margelo/react-native-vision-camera #376: 🐛 app crash on Android: building a face-detector in mlkit. how to  debug jni?

rodgomesc, 2021-08-19:
### What were you trying to do?

build a face detector using mlkit and frame processors.

### Reproduceable Code

```tsx
package com.visioncamerafacedetector;

import android.annotation.SuppressLint;
import android.graphics.PointF;
import android.graphics.Rect;
import android.media.Image;
import android.util.Log;

import androidx.camera.core.ImageProxy;

import com.facebook.react.bridge.Arguments;
import com.facebook.react.bridge.WritableMap;
import com.facebook.react.bridge.WritableNativeArray;
import com.facebook.react.bridge.WritableNativeMap;
import com.google.android.gms.tasks.Task;
import com.google.android.gms.tasks.Tasks;
import com.google.mlkit.vision.common.InputImage;
import com.google.mlkit.vision.face.Face;
import com.google.mlkit.vision.face.FaceContour;
import com.google.mlkit.vision.face.FaceDetection;
import com.google.mlkit.vision.face.FaceDetector;

import com.google.mlkit.vision.face.FaceDetectorOptions;
import com.mrousavy.camera.frameprocessor.FrameProcessorPlugin;

import java.util.List;

public class VisionCameraFaceDetectorPlugin extends FrameProcessorPlugin {

  FaceDetectorOptions options =
    new FaceDetectorOptions.Builder()
      .setPerformanceMode(FaceDetectorOptions.PERFORMANCE_MODE_ACCURATE)
      .setContourMode(FaceDetectorOptions.CONTOUR_MODE_ALL)
      .setClassificationMode(FaceDetectorOptions.CLASSIFICATION_MODE_ALL)
      .setMinFaceSize(0.15f)
      .build();

  FaceDetector faceDetector = FaceDetection.getClient(options);

  private WritableMap processBoundingBox(Rect boundingBox) {
    WritableMap bounds = Arguments.createMap();

    bounds.putInt("x", boundingBox.left);
    bounds.putInt("y", boundingBox.top);
    bounds.putInt("width", boundingBox.width());
    bounds.putInt("height", boundingBox.height());

    return bounds;
  }

  private WritableMap  processFaceContours(Face face) {
    // All faceContours
    int[] faceContoursTypes =
      new int[] {
        FaceContour.FACE,
        FaceContour.LEFT_EYEBROW_TOP,
        FaceContour.LEFT_EYEBROW_BOTTOM,
        FaceContour.RIGHT_EYEBROW_TOP,
        FaceContour.RIGHT_EYEBROW_BOTTOM,
        FaceContour.LEFT_EYE,
        FaceContour.RIGHT_EYE,
        FaceContour.UPPER_LIP_TOP,
        FaceContour.UPPER_LIP_BOTTOM,
[... 297 lines cut ...]
08-19 13:06:40.763 18404 18404 F DEBUG   :       #94 pc 0000000000594622  /apex/com.android.art/lib64/libart.so (art::interpreter::Execute(art::Thread*, art::CodeItemDataAccessor const&, art::ShadowFrame&, art::JValue, bool, bool)+306) (BuildId: f2321c7c4ca8d13e570046cc41560503)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #95 pc 0000000000958e6f  /apex/com.android.art/lib64/libart.so (artQuickToInterpreterBridge+1007) (BuildId: f2321c7c4ca8d13e570046cc41560503)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #96 pc 00000000003a01bc  /apex/com.android.art/lib64/libart.so (art_quick_to_interpreter_bridge+140) (BuildId: f2321c7c4ca8d13e570046cc41560503)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #97 pc 0000000000394d14  /apex/com.android.art/lib64/libart.so (art_quick_invoke_stub+756) (BuildId: f2321c7c4ca8d13e570046cc41560503)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #98 pc 000000000041d6fa  /apex/com.android.art/lib64/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+218) (BuildId: f2321c7c4ca8d13e570046cc41560503)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #99 pc 000000000081a16e  /apex/com.android.art/lib64/libart.so (art::JValue art::InvokeVirtualOrInterfaceWithJValues<art::ArtMethod*>(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, art::ArtMethod*, jvalue const*)+478) (BuildId: f2321c7c4ca8d13e570046cc41560503)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #100 pc 000000000087974f  /apex/com.android.art/lib64/libart.so (art::Thread::CreateCallback(void*)+1343) (BuildId: f2321c7c4ca8d13e570046cc41560503)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #101 pc 00000000000c783a  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+58) (BuildId: 85fd828dbc591a601280b411dd4616a6)
08-19 13:06:40.763 18404 18404 F DEBUG   :       #102 pc 0000000000060017  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+55) (BuildId: 85fd828dbc591a601280b411dd4616a6)
```

### Device

Android Emulator (Android 12)

### VisionCamera Version

^2.5.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-20:
This looks like it's the same error as in https://github.com/mrousavy/react-native-vision-camera/issues/350, I tried to fix it here: https://github.com/mrousavy/react-native-vision-camera/pull/351 but that fix apparently didn't work. I'm not quite sure where I am using that stale reference, would be interesting to see which line is causing the crash...

rodgomesc, 2021-08-20:
I did some tests and found something interesting

`faceContoursTypesMap.putArray(faceContoursTypesStrings[contour.getFaceContourType() - 1], pointsArray);`

it's a lot of `x` and `y` values going into that array. if I put this static value like below, the `putArray`  only keep last value and the the app takes about 2 minutes longer to break

`faceContoursTypesMap.putArray("test", pointsArray)`;

here is a sample of the output in javascript if that helps;
https://pastebin.com/ue73U9a2

mrousavy, 2021-08-20:
So since it breaks after some specific amount of time, and by re-using arrays to avoid excessive allocation I can conclude that it's definitely a problem with the Java Garbage Collector. There is a weak reference somewhere in the C++ code which is being used without checking for nullability and without holding a strong reference to it, and once the Java GC runs, that reference is gone. 

Maybe somewhere I use `alias_ref` where it should be `local_ref` or `global_ref` instead? 🤔🤔🤔

rodgomesc, 2021-08-20:
I believe you'll find out soon, I'm very excited about this lib  :)

closed 2021-08-20 by mrousavy

mrousavy, 2021-09-24:
The C++ crash that occured after running a Frame Processor for a long time (on Hermes 0.8.x+) is finally fully fixed in #455 + [Reanimated v2.3.0-beta.1](https://github.com/software-mansion/react-native-reanimated/releases/tag/2.3.0-beta.1).