margelo/react-native-vision-camera #382: ❓ TakePhoto return base64 instead of a file instance

rodgomesc, 2021-08-22:
### What feature or enhancement are you suggesting?

do you have plans to implement base64 format for `Camera.takePhoto()?`

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

i don't want to save the image taken in my file system

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-08-23:
With the current Bridge architecture, returning base64 from a native function will be way slower than writing to a file and returning it's path because the entire base64 string has to be serialized and copied a lot.

I might consider this when TurboModules land and the serialization can be avoided, but I'd prefer to implement this instead: https://github.com/mrousavy/react-native-vision-camera/issues/69

rodgomesc, 2021-08-23:
thanks for the quick feedback, so i will close this for now

closed 2021-08-23 by rodgomesc

rgomesrn, 2022-08-25:
this is achievable at that point?