margelo/react-native-vision-camera #388: C/C++ debug|x86 : CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

trta2352, 2021-08-23:
### What were you trying to do?

Tried to build the app in Android Studio. On iOS it works fine. 

Using react-native: 0.64.2

### What happened instead?

Build failed with error: 

/Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ debug|x86 : CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android
FOLLY_JSON_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android
JS_ENGINE_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android
REACT_NATIVE_JNI_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android
REANIMATED_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android

### Device

Samsung s8

### VisionCamera Version

2.6.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-08-23:
does it work if you try to build twice?

trta2352, 2021-08-23:
Yes I tried to build it multiple times together with cleaning the project.

trta2352, 2021-08-24:
I deleted and reinstalled node_modules and pods. The build went trough but when I wanted to use the camera a different error occurred. 

[image]

mrousavy, 2021-08-24:
Do you have reanimated installed? Can you run the `example/` app here? If yes, compare your project to the example project to see what's different...

trta2352, 2021-08-24:
I tried to build the example project and the exact same error occurred. 

C/C++ debug|x86 : CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/desktop/temp/react-native-vision-camera/android
FOLLY_JSON_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/desktop/temp/react-native-vision-camera/android
REACT_NATIVE_JNI_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/desktop/temp/react-native-vision-camera/android
REANIMATED_LIB
    linked by target "VisionCamera" in directory /Users/jakatertinek/desktop/temp/react-native-vision-camera/android

mrousavy, 2021-08-24:
Can you describe how you build the project? Did you run `yarn` in the directory? Did gradle sync successfully? Did you actually hit run in Android Studio? 

Maybe check if you have an outdated Android SDK/Studio/NDK version

trta2352, 2021-08-24:
This is how I ran my project and where the issue happened. I opened the android folder in Android Studio. Project gradle sync worked without issue. I ran "Make project" and then ran it on device. It ran without issue and installed correctly. When I pressed the button that opens the camera, the **error CameraViewManager cannot be cast to ViewGroupManager happened**.  I went back to Android Studio, ran "Clean project" and then "Refresh linked C++ projects". This is where the error with the missing variables keeps happening. 

I have the latest version of Android Studio, SDK and NDK installed. Although I'm not using them in the project. 

```
buildscript {
   ext {
         RNNKotlinVersion = "1.3.61"
        buildToolsVersion = "30.0.3"
        minSdkVersion = 21
        compileSdkVersion = 30
        targetSdkVersion = 30
        ndkVersion = "20.1.5948944"
    }
```
Now I switched from the Android Studio M1 version to the normal one and the same occurs as before (so error CameraViewManager cannot be cast to ViewGroupManager happened) but when refreshing the c++ linking this happens
```
* What went wrong:
Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'.
> Build command failed.
  Error while executing process /Users/jakatertinek/Library/Android/sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a VisionCamera}
  ninja: Entering directory `/Users/jakatertinek/Desktop/ICC/mobile-cop26-digital-id-verification/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a'
  
  ninja: error: '../../../../build/react-native-0.64.2.aar/jni/armeabi-v7a/libjscexecutor.so', needed by '../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it

```

Now I did also try running the example project. Downloaded it and ran yarn install. Try running it on the phone with "Run 'app'" and got error: 

```
Build command failed.
Error while executing process /Users/jakatertinek/Library/Android/sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Users/jakatertinek/Desktop/temp/react-native-vision-camera/android/.cxx/cmake/debug/arm64-v8a VisionCamera}
ninja: Entering directory "/Users/jakatertinek/Desktop/temp/react-native-vision-camera/android/.cxx/cmake/debug/arm64-v8a"

ninja: error: "../../../../build/third-party-ndk/hermes/jni/arm64-v8a/libhermes.so", needed by "../../../../build/intermediates/cmake/debug/obj/arm64-v8a/libVisionCamera.so", missing and no known rule to make it

```

After this error happened I ran "Refresh linked C++ projects" and the app ran on the phone. Now on the phone the app opens. You grant permission... But another error "parallel-video-processing-not-supported". 

I will try to find if there are any obvious differences between my project and the example project.

mrousavy, 2021-08-24:
Okay so first off those are all very different errors.

> the error CameraViewManager cannot be cast to ViewGroupManager happened

This seems really weird. I have never seen that error before and do not know how to reproduce it, if you can provide me a GitHub repository that I can run to reproduce it that would help me.

> but when refreshing the c++ linking this happens
>  `ninja: error: '../../../../build/react-native-0.64.2.aar/jni/armeabi-v7a/libjscexecutor.so', needed by '../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libVisionCamera.so', missing and no known rule to make it`

This should be fixed by running the build twice. See https://github.com/mrousavy/react-native-vision-camera/issues/341

> Now I did also try running the example project. Downloaded it and ran yarn install. Try running it on the phone with "Run 'app'" and got error:
> `ninja: error: "../../../../build/third-party-ndk/hermes/jni/arm64-v8a/libhermes.so", needed by "../../../../build/intermediates/cmake/debug/obj/arm64-v8a/libVisionCamera.so", missing and no known rule to make it`

Same as above. See See https://github.com/mrousavy/react-native-vision-camera/issues/341. Also you want to run `yarn` in both the root directory and in the `example/` directory. Then open `example/android/` in Android Studio.

> But another error "parallel-video-processing-not-supported".

This happens in the Example app? See ["the `supportsParallelVideoProcessing` prop"](https://mrousavy.github.io/react-native-vision-camera/docs/guides/devices#the-supportsparallelvideoprocessing-prop), or related issue: https://github.com/mrousavy/react-native-vision-camera/issues/315

Maybe this whole thing has to do with you using an M1 laptop? the `CameraViewManager cannot be cast to ViewGroupManager` still seems so weird to me....

HoraceDiesel, 2021-08-25:
I am on M1, i see the same build error often.

`CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
linked by target "VisionCamera" in directory {app}/node_modules/react-native-vision-camera/android
FOLLY_JSON_LIB
linked by target "VisionCamera" in directory {app}/node_modules/react-native-vision-camera/android
REACT_NATIVE_JNI_LIB
linked by target "VisionCamera" in directory {app}/node_modules/react-native-vision-camera/android
REANIMATED_LIB
linked by target "VisionCamera" in directory {app}/node_modules/react-native-vision-camera/android`

trta2352, 2021-08-25:
> I am on M1, i see the same build error often.
>
> `CMake Error: The following variables are used in this project, but they are set to NOTFOUND. Please set them or make sure they are set and tested [...]

So is there any way to work around this?

HoraceDiesel, 2021-08-25:
> So is there any way to work around this?

I do gradlew clean and sync until it doesn't complain about it. otherwise rm the whole node modules and clean the project, then do full setup of the project.

mrousavy, 2021-08-25:
hmm that's really weird, can you provide a repo where I can reproduce this? again, in this repo (`example/`) everything builds fine for me and for the GitHub Action (CI)...

Just to confirm @HoraceDiesel can you build the `example/` app here?

bear-junior, 2021-09-04:
@mrousavy  I believe the problem here is that on AppCenter (or random CI) the command is `./gradlew clean assembleRelease` , but the tasks from gradle are executed if the task name contains 'externalNativeBuild', so the tasks execution is:

> Task :react-native-vision-camera:extractAARHeaders
> Task :react-native-vision-camera:extractJNIFiles
> Task :react-native-vision-camera:prepareHermes
> Task :react-native-vision-camera:prepareJSC
> Task :react-native-vision-camera:prepareThirdPartyNdkHeaders
> Task :react-native-vision-camera:externalNativeBuildCleanDebug
> Task :react-native-vision-camera:externalNativeBuildCleanRelease
> Task :react-native-vision-camera:clean

so everything that has been added to the build folder is not here anymore, so the paths from CMake file are incorrect and it returns NOTFOUND which makes totally sense in this case. Those tasks should be called after clean task is done.

So the problem here is not with the paths, but with the tasks order and btw I suggest to add doLast to prepareHermes as well.

The issue is visible for /example project as well, just run ./gradlew clean assembleRelease.

mrousavy, 2021-09-04:
@bear-junior nice observation, could you create a PR that uses doLast instead? Then we'll also see if the CI here (GitHub Actions) still builds

mrousavy, 2021-09-04:
@bear-junior created a PR to fix this here: #407 🎉 
can anyone here confirm that the PR (#407) fixes this issue?

dkocich, 2021-09-11:
- the same issue here in Linux mint 20.2

```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:configureCMakeDebug'.
> C/C++: /home/dk/it/PhotoMap/node_modules/react-native-vision-camera/android/CMakeLists.txt debug|armeabi-v7a : CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
  Please set them or make sure they are set and tested correctly in the CMake files:
  FBJNI_LIB
      linked by target "VisionCamera" in directory /home/dk/it/PhotoMap/node_modules/react-native-vision-camera/android
  FOLLY_JSON_LIB
      linked by target "VisionCamera" in directory /home/dk/it/PhotoMap/node_modules/react-native-vision-camera/android
  JS_ENGINE_LIB
      linked by target "VisionCamera" in directory /home/dk/it/PhotoMap/node_modules/react-native-vision-camera/android
  REACT_NATIVE_JNI_LIB
      linked by target "VisionCamera" in directory /home/dk/it/PhotoMap/node_modules/react-native-vision-camera/android
  REANIMATED_LIB
      linked by target "VisionCamera" in directory /home/dk/it/PhotoMap/node_modules/react-native-vision-camera/android
```

bear-junior, 2021-09-13:
@dkocich do you use [v2.7.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.7.0)?

alexstanbury, 2021-09-23:
FWIW I'm having a hell of a time getting my own frame processor plugins working, during my attempts I've noticed that updating the Android Gradle Plugin version from 4.2.1 to 7.0.2 when prompted by Android Studio Arctic Fox immediately causes the build to fail with the Cmake errors above.

mrousavy, 2021-09-23:
> updating the Android Gradle Plugin version from 4.2.1 to 7.0.2 

that's a pretty high version bump, afaik React Native uses 4.x.x, so bumping to 7.x.x can only introduce breaking changes.

Hirbod, 2021-12-20:
This error is still happening for me on 2.10.0 (Expo SDK 44, Custom Dev Client)

closed 2022-01-02 by mrousavy

proffnick, 2022-07-28:
For some reasons for me, the issue got resolved by building the project twice! like asked by @mrousavy

ucheNkadiCode, 2022-11-15:
I was experiencing this problem in CMAKE for Reanimated. To solve this I performed from the root directory

`cd android &&  ./gradlew clean assembleRelease`

And some how this worked

najamuddin11, 2023-10-24:
go to node_modules/libraray/android/ which is causing the problem.

make CMakeLists.txt empty. and remove .cxx folder. 

for me it worked. you have to repeat the process for each 
npm run android.

PR #407 (2021-09-06): fix: Fix Android tasks execution
## What
This PR fixes Android tasks execution.

## Changes
This PR changes the tasks order and prepare everything correctly to be executed by task externalNativeBuild.

## Tested on
Local machine for example project and on AppCenter CI.

## Related issues
    * Fixes #338 
    * Fixes #388
PR #694 (2022-01-02): fix: Fix Android build on React Native 0.65 and older

## What

## Changes

## Tested on

## Related issues

* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/654
* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/588
* Fixes #546 
* Fixes #513 
* Fixes #422 
* Fixes #390 
* Fixes #388 
* Fixes #369 
* Fixes #365 
* Fixes #341 
* Fixes #338 
* Fixes #269 
* Overrides #370 
* Overrides #676 
* Resolves https://github.com/mrousavy/react-native-vision-camera/issues/663