margelo/react-native-vision-camera #389: 🐛 I try to use Frame Processors but I have this error: Frame Processors are not enabled

Raidouane, 2021-08-23:
### What were you trying to do?

I'm trying to use Frame Processors to implement face detection and then a text OCR, but I have an error on Frame Processors.

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
        'worklet'
        console.log('processFaces', frame);
        // const faces = processFaces(frame)
        // if (qrCodes.length > 0) {
        // runOnJS(onQRCodeDetected)(qrCodes[0])
        // }
    }, [processFaces]);

return (
  <Camera
                        style={styles.preview}
                        device={device}
                        isActive
                        photo
                        ref={cameraRef}
                        frameProcessor={frameProcessor}
                    />
);
```

### What happened instead?

Error: Frame Processors are not enabled. See https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting
    at Camera.assertFrameProcessorsEnabled (index.bundle?platfor…minify=false:395335)
    at Camera.setFrameProcessor (index.bundle?platfor…minify=false:395341)
    at Camera.onLayout (index.bundle?platfor…minify=false:395359)
    at Object.invokeGuardedCallbackImpl (index.bundle?platfor…&minify=false:12851)
    at invokeGuardedCallback (index.bundle?platfor…&minify=false:12945)
    at invokeGuardedCallbackAndCatchFirstError (index.bundle?platfor…&minify=false:12949)
    at executeDispatch (index.bundle?platfor…&minify=false:13021)
    at executeDispatchesInOrder (index.bundle?platfor…&minify=false:13041)
    at executeDispatchesAndRelease (index.bundle?platfor…&minify=false:14116)
    at executeDispatchesAndReleaseTopLevel (index.bundle?platfor…&minify=false:14125)

### Device

Iphone X - ios 14.7.1

### VisionCamera Version

0.1.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rodgomesc, 2021-08-23:
can you please provide:

- a valid visio-camera version
- a full code example?

i can see in your example that you are not checking if device is not null,

please give a look in this example;
https://github.com/mrousavy/react-native-vision-camera/tree/main/example

mrousavy, 2021-08-23:
> VisionCamera Version: 0.1.1

what? 🤨 latest is 2.6.1

Raidouane, 2021-08-23:
> > VisionCamera Version: 0.1.1
> 
> what? 🤨 latest is 2.6.1

HI, sorry It was another version package, I edited my post

Raidouane, 2021-08-23:
> can you please provide:
> 
> * a valid visio-camera version
> * a full code example?
> 
> i can see in your example that you are not checking if device is not null,
> 
> please give a look in this example;
> https://github.com/mrousavy/react-native-vision-camera/tree/main/example

`
const IdentityVerification = ({
                                  isLoading,
                                  error,
                              }: any) => {
    const {t} = useTranslation();
    const cameraRef = useRef(null);
    const devices = useCameraDevices();
    const device = devices.back;
    const frameProcessor = useFrameProcessor((frame) => {
        'worklet'
        console.log('processFaces', frame);
        // const faces = processFaces(frame)
        // if (qrCodes.length > 0) {
        // runOnJS(onQRCodeDetected)(qrCodes[0])
        // }
    }, [processFaces]);
    console.log('device.supportsParallelVideoProcessing', device.supportsParallelVideoProcessing)
    const handleTakePicture = async () => {
        if (!cameraRef.current)  return;

        const photo = await cameraRef.current.takePhoto({
            flash: 'off',
            enableAutoStabilization: true,
        });
        console.log('handleTakePicture uri', photo);
    };
    if (!device) return null;

    return (
        <Block style={styles.container}>
            <Header
                back
                transparent
                white
            />
            <Block flex={2} style={styles.cameraContainer}>
                <Block style={styles.previewContainer}>
                    <Camera
                        style={styles.preview}
                        device={device}
                        isActive
                        photo
                        ref={cameraRef}
                        frameProcessor={device.supportsParallelVideoProcessing ? frameProcessor : undefined}
                    />
                </Block>
            </Block>
            <Block area style={{flex: 0, flexDirection: 'row', justifyContent: 'center'}}>
                <TouchableOpacity onPress={handleTakePicture} style={styles.capture}>
                    <Icon
                        family='Feather'
                        size={50}
                        name='camera'
                        color={materialTheme.COLORS.PRIMARY_INVERT}
                    />
                </TouchableOpacity>
            </Block>
        </Block>
    );
};

export default IdentityVerification;
`

mrousavy, 2021-08-24:
1. Do you have react-native-reanimated installed? If yes, which version?
2. Is that the first screen in the app? Maybe it's a race condition, could you somehow delay the loading of the `device` with a custom `setTimeout` for something like 5 seconds?

Raidouane, 2021-08-25:
> 1. Do you have react-native-reanimated installed? If yes, which version?
> 2. Is that the first screen in the app? Maybe it's a race condition, could you somehow delay the loading of the `device` with a custom `setTimeout` for something like 5 seconds?
> 3. Could you please share the native Xcode logs? (Log window at the bottom in Xcode)
> 
> I have a feeling that you have an incorrect Reanimated version, so it disables Frame Processors here:
> 
> https://github.com/mrousavy/react-native-vision-camera/blob/f3e1df009e5ec4eaf02744a216695de7d69ff05c/ios/Frame%20Processor/FrameProcessorRuntimeManager.mm#L22-L35

closed 2021-08-25 by Raidouane

mrousavy, 2021-08-25:
Oh, yes, the remote JS debugger does not work with JSI. You will need to use Flipper or the Safari Debugger if you use JSC.

rodgomesc, 2021-09-30:
the same is happening to me but only on back camera