margelo/react-native-vision-camera #392: ❓Is this library support square video?

AlkanV, 2021-08-29:
### Question

Hello,

is it possible to record square videos with this package?

thanks

### VisionCamera Version

Latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-08-30:
Yes it is. Try choosing a `format` that has square video dimensions (see ["Camera Formats"](https://mrousavy.github.io/react-native-vision-camera/docs/guides/formats)).

If none is available, then it isn't supported by your Phone and you will have to manually crop the video somehow (post-processing)

AlkanV, 2021-08-30:
Hello, @mrousavy,

Thank you so much for the answer. 

Best wishes,

closed 2021-08-30 by mrousavy

nica0012, 2022-03-27:
Hi @AlkanV What did you end up using to get square video?

powerpovar, 2022-03-28:
Hi @AlkanV, same question, how did you achieve the square video?

AlkanV, 2022-05-22:
@nica0012 @powerpivar i did not use this library yet so i have no answer for it.