margelo/react-native-vision-camera #401: ❓ Integrate MLKit to swift in IOS for FrameProcessor Plugin

rodgomesc, 2021-09-02:
### Question

after finishing the implementation of faceDetector on android, it's ios' turn, after looking between swift and Objective-C I decided to use swift which seemed easier, since I've never worked with either of them.

I have a problem that as silly as it seems took me a few hours and I still can't solve it, I did the implementation of s.dependency `GoogleMLKit/FaceDetection` in my [vision-camera-face-detector.podspec](https://github.com/rodgomesc/vision-camera-face-detector/blob/master/vision-camera-face-detector.podspec), and I see the [mlkit dependencies](https://github.com/rodgomesc/vision-camera-face-detector/blob/dd103a6b2615147af34b0cf4eeda814f3610be21/example/ios/Podfile.lock#L64) being installed by cocoapods, but when I go to compile my example project that uses frameprocessor, I always run into this error;

[image]

everything seems right to me since the mlkit documentation itself redirects me [to this lib](https://github.com/googlesamples/mlkit/blob/master/ios/quickstarts/vision/VisionExample/CameraViewController.swift) that does the import in the same way

I've already noticed the other plugins created for vision camera but they all use objective-C and the example you have of swift is a code example only that doesn't show an implementation of dependencies.

to look in my poc project just look at [this repo](https://github.com/rodgomesc/vision-camera-face-detector)

Cheers.

### VisionCamera Version

^2.6.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-09-02:
A workaround would be to add `pod GoogleMLKit/FaceDetection` to your example/ios/Podfile, but that shouldn't be needed... 🤔

rodgomesc, 2021-09-02:
> This might be a dumb question - but did you run `pod install` in the example/ios app? What happens when you do that?

sure, all pods are installed normally including mlkit dependencies

rodgomesc, 2021-09-02:
> A workaround would be to add `pod GoogleMLKit/FaceDetection` to your example/ios/Podfile, but that shouldn't be needed... 🤔

that doesn't work too

rodgomesc, 2021-09-03:
seems like iam not the only one experiencing this.

https://stackoverflow.com/questions/67363271/missing-required-module-mlkit-error-while-using-a-framework-that-uses-mlkit

rodgomesc, 2021-09-03:
solved.

for some reason, google quickstart project of mlkit using swift, and other fonts on internet uses

`import MLKit`

when the correct way should be 

`
import MLKitFaceDetection
import MLKitVision
`

maybe it's a undocumented recent changes.

closed 2021-09-03 by rodgomesc

CodingWithNobody, 2023-05-01:
Try this https://www.youtube.com/watch?v=mv4c2uTxR4U