margelo/react-native-vision-camera #405: Microphone permission was denied! on Example app🐛

redlac, 2021-09-02:
### What were you trying to do?

I was trying to run the Example app for Android. 

### What happened instead?

After running the app from Android Studio, recording fails and the following error message appears:

### Relevant log output

```shell
2021-09-02 11:14:46.542 25423-25467/com.mrousavy.camera.example E/ReactNativeJS: 'Recording failed!', { userInfo: null, message: '[permission/microphone-permission-denied] The Microphone permission was denied! If you want to record Video without sound, pass audio={false}.', cause: { stacktrace: 'com.mrousavy.camera.MicrophonePermissionError: [permission/microphone-permission-denied] The Microphone permission was denied! If you want to record Video without sound, pass audio={false}.\n\tat com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:28)\n\tat com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:89)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)\n', message: '[permission/microphone-permission-denied] The Microphone permission was denied! If you want to record Video without sound, pass audio={false}.' }, code: 'permission/microphone-permission-denied' }
```

### Device

Samsung Galaxy S8

### VisionCamera Version

2.6.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-09-02:
Did it ask you for microphone permission/did you deny?

redlac, 2021-09-03:
I did not see a dialog for permissions.

PR #418 (2021-09-11): fix: Fix Microphone usage request not showing on Android 11
## What

This PR fixes an issue on Android 11 which wasn't showing the instruction and button to grant microphone permission usage. Fix made on example app.

## Changes
Now microphone permission usage is verified looking if it has been authorized or not rather than compares with an initial state value `not-determined`.

## Tested on
Samsung A10, Android 11

## Related issues
Closes #405
closed 2021-09-11 by mrousavy