margelo/react-native-vision-camera #408: ❓  Barcode scanning frame processor plugin (mlkit) for Android, throws an error

Curti-s, 2021-09-06:
### Question

How to resolve unkown JNIS type errors.

### What I tried

Hi and many thanks for this awesome lib.

I have tried setting up the frame processor plugin for barcode scanning using mlkit (https://developers.google.com/ml-kit/vision/barcode-scanning/android#before-you-begin) in Android and each time I launch the camera, the frame processor throws an error:

`Error: Exception in HostFunction: Received unknown JNI type "class com.google.android.gms.tasks.zzw"! Cannot convert to jsi::Value.`

The implementation for this can be found in this repo https://github.com/Curti-s/test_react_native_vision_camera
Please advice what I could be doing wrong. Thanks.

### VisionCamera Version

2.6.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-09-06:
Hi!

The error message is pretty descriptive I think? Your frame processor plugin is returning a Java value of type `com.google.android.gms.task.zzw` (`Task<T>`), and the JSI Runtime doesn't know how it can convert that to a JavaScript value. It shouldn't know that, because that cannot be represented as a JS value, what do you expect it to be?

Here's where your frame processor plugin is returning that `Task<T>` which cannot be converted to a JS value:

https://github.com/Curti-s/test_react_native_vision_camera/blob/9959a328475ed10963ced081e31a72be2807ff20/android/app/src/main/java/com/test_vision_camera/QRcodeFrameProcessor.java#L76

 I assume you want to return a list of objects (results), but looking at your code it seems like you use the _asynchronous_ API for the Task:

https://github.com/Curti-s/test_react_native_vision_camera/blob/9959a328475ed10963ced081e31a72be2807ff20/android/app/src/main/java/com/test_vision_camera/QRcodeFrameProcessor.java#L49

instead of the _synchronous_ one.

You have two possibilities:

### Use the synchronous API, directly return Barcodes from Frame Processor Plugin

This API will look like this for JS:

```js
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  const barcodes = scanBarcodes(frame)
  console.log(`Detected: ${barcodes}`)
}, [])
```

To achieve this, you must not add a `addOnSuccessListener` and `addOnFailureListener`, but instead use the `Tasks.await` method to synchronously await the result instead of receiving it in the on success listener.

```java
Task<List<Barcode>> task = scanner.process(image);
List<Barcode> barcodes = Tasks.await(task);
```

This works perfectly fine if you're not doing any other work in your frame processor (e.g. facial recognition)

### Use an asynchronous event system

This API will look something like this for JS:

```js
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  BarcodeAPI.scanBarcodes(frame)
}, [])

useEffect(() => {
  BarcodeAPI.addOnDetectListener((barcodes) => {
    console.log(`Detected: ${barcodes}`)
  }
}, [])
```

For this, you can continue to use the asynchronous API (`addOnSuccessListener` and `addOnFailureListener`) which executes on a separate thread, and then you can use the React Event API (see [the React Native docs: "Sending Events to JavaScript"](https://reactnative.dev/docs/native-modules-android#sending-events-to-javascript)) to send the actual detected barcodes to the `BarcodeAPI.addOnDetectListener` call in the `useEffect` hook at some later point in time (when they're actually detected). This then runs on the normal React-JS thread.

Use whatever API you feel like looks easier to you, but just know that there are two possibilities for async and sync APIs.

Curti-s, 2021-09-06:
Thanks @mrousavy for this and for the pointers as well. Appreciate that.

closed 2021-09-06 by Curti-s

mrousavy, 2021-09-06:
you're welcome my g let me know if u got it working