margelo/react-native-vision-camera #411: ❓ Is react-native-reanimated now required?

jzxchiang1, 2021-09-06:
### Question

Is react-native-reanimated now required?

### What I tried

In 2.4.1, I could build my iOS and Android app without having react-native-reanimated installed.

However, when I upgraded to 2.7.0 just now, both my iOS and Android builds failed due to a) a missing RNReanimated header for iOS, and b) a missing ":react-native-reanimated" project for Android.

According to the docs, the Reanimated library is only needed for frame processors, so I thought it was optional. 😢

### VisionCamera Version

2.7.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-09-06:
Hi!
Yes, reanimated is required for VisionCamera since 2.5.0.
It might be possible to make Reanimated an optional dependency, (definitely possible on iOS, but a bit difficult on Android... no idea how to fix the CMake/build.gradle build when REA is not found)

I'm going to take a look at it, to find out whether I can make REA optional again or have to mention in the docs to state otherwise.

jzxchiang1, 2021-09-06:
Oh man, gotcha 😞. I'm curious: why was it made required when it worked just fine before as optional?

Yeah, unfortunately I can't install reanimated quite yet (since it requires JSC or Hermes, but my Android app uses v8)...

Please let me know what you find!

jzxchiang1, 2021-09-06:
Also would it be possible to use reanimated v1 or does it have to be v2?

mrousavy, 2021-09-06:
It has to be v2 because Frame Processors require parts of Reanimated's Worklet API to work.
I need to somehow pass a flag to the native C++ code when CMake/Gradle cannot find REA to skip linking that, no idea how that'll work...

jzxchiang1, 2021-09-06:
Hm... why is it when I disable Frame Processors in the GCC flag, it still requires linking Reanimated?

closed 2022-01-02 by mrousavy