margelo/react-native-vision-camera #413: ❓ Error: Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError: [system/view-not-found] The given view (ID 165) was not found in the view manager.

Curti-s, 2021-09-07:
### Question

Each time I navigate from the Vision Camera screen, this error pops up.

### What I tried

This is the full error message for more clarity

`  Error: Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError: [system/view-not-found] The given view (ID 165) was not found in the view manager.

This error is located at:
    in Camera (at createAnimatedComponent.js:514)
    in AnimatedComponent(Camera) (at createAnimatedComponent.js:525)
    in AnimatedComponentWrapper (at CameraPage.tsx:206)
    in TapGestureHandler (at CameraPage.tsx:205)
    in RCTView (at View.js:32)
    in View (at createAnimatedComponent.js:514)
    in AnimatedComponent(View) (at createAnimatedComponent.js:525)
    in AnimatedComponentWrapper (at CameraPage.tsx:204)
    in PinchGestureHandler (at CameraPage.tsx:203)
    in RCTView (at View.js:32)
    in View (at CameraPage.tsx:201)
    in CameraPage (at gestureHandlerRootHOC.tsx:13)
    in GestureHandlerRootView (at GestureHandlerRootView.android.tsx:26)
    in GestureHandlerRootView (at gestureHandlerRootHOC.tsx:12)
    in gestureHandlerRootHOC(CameraPage) (created by WrappedComponent)
    in WrappedComponent (at renderApplication.js:48)
    in RCTView (at View.js:32)
    in View (at AppContainer.js:106)
    in RCTView (at View.js:32)
    in View (at AppContainer.js:133)
    in AppContainer (at renderApplication.js:41)
    in CameraPage(RootComponent) (at renderApplication.js:57), js engine: hermes`

### VisionCamera Version

2.6.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-09-07:
Why is this a question? This should be a bug report.

I think this is a race condition in the native code, I'll try to take a look at that, thanks

Curti-s, 2021-09-07:
I'v filed one here https://github.com/mrousavy/react-native-vision-camera/issues/414
closing this now.

closed 2021-09-07 by Curti-s