margelo/react-native-vision-camera #414: 🐛 ❓ Error: Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError: [system/view-not-found] The given view (ID 165) was not found in the view manager.

Curti-s, 2021-09-07:
### What were you trying to do?

Each time I navigate from the Vision Camera screen, this error pops up.

This is the full error message for more clarity

`  Error: Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError: [system/view-not-found] The given view (ID 165) was not found in the view manager.
`

### Reproduceable Code

```tsx
`return (
    <View style={styles.container}>
      <Reanimated.View style={StyleSheet.absoluteFill}>
        <ReanimatedCamera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          animatedProps={animatedProps}
          frameProcessor={frameProcessor}
          frameProcessorFps={30}
          fps={5}
        />
      </Reanimated.View>
    </View>`
```

### What happened instead?

The app crashes w/ a 
`  Error: Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError: [system/view-not-found] The given view (ID 165) was not found in the view manager.
`

### Relevant log output

```shell
`
This error is located at:
    in Camera (at createAnimatedComponent.js:514)
    in AnimatedComponent(Camera) (at createAnimatedComponent.js:525)
    in AnimatedComponentWrapper (at CameraPage.tsx:206)
    in TapGestureHandler (at CameraPage.tsx:205)
    in RCTView (at View.js:32)
    in View (at createAnimatedComponent.js:514)
    in AnimatedComponent(View) (at createAnimatedComponent.js:525)
    in AnimatedComponentWrapper (at CameraPage.tsx:204)
    in PinchGestureHandler (at CameraPage.tsx:203)
    in RCTView (at View.js:32)
    in View (at CameraPage.tsx:201)
    in CameraPage (at gestureHandlerRootHOC.tsx:13)
    in GestureHandlerRootView (at GestureHandlerRootView.android.tsx:26)
    in GestureHandlerRootView (at gestureHandlerRootHOC.tsx:12)
    in gestureHandlerRootHOC(CameraPage) (created by WrappedComponent)
    in WrappedComponent (at renderApplication.js:48)
    in RCTView (at View.js:32)
    in View (at AppContainer.js:106)
    in RCTView (at View.js:32)
    in View (at AppContainer.js:133)
    in AppContainer (at renderApplication.js:41)
    in CameraPage(RootComponent) (at renderApplication.js:57), js engine: hermes``
```

### Device

Nokia 5.1 model:TA_1053 

### VisionCamera Version

2.6.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-09-08:
Thanks for the bug report. I will take a look. 

This is a race condition in the native code, and you can fix this on your end by removing the `frameProcessor` from the Camera.

closed 2021-09-27 by mrousavy

mrousavy, 2021-09-27:
Fixed in 2.8.3 🎉 🚀

PR #459 (2021-09-27): fix: Fix `system/view-not-found` error caused by `setFrameProcessor` race condition

## What

## Changes

## Tested on

## Related issues

* Fixes #414
* Fixes #457