margelo/react-native-vision-camera #421: ❓

Ankit-Micron, 2021-09-13:
### Question

Hey 
I am not able to use react-native-vision-camera in my project. Whenever I try to install and run it, its showing this error:

> Configure project :react-native-vision-camera
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.

> Task :react-native-vision-camera:compileDebugKotlin FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:compileDebugKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

### What I tried

I upgraded the gradle to 6.7.1 which was working in (Colorwaver app) and I also updated the app with latest sdkVersion.

### VisionCamera Version

2.7.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-09-13:
Hi! This should be a bug report, not a question.

Can you share the full logs? There is no error included in what you shared, only a warning. Also compare your project with the `example/` project here to find out if there are some version mismatches

closed 2021-09-13 by Ankit-Micron

bekatd, 2022-04-22:
#1012 issuecomment-1106575501