margelo/react-native-vision-camera #422: 🐛 A problem occurred configuring root project 'rn_starter'.

Ankit-Micron, 2021-09-13:
### What were you trying to do?

I was trying to use react-native-vision-camera in my project

### What happened instead?

* What went wrong:
A problem occurred configuring root project 'rn_starter'.        
> Could not resolve all artifacts for configuration ':classpath'.
   > Could not find org.jetbrains.kotlin:kotlin-gradle-plugin:1.1.16.
       - https://dl.google.com/dl/android/maven2/org/jetbrains/kotlin/kotlin-gradle-plugin/1.1.16/kotlin-gradle-plugin-1.1.16.pom
       - https://jcenter.bintray.com/org/jetbrains/kotlin/kotlin-gradle-plugin/1.1.16/kotlin-gradle-plugin-1.1.16.pom    
     Required by:
         project :

### Device

Android and iOS

### VisionCamera Version

2.7.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-09-13:
Can you share the full logs? There is no error included in what you shared, only a warning. Also compare your project with the `example/` project here to find out if there are some version mismatches.

Do you have `jcenter()` or `mavenCentral()` added in your `build.gradle`?

Ankit-Micron, 2021-09-14:
Thanks for the prompt response 

Yes I have jcenter() in build.gradle 

This is the full error:

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring root project 'rn_starter'.
> Could not resolve all artifacts for configuration ':classpath'.
   > Could not find org.jetbrains.kotlin:kotlin-gradle-plugin:1.1.16.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/org/jetbrains/kotlin/kotlin-gradle-plugin/1.1.16/kotlin-gradle-plugin-1.1.16.pom
       - https://jcenter.bintray.com/org/jetbrains/kotlin/kotlin-gradle-plugin/1.1.16/kotlin-gradle-plugin-1.1.16.pom
     Required by:
         project :

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 24s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup. Run CLI with --verbose flag for more details.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8088

PR #694 (2022-01-02): fix: Fix Android build on React Native 0.65 and older

## What

## Changes

## Tested on

## Related issues

* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/654
* Fixes https://github.com/mrousavy/react-native-vision-camera/issues/588
* Fixes #546 
* Fixes #513 
* Fixes #422 
* Fixes #390 
* Fixes #388 
* Fixes #369 
* Fixes #365 
* Fixes #341 
* Fixes #338 
* Fixes #269 
* Overrides #370 
* Overrides #676 
* Resolves https://github.com/mrousavy/react-native-vision-camera/issues/663
closed 2022-01-02 by mrousavy