margelo/react-native-vision-camera #431: Can I run multiple Frame Processor plugins?

ldstein, 2021-09-14:
### Question

Hi,

Hi created two test plugins, ([Plugin1](https://github.com/ldstein/MeshCameraTest/blob/master/android/app/src/main/java/com/meshcameratest/frameprocessors/ExamplePlugin1.java), [Plugin2](https://github.com/ldstein/MeshCameraTest/blob/master/android/app/src/main/java/com/meshcameratest/frameprocessors/ExamplePlugin2.java)) based based on the [ExamplePlugin](https://github.com/mrousavy/react-native-vision-camera/blob/main/example/src/frame-processors/ExamplePlugin.ts).

They have been added a to [Package](https://github.com/ldstein/MeshCameraTest/blob/master/android/app/src/main/java/com/meshcameratest/frameprocessors/FrameProcessorPackage.java) and added to the [MainApplication](https://github.com/ldstein/MeshCameraTest/blob/master/android/app/src/main/java/com/meshcameratest/MainApplication.java) as per the examples provided.

In my App I call both of them [here](https://github.com/ldstein/MeshCameraTest/blob/8104b30e2bf3cac3a8677b7ad94c3361d4d31cfe/App.js#L45-L58), which causes the app to consistently crash with the following error:

```
2021-09-14 21:17:03.138 12726-12883/com.meshcameratest A/.meshcamerates: java_vm_ext.cc:577] JNI DETECTED ERROR IN APPLICATION: can't call java.lang.Object com.meshcameratest.frameprocessors.ExamplePlugin1.callback(androidx.camera.core.ImageProxy, java.lang.Object[]) on instance of com.meshcameratest.frameprocessors.ExamplePlugin2
    java_vm_ext.cc:577]     in call to CallObjectMethodV
    java_vm_ext.cc:577]     from void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy)
```

The crash will always occur on the second call regardless of order or plugin being called.

It's very possible I'm missing some fundamental understanding of what you can and cannot do within a worklet, hence this is phrased as a question rather than a bug.

Any guidance appreciated. 

Example repo available at https://github.com/ldstein/MeshCameraTest

### What I tried

- Register plugins directly in onCreate
- Moved packages around
- Clear cache in metro, gradle 
- Ensure react-reanimated plugin + setup is complete

### VisionCamera Version

2.7.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-09-14:
Hi! Thanks for the _very detailed_ bug report and clear reproduceable example! 🖤 

You should be able to call both plugins, yes, so this is a bug. Does this work on iOS?

mrousavy, 2021-09-14:
Can you share the full adb (or Android Studio) logs? At the start of the app there should be two logs saying "Installing Frame Processor Plugin ..."

ldstein, 2021-09-14:
No problem. Really excited by what your lib solves.

I added the adb logcat [here](https://github.com/ldstein/MeshCameraTest/blob/master/logs/crash.log). The "Installing Frame Processor Plugin" lines start  [here](https://github.com/ldstein/MeshCameraTest/blob/5013c323c6ef45a477f1802f919713c9b281b4fc/logs/crash.log#L45-L50).

rodgomesc, 2021-09-14:
ohh no, this is crucial for a project that i'm involved :/

mrousavy, 2021-09-15:
@rodgomesc can you test iOS?

mrousavy, 2021-09-15:
You guys wanna see something really weird? I tried to reproduce this here in the `example/` app and called 2 separate frame processor plugins like this:

```ts
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const values = examplePlugin(frame);
    console.log(`Return Values: ${JSON.stringify(values)}`);
    const another = anotherPlugin(frame);
    console.log(`Another plugin: ${another}`);
  }, []);
```

and it works. No crash. [Here's the branch](https://github.com/mrousavy/react-native-vision-camera/tree/fix/two-plugins)

ldstein, 2021-09-15:
Interesting. Crashing on my end if I enable two plugins but okay when running a single plugin.

I noticed it's crashing if I close the App then relaunch. Seems to be ok if the Camera/Mic permissions screen is displayed first.

Could it possibly be related to me building in Windows? 

[react-native-vision-camera-two-plugins-crash.log](https://github.com/mrousavy/react-native-vision-camera/files/7169665/react-native-vision-camera-two-plugins-crash.log)

mrousavy, 2021-09-15:
Hmm, did you clean your metro cache before launching the app? `yarn start --reset-cache`

ldstein, 2021-09-15:
Tried a few things this morning:

- Clean install of your two plugins branch
- Build from Android Studio and `npm run android`
- `yarn start --reset-cache` on every attempt

I'm observing two crashes: 

1. Occurs when second plugin executes (consistent)

```
 -------------------------------------------------------
 react-native-vision-camera-two-plugins-crash2.log:
 react-native-vision-camera-two-plugins-crash3.log:
 ------------------------------------------------------- 
 LOG  Running "CameraPage" with {"initialProps":{"componentId":"Component1"},"rootTag":1}
 LOG  re-rendering camera page without active camera
 LOG  Re-rendering camera page with active camera. Device: "back (0)" (4000x2000 @ 60fps)
 LOG  Camera initialized!
 LOG  Re-rendering camera page with active camera. Device: "back (0)" (4000x2000 @ 60fps)
 LOG  Return Values: {"example_str":"Test","example_array":["Hello!",true,17.38],"example_double":5.3,"example_bool":true}
 BUNDLE  ./index.js
```

2. App loads to camera view, Permission was granted in previous use.
[react-native-vision-camera-two-plugins-crash3.log](https://github.com/mrousavy/react-native-vision-camera/files/7173817/react-native-vision-camera-two-plugins-crash3.log)
[react-native-vision-camera-two-plugins-crash4.log](https://github.com/mrousavy/react-native-vision-camera/files/7173818/react-native-vision-camera-two-plugins-crash4.log)
[react-native-vision-camera-two-plugins-crash2.log](https://github.com/mrousavy/react-native-vision-camera/files/7173819/react-native-vision-camera-two-plugins-crash2.log)
 (Intermittent):
```
 -------------------------------------------------------
 react-native-vision-camera-two-plugins-crash4.log:
 -------------------------------------------------------
 LOG  Running "CameraPage" with {"initialProps":{"componentId":"Component1"},"rootTag":1}
 LOG  re-rendering camera page without active camera
 LOG  Re-rendering camera page with active camera. Device: "back (0)" (4000x2000 @ 60fps)
```

I'm going to pull my old Macbook Pro out of storage, see if OS makes a difference.

mrousavy, 2021-09-16:
I'm using Node v14.17.4. This might also be an NDK/JNI/[fbjni](https://github.com/facebookincubator/fbjni) bug....

ldstein, 2021-09-17:
Small update. I noticed disabling hermes greatly improves the stability of FrameProcessors.

I have a simple QR Scanner App based on Google Vision processing 2K images at 30FPS for over an hour. 

If I enable Hermes, it crashes out with JNI error.

Still working on procuring a Macbook for further testing. It's currently been hijacked for remote learning by my daughter.

mrousavy, 2021-09-17:
> Small update. I noticed disabling hermes greatly improves the stability of FrameProcessors.

Huh, then it's a Hermes bug? Maybe the Hermes garbage collector (Hades) does not run on the same Thread? I'd have to coordinate that with the guys from the Hermes team, are you [still on hermes-engine 8.1.0](https://github.com/ldstein/MeshCameraTest/blob/5013c323c6ef45a477f1802f919713c9b281b4fc/package-lock.json#L7377)?

> I have a simple QR Scanner App based on Google Vision processing 2K images at 30FPS for over an hour.

Wow! Awesome, I really did something cool with the Frame Processor API 😄 

> Unfortunately it doesn't solve the multi-FrameProcessor bug. That still occurs regardless of hemes being used or not.

That's so weird. I wish I could reproduce it somehow... Did you try to run it on other Android phones maybe? What Android Studio, what Android SDK, what JDK and what NDK versions are you using? All latest?

> It's been hijacked for remote learning by my daughter

😄 

> I suspect the issue lies somewhere within my development environment.

 I would normally say I doubt that, but we'll see.

ldstein, 2021-09-18:
Small update:

- Brand new install of Android Studio and Node on the trusty 2011 Macbook Pro. No other development tools on this machine
- Cloned https://github.com/mrousavy/react-native-vision-camera/tree/fix/two-plugins
- `npm install` then `cd example` then `npm install`
- `npm run android`

No other changes.

Compiles and installs on the phone ok, grant permission, camera fades in and crashes on second frameprocessor call  with identical error as on Windows machine.

Tested on both a Pixel 3 and Pixel 2 with similar result. Going to find myself some non-Pixel phones to test.

Also going to mess around with the Android SDKs / ndk / Java versions as per your suggestion.

alexstanbury, 2021-09-21:
If it's any help, I did the above steps using a Samsung Galaxy Tab A7 and had a similar crash.

[Error Logcat here](https://gist.github.com/alexstanbury/7f5d06e1084e15af24bd6f774cb0cc2f)

icc-guerrero, 2021-09-22:
I tried with 2.8.0 and Im also suffering the same JNI crash, @mrousavy. If I enable a second plugin, it crashes. I checked @alexstanbury logs and seems the same problem. This log entry could might help you identifying the issue:

JNI DETECTED ERROR IN APPLICATION: can't call java.lang.Object com.visioncameraocr.VisionCameraOcrPlugin.callback(androidx.camera.core.ImageProxy, java.lang.Object[]) on instance of com.visioncamerafacedetector.VisionCameraFaceDetectorPlugin

Why the callback from one plugin is been called from the other one?

mrousavy, 2021-09-22:
> Why the callback from one plugin is been called from the other one?

Yeah I also don't get that. The past few days have been really busy for me so I couldn't investigate this further, I hope I can find some free time to do that soon.

alexstanbury, 2021-09-23:
Running two frame processors appears to work fine on iOS, tested on an iPhone 7 device running iOS 13.6.1. I had to downgrade the reanimated version to 2.2.2 (from the alpha used in the examples) as it seems the reanimated pods reference react-native-screens for some reason which gave a build error. 

So it looks like this is purely an Android issue. I tried separating the two test plugins into their own Android packages to see if that made a difference but it didn't.

mrousavy, 2021-09-23:
@alexstanbury in the logs you shared I can see the following:

```
2021-09-21 17:03:24.669 16150-19603/com.mrousavy.camera.example I/VisionCamera: Calling Plugin "class com.mrousavy.camera.example.ExampleFrameProcessorPlugin"...
2021-09-21 17:03:24.669 16150-19603/com.mrousavy.camera.example D/ExamplePlugin: 640 x 480 Image with format #35. Logging 6 parameters:
2021-09-21 17:03:24.669 16150-19603/com.mrousavy.camera.example D/ExamplePlugin:   -> hello! (java.lang.String)
2021-09-21 17:03:24.669 16150-19603/com.mrousavy.camera.example D/ExamplePlugin:   -> parameter2 (java.lang.String)
2021-09-21 17:03:24.669 16150-19603/com.mrousavy.camera.example D/ExamplePlugin:   -> true (java.lang.Boolean)
2021-09-21 17:03:24.670 16150-19603/com.mrousavy.camera.example D/ExamplePlugin:   -> 42.0 (java.lang.Double)
2021-09-21 17:03:24.671 16150-19603/com.mrousavy.camera.example D/ExamplePlugin:   -> { NativeMap: {"test":0,"second":"test"} } (com.facebook.react.bridge.ReadableNativeMap)
2021-09-21 17:03:24.671 16150-19603/com.mrousavy.camera.example D/ExamplePlugin:   -> ["another test",5] (com.facebook.react.bridge.ReadableNativeArray)
2021-09-21 17:03:24.674 16150-19603/com.mrousavy.camera.example I/VisionCamera: Calling Plugin "class com.mrousavy.camera.example.AnotherPlugin"...
2021-09-21 17:03:24.674 16150-19603/com.mrousavy.camera.example A/.camera.exampl: java_vm_ext.cc:578] JNI DETECTED ERROR IN APPLICATION: can't call java.lang.Object com.mrousavy.camera.example.ExampleFrameProcessorPlugin.callback(androidx.camera.core.ImageProxy, java.lang.Object[]) on instance of com.mrousavy.camera.example.AnotherPlugin
    java_vm_ext.cc:578]     in call to CallObjectMethodV
    java_vm_ext.cc:578]     from void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy)
2021-09-21 17:03:24.674 16150-19461/com.mrousavy.camera.example I/ReactNativeJS: Return Values: {"example_str":"Test","example_array":["Hello!",true,17.38],"example_double":5.3,"example_bool":true}
2021-09-21 17:03:24.992 16150-19603/com.mrousavy.camera.example A/.camera.exampl: runtime.cc:677] Runtime aborting...
```

So at first it's calling

```
2021-09-21 17:03:24.669 16150-19603/com.mrousavy.camera.example I/VisionCamera: Calling Plugin "class com.mrousavy.camera.example.ExampleFrameProcessorPlugin"...
```

which works fine, then it's calling

```
2021-09-21 17:03:24.674 16150-19603/com.mrousavy.camera.example I/VisionCamera: Calling Plugin "class com.mrousavy.camera.example.AnotherPlugin"...
```

which crashes with the error 

```
can't call java.lang.Object com.mrousavy.camera.example.ExampleFrameProcessorPlugin.callback(androidx.camera.core.ImageProxy, java.lang.Object[]) on instance of com.mrousavy.camera.example.AnotherPlugin
```

wtf?

mrousavy, 2021-09-23:
Can you try again with the latest change in https://github.com/mrousavy/react-native-vision-camera/pull/435?

icc-guerrero, 2021-09-23:
#435 just tested. Same issue :(
2021-09-23 12:20:28.264 12948-13338/org.unfccc.cop26idcheck A/cc.cop26idchec: runtime.cc:638] JNI DETECTED ERROR IN APPLICATION: can't call java.lang.Object com.visioncameraocr.VisionCameraOcrPlugin.callback(androidx.camera.core.ImageProxy, java.lang.Object[]) on instance of com.visioncamerafacedetector.VisionCameraFaceDetectorPlugin

alexstanbury, 2021-09-23:
> Can you try again with the latest change in #435?

I couldn't get the example to build, I get the following build error:
```
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'.
> Build command failed.
  Error while executing process /Users/alexstanbury/Library/Android/sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C /Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a VisionCamera}
  ninja: Entering directory `/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPlugin.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPlugin.cpp.o 
  /Users/alexstanbury/Library/Android/sdk/ndk/20.1.5948944/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=armv7-none-linux-androideabi21 --gcc-toolchain=/Users/alexstanbury/Library/Android/sdk/ndk/20.1.5948944/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/alexstanbury/Library/Android/sdk/ndk/20.1.5948944/toolchains/llvm/prebuilt/darwin-x86_64/sysroot  -DVisionCamera_EXPORTS -I../../../../build/fbjni-0.2.2-headers.jar -I../../../../build/third-party-ndk/boost -I../../../../build/third-party-ndk/double-conversion -I../../../../build/third-party-ndk/folly -I../../../../build/third-party-ndk/glog -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native/React -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native/React/Base -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native/ReactAndroid/src/main/jni -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native/ReactAndroid/src/main/java/com/facebook/react/turbomodule/core/jni -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native/ReactCommon -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native/ReactCommon/callinvoker -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native/ReactCommon/jsi -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/hermes-engine/android/include -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native-reanimated/Common/cpp/headers/Tools -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native-reanimated/Common/cpp/headers/SpecTools -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native-reanimated/Common/cpp/headers/SharedItems -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native-reanimated/Common/cpp/headers/Registries -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native-reanimated/Common/cpp/headers/LayoutAnimations -I/Users/alexstanbury/Desktop/pluginfix/react-native-vision-camera/example/android/../node_modules/react-native-reanimated/Common/cpp/hidden_headers -I../../../../src/main/cpp -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -O0 -fno-limit-debug-info  -fPIC   -std=gnu++14 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPlugin.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPlugin.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorPlugin.cpp.o -c ../../../../src/main/cpp/FrameProcessorPlugin.cpp
  ../../../../src/main/cpp/FrameProcessorPlugin.cpp:33:125: error: no member named 'c_str' in 'facebook::jni::basic_strong_ref<facebook::jni::JClass, facebook::jni::LocalReferenceAllocator>'
    __android_log_print(ANDROID_LOG_INFO, TAG, "Plugin \"%s\" (actual: \"%s\"): Calling callback...", typeName.c_str(), clzzz.c_str());
                                                                                                                        ~~~~~ ^
  ../../../../src/main/cpp/FrameProcessorPlugin.cpp:33:41: error: use of undeclared identifier 'TAG'
    __android_log_print(ANDROID_LOG_INFO, TAG, "Plugin \"%s\" (actual: \"%s\"): Calling callback...", typeName.c_str(), clzzz.c_str());
                                          ^
  2 errors generated.
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/CameraView.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxyHostObject.cpp.o
  [6/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  [7/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  [8/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  [9/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o
  ninja: build stopped: subcommand failed.
  ```

So I commented the offending line and it built ok, but unfortunately I'm still getting the same issue.  [Logs attached](
https://gist.github.com/alexstanbury/c8dffbff2b84373ad82e2ce298c4f95c)

ldstein, 2021-09-23:
Also a compiler crash for me:

https://gist.github.com/ldstein/00feb603c851e8e3518aae05c322a286

mrousavy, 2021-09-23:
Whoops, can you replace `TAG` with something like `"testtest"` and try again? I need that log.

ldstein, 2021-09-23:
Different error this time (C++ is completely foreign to me)

[compile-error.log](https://github.com/mrousavy/react-native-vision-camera/files/7217415/compile-error.log)

mrousavy, 2021-09-23:
ah shit sorry again! doing this on mobile 😅 try again now

alexstanbury, 2021-09-23:
> ah shit sorry again! doing this on mobile 😅 try again now

[Here you go](https://gist.github.com/alexstanbury/6a65d22c396edf8804f477194f43946d)

mrousavy, 2021-09-23:
what the f

mrousavy, 2021-09-23:
gonna report abuse on that gist because that's the weirdest thing I've ever seen 
[image]

mrousavy, 2021-09-23:
Trying a really weird thing - does the latest commit in the PR fix the issue for you guys? cc @alexstanbury

alexstanbury, 2021-09-23:
> Trying a really weird thing - does the latest commit in the PR fix the issue for you guys? cc @alexstanbury

Yep, that works:

```2021-09-23 14:21:59.789 23302-23302/? I/.camera.exampl: Late-enabling -Xcheck:jni
2021-09-23 14:21:59.789 23302-23302/? I/.camera.exampl: Late-enabling -Xcheck:jni
2021-09-23 14:21:59.842 23302-23302/? E/.camera.exampl: Unknown bits set in runtime_flags: 0x8000
2021-09-23 14:21:59.843 23302-23302/? W/.camera.exampl: Unexpected CPU variant for X86 using defaults: x86
2021-09-23 14:21:59.842 23302-23302/? E/.camera.exampl: Unknown bits set in runtime_flags: 0x8000
2021-09-23 14:21:59.843 23302-23302/? W/.camera.exampl: Unexpected CPU variant for X86 using defaults: x86
2021-09-23 14:21:59.789 23302-23302/? I/.camera.exampl: Late-enabling -Xcheck:jni
2021-09-23 14:21:59.842 23302-23302/? E/.camera.exampl: Unknown bits set in runtime_flags: 0x8000
2021-09-23 14:21:59.843 23302-23302/? W/.camera.exampl: Unexpected CPU variant for X86 using defaults: x86
2021-09-23 14:22:00.654 23302-23302/com.mrousavy.camera.example D/SoLoader: init start
2021-09-23 14:22:00.655 23302-23302/com.mrousavy.camera.example D/SoLoader: adding system library source: /vendor/lib
2021-09-23 14:22:00.655 23302-23302/com.mrousavy.camera.example D/SoLoader: adding system library source: /system/lib
2021-09-23 14:22:00.656 23302-23302/com.mrousavy.camera.example D/SoLoader: adding application source: com.facebook.soloader.DirectorySoSource[root = /data/app/com.mrousavy.camera.example-WaOyZZrW7R7-A-kZSKcMCg==/lib/x86 flags = 0]
2021-09-23 14:22:00.658 23302-23302/com.mrousavy.camera.example D/SoLoader: adding backup source from : com.facebook.soloader.ApkSoSource[root = /data/data/com.mrousavy.camera.example/lib-main flags = 1]
2021-09-23 14:22:00.658 23302-23302/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /system/lib flags = 2]
2021-09-23 14:22:00.658 23302-23302/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /vendor/lib flags = 2]
2021-09-23 14:22:00.658 23302-23302/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.DirectorySoSource[root = /data/app/com.mrousavy.camera.example-WaOyZZrW7R7-A-kZSKcMCg==/lib/x86 flags = 0]
2021-09-23 14:22:00.658 23302-23302/com.mrousavy.camera.example D/SoLoader: Preparing SO source: com.facebook.soloader.ApkSoSource[root = /data/data/com.mrousavy.camera.example/lib-main flags = 1]
2021-09-23 14:22:00.659 23302-23302/com.mrousavy.camera.example V/fb-UnpackingSoSource: locked dso store /data/user/0/com.mrousavy.camera.example/lib-main
2021-09-23 14:22:00.660 23302-23302/com.mrousavy.camera.example V/fb-UnpackingSoSource: deps mismatch on deps store: regenerating
2021-09-23 14:22:00.660 23302-23302/com.mrousavy.camera.example V/fb-UnpackingSoSource: so store dirty: regenerating
2021-09-23 14:22:00.723 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libVisionCamera.so: deferring to libdir
2021-09-23 14:22:00.723 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libbetter.so: deferring to libdir
2021-09-23 14:22:00.724 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libc++_shared.so: deferring to libdir
2021-09-23 14:22:00.724 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libfabricjni.so: deferring to libdir
2021-09-23 14:22:00.724 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libfb.so: deferring to libdir
2021-09-23 14:22:00.724 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libfbjni.so: deferring to libdir
2021-09-23 14:22:00.724 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libfolly_futures.so: deferring to libdir
2021-09-23 14:22:00.725 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libfolly_json.so: deferring to libdir
2021-09-23 14:22:00.725 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libglog.so: deferring to libdir
2021-09-23 14:22:00.725 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libglog_init.so: deferring to libdir
2021-09-23 14:22:00.726 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libhermes-executor-common-debug.so: deferring to libdir
2021-09-23 14:22:00.726 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libhermes-executor-common-release.so: deferring to libdir
2021-09-23 14:22:00.726 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libhermes-executor-debug.so: deferring to libdir
2021-09-23 14:22:00.726 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libhermes-executor-release.so: deferring to libdir
2021-09-23 14:22:00.726 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libhermes-inspector.so: deferring to libdir
2021-09-23 14:22:00.727 23302-23302/com.mrousavy.camera.example D/ApkSoSource: not allowing consideration of lib/x86/libhermes.so: deferring to libdir
[... cut ...]
2021-09-23 14:22:10.785 23302-23352/com.mrousavy.camera.example I/ReactNativeJS: Return Values: {"example_str":"Test","example_array":["Hello!",true,17.38],"example_double":5.3,"example_bool":true}
2021-09-23 14:22:10.786 23302-23379/com.mrousavy.camera.example I/VisionCamera: Calling Plugin "class com.mrousavy.camera.example.AnotherPlugin"...
2021-09-23 14:22:11.515 23302-23379/com.mrousavy.camera.example D/AnotherPlugin: AnotherPlugin called with 3 parameters.
2021-09-23 14:22:11.517 23302-23352/com.mrousavy.camera.example I/ReactNativeJS: Another plugin: 5```

mrousavy, 2021-09-23:
lol, fresh

will clean up the PR and merge when I get home

alexstanbury, 2021-09-23:
Brilliant, thanks for sorting it.

mrousavy, 2021-09-23:
thanks for testing + reporting 🙏

icc-guerrero, 2021-09-23:
@mrousavy: I can confirm that is also works for me.  I mentioned a few hours ago that was not working, but was clearly my fault: C++ libraries were not properly cleaned up. After deleting cached ones it worked as expected. Thanks!

ldstein, 2021-09-23:
Nice one, all good on this end too.

closed 2021-09-24 by mrousavy

mrousavy, 2021-09-24:
I've fixed this in [**2.8.1**](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.8.1) 🎉🚀

mhorowitz, 2021-09-28:
> Maybe the Hermes garbage collector (Hades) does not run on the same Thread?

That's correct.  From jsi.h: "The [HostObject] C++ object's dtor will be called when the GC finalizes this object.  (This may be as late as when the Runtime is shut down.) You have no control over which thread it is called on."  Before Hades, this would always happen synchronously, but one of the ways Hades reduces JS stalls is by sometimes calling dtors on a separate thread.  If you need to delete an fbjni global_ref, then you need to arrange to queue that work on a Java-registered thread (preferred), or use ThreadScope::WithClassLoader to temporarily register the current thread.

mrousavy, 2021-09-28:
Thanks for elaborating @mhorowitz! I actually use WithClassLoader in the dtor now, but how does "you need to arrange to queue that work on a Java-registered thread (preferred)" work? Register the Hades GC Thread with JNI?

mhorowitz, 2021-09-28:
You can't usefully register the Hades GC thread, because there's no way to safely unregister it.

I mean you would have some queue which you could add to in native from any thread, and read from on a java thread, which would destroy the references added to the queue.  This is mostly independent from JNI.

mrousavy, 2021-09-28:
I see, thanks for clarifying! I'm going to stick with the fbjni Class Loader solution for now.

PR #435 (2021-09-24): fix: Fix calling multiple Plugins in a single Frame Processor

## What

Calling multiple plugins in a single frame processor crashes with the following error:

```
2021-09-14 23:12:50.060 30880-31018/com.meshcameratest A/.meshcamerates: java_vm_ext.cc:577] JNI DETECTED ERROR IN APPLICATION: can't call java.lang.Object com.meshcameratest.frameprocessors.ExamplePlugin1.callback(androidx.camera.core.ImageProxy, java.lang.Object[]) on instance of com.meshcameratest.frameprocessors.ExamplePlugin2
```

This PR fixes this so you can call unlimited amount of plugins in a Frame Processor

## Changes

# [... cut]