margelo/react-native-vision-camera #434: 🐛 Preview does not and hangs on the capture process

developerantoniosousa, 2021-09-15:
### What were you trying to do?

I ran the example app, granted the camera and microphone permissions, then tried to capture a photo. The preview didn't load.  By pressing the capture button the process kept hanging.

### What happened instead?

The app didn't crash. It kept hanging on the capture process and the preview does not appear since the app opening - it kept a black screen all the time.
[image]

### Device

Samsung A10, Android 11

### VisionCamera Version

v2.7.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-09-15:
Hmm, does it work if you remove the `frameProcessor`?

developerantoniosousa, 2021-09-18:
Actually, frameProcessor is not supported by what I could see here.

I've noticed if fps is equal to 60 (which is not supported by my phone) the preview doesn't show up or update. My phone also doesn't support `supportsLowLightBoost` and `supportsParallelVideoProcessing`.

I got it working properly only passing 30 fps (by prop to Camera).

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy