margelo/react-native-vision-camera #437: ✨ Framebuffer emulation support

xRiot, 2021-09-15:
### What feature or enhancement are you suggesting?

The ability to emulate a framebuffer for a virtual device in xCode's Simulator and Android Studio's Emulator would make it a lot easier to test if the app behaves correctly in a CICD environment. For instance, this could be achieved by adding support for debug mode with a prerecorded video or image file.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

None

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-09-15:
Hi! Thanks for the feature request, that's indeed something I'm also missing from VisionCamera! 

If you want to take a look at this and implement this yourself, I'd gladly give any pointers and help you out! 💪 

Until then, it stays a feature request until I have some free time to take a look at this (hopefully soon 😉)

rodgomesc, 2021-09-15:
it would be fantastic if we could emulate a virtual scene in  visio camera similar to the one we have in the android emulator, today it is impossible to test camera on ios without physical hardware, and this kind of functionality would be a differential that no other camera lib for react native has

xeroxoid, 2021-11-12:
My 2 cents: https://github.com/teslamotors/react-native-camera-kit somehow does it

AdamSheaHewett, 2022-06-02:
Please make this happen, it would make development so much easier

rgomesrn, 2022-06-02:
yep, double thumbs up for this feature

fervailanti, 2022-07-11:
+1 to the request! 👍🏼

devartwa, 2022-11-14:
+1, would be nice.

salorsino, 2022-12-27:
+2 would be really nice to have this, not using this library because of the lack of simulator support

mrousavy, 2022-12-28:
Please stop spamming. This is a really nice feature request / nice-to-have, but I currently have other priorities ([VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376) and my company [Margelo](https://margelo.io)), if anyone wants to pay for the development of this feature let me know.

Otherwise I'll leave this open until someone submits a PR to implement this, or I have some free time sometime in the future.

mrousavy, 2023-09-30:
Update; VisionCamera will never support simulating the Camera on the iOS simulator as it is insanely complex to do that in VisionCamera since it is built upon a Devices API. More info here: https://github.com/mrousavy/react-native-vision-camera/pull/1848#issuecomment-1735162249

closed 2023-09-30 by mrousavy