margelo/react-native-vision-camera #438: 🐛 Samsung s20 + Unknown Camera Error

maguirekrist, 2021-09-17:
### What were you trying to do?

Calling both useCameraDevices() and Camera.getAvailableCameraDevices().

### Reproduceable Code

```tsx
`
try {
                cameraDevices = await Camera.getAvailableCameraDevices();
                
            } catch (err) {
                console.log("Camera Error!!");
                console.log(err.code);
                console.log(err);
            }
`
```

### What happened instead?

Well, if I didn't wrap the function in a try, I get a unhandled promise rejection of "Unknown error". The output from the previous code snippet is....
`
unknown/unknown
[unknown/unknown: [unknown/unknown] An unknown camera error occured.]
`  

### Device

Samsung Galaxy s20 Plus

### VisionCamera Version

2.7.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-09-17:
😐

mrousavy, 2021-09-17:
1. You did not provide a reproduceable sample. You pasted a snippet which crashes for you, but this is not reproduceable. Obviously that code works for me if it's in the `example/` app. I need a repo/code that I can run which also "crashes" for me.
2. You intentionally skipped the "**Relevant log output**" section and only attached a "unknown/unknown" error message, which is as meaningless as it can get. I need the output from `adb` because your error message says literally nothing:

[image]

maguirekrist, 2021-09-17:
Firstly, it is reproducible with just that snippet when ran on a Samsung Galaxy S20 + (honest to god, you only need to request a camera device with your function to precipitate an error, but only on specific Android devices); your camera solution is great on the Android Emulator and all IOS Devices. Unfortunately, I cannot present you any ADB output because I no longer have the device with me, I had to travel an hour to test the application on a Android device as everyone that I know in my area uses IOS, including myself. I'll try to go back by next week and give it a run, but until then, I may have to rewrite my application using another react-native camera module in hopes of getting this to work on "supported" android devices (which I feel will also be inauspicious on Samsung devices, considering your module has received great acclaim). I had to intentionally skip relevant log output, logcat seems all gibberish when I ran it (it just kept spitting on information, I did not know where I should copy considering how much text info there was), and your own error catching solution presented in your documentation provided me with a "unknown/unknown" code.

P.S: I understand it's frustrating to have someone tell you "unknown error" on your repo, it isn't much help, but this bug is so elusive because it seems to be happening only on Samsung devices as stated on another open ticket by another user on this repo.

mrousavy, 2021-09-17:
Well it's not just frustrating to have no information, it's just that I can't help you if I don't have an error message. And you get this library for free, there's not much I can do without log output.

If you share the logcat/adb logs (just share all of them.) I can take a look at this :)

maguirekrist, 2021-09-17:
I'll give you the log output when I get the chance, until then, maybe my error is related to what another user posted also using a Samsung device and getting "unknown error" as posted here: https://github.com/mrousavy/react-native-vision-camera/issues/238

xRiot, 2021-09-23:
Same issue here with my Samsung Galaxy S20+ / Android 11 @mrousavy
Here's the Logcat from react-native-vision-camera/example after granting microphone + camera permissions and restarting the app:
https://gist.githubusercontent.com/xRiot/abaeab52e43c9251209f19eb5cb1c565/raw/84aa0b7ea79c159b6b9e3f5ec7aee2be62dc57ae/react-native-vision-camera-example

mrousavy, 2021-09-23:
@xRiot thank you for the Logcat logs! Because of the logs I was able to fix the issue: https://github.com/mrousavy/react-native-vision-camera/pull/449.

I will create a patch release shortly.

closed 2021-09-23 by mrousavy

mrousavy, 2021-09-24:
I've fixed this in [**2.8.1**](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.8.1) 🎉🚀

PR #449 (2021-09-23): fix: Fix Android Stabilization Modes

## What

On some Samsung devices, getting the Digital Lens Stabilization Modes returns null, even though the Android Docs say it cannot be null:

[image]

So the docs are lying.

## Changes

* Fixes Stabilization Modes being `null` on some Samsung devices (even though the Android docs say it cannot be null!)
* Always returns at least `off` for stabilization modes
* Returns `cinematic` for optical stabilization mode

## Tested on

## Related issues

* Fixes #438
* Fixes #238