margelo/react-native-vision-camera #448: How I can record video with audio while another video is playback on IOS?❓

HorbachAndrii, 2021-09-23:
### Question

How I can record video with audio in case when I'm running record video and then run play video?

### What I tried

I can record video with audio while another video is playback on IOS but in case I'm running play video before start record video.

But I can't record video with audio in case when I'm running record video and then run play video.

### VisionCamera Version

2.8.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-09-23:
Hey! Sorry I don't understand the question. Recording a Video while playing any kind of audio in the background (Spotify, or even a Video) works for me in the `example/` app.

HorbachAndrii, 2021-09-23:
Sorry for my English(.

I want to record a video while playing any kind of video in the foreground.
For example. 
I have 2 components on one screen:
 1 - it's <VLCPlayer/>
 2 - it's <Camera/>
 
What I do:
1. start recording video(from component <Camera/>).
2. start play video(from component <VLCPlayer/>).
3. stop record video.
In this case, the microphone will turn off when I start to play video(from component <VLCPlayer/>).

mrousavy, 2021-09-23:
This might be an issue with the VLCPlayer component - if it does not set the Audio Session correctly it will force the Camera to loose audio focus. Here's how I set up the Audio Session:

https://github.com/mrousavy/react-native-vision-camera/blob/e621422389d54c315455051affbf7f797fe8d76a/ios/CameraView%2BAVAudioSession.swift#L90-L94

Ideally, VLCPlayer should not touch the audio session at all (or set it up the same way I do, important parts are `playAndRecord` and option `.mixWithOthers`)

HorbachAndrii, 2021-09-23:
Thanks for your support.

I found line where mentioned `AVAudioSession` in the VLCPlayer library. 

Could you look at this line and say your opinion?

https://github.com/razorRun/react-native-vlc-media-player/blob/4a83c5116ff86eec9208cb2423b4ee29b356d2f2/ios/RCTVLCPlayer/RCTVLCPlayer.m#L110

mrousavy, 2021-09-24:
yea, if you remove that line from the lib it should work

HorbachAndrii, 2021-09-24:
thank you, I will try to remove [this](https://github.com/razorRun/react-native-vlc-media-player/blob/4a83c5116ff86eec9208cb2423b4ee29b356d2f2/ios/RCTVLCPlayer/RCTVLCPlayer.m#L110) line.

p.c.
By the way... 
When I recording video and using WebView for playback video. In this case, video recording video works fine.

closed 2021-10-05 by mrousavy