margelo/react-native-vision-camera #452: Recorded video is stretched🐛

HorbachAndrii, 2021-09-23:
### What were you trying to do?

Record video.

### Reproduceable Code

```tsx
<Camera
 ref={cameraRef}
 style={styles.recordVideo}
 device={devices.front}
 video={true}
 audio={true}
 isActive={isActiveRecordVideo}
 preset="iframe-1280x720"
/>

or Full code

import React, {useLayoutEffect, useRef, useState} from 'react';
import {
  Platform,
  Pressable,
  SafeAreaView,
  ScrollView,
  StatusBar,
  StyleSheet,
  Text,
  View,
} from 'react-native';
import SplashScreen from 'react-native-splash-screen';
import {VLCPlayer} from 'react-native-vlc-media-player';
import {Camera, useCameraDevices} from 'react-native-vision-camera';
import {RNCamera} from 'react-native-camera';

const AppVisionCamera = () => {
  const devices = useCameraDevices();

  const [isRecordingVide, setIsRecordingVide] = useState(false);
  const [recordedVideo, setRecordedVideo] = useState(null);

  const cameraRef = useRef();

  useLayoutEffect(() => {
    const getPermissions = async () => {
      const cameraPermission = await Camera.getCameraPermissionStatus();
      if (cameraPermission === 'not-determined') {
        const newCameraPermission = await Camera.requestCameraPermission();
        console.log('newCameraPermission', newCameraPermission);
      }
      const microphonePermission = await Camera.getMicrophonePermissionStatus();
      console.log('microphonePermission', microphonePermission);
      if (microphonePermission === 'not-determined') {
        const newMicrophonePermission =
          await Camera.requestMicrophonePermission();
        console.log('newMicrophonePermission', newMicrophonePermission);
      }
    };
    getPermissions();
    SplashScreen.hide();
  }, []);

  const onStartRecording = async () => {
    setRecordedVideo(null);
    setIsRecordingVide(true);
    console.log('cameraRef', cameraRef.current);
    cameraRef.current.startRecording({
      fileType: 'mp4',
      onRecordingFinished: async (video) => {
        console.log('onRecordingFinished', video);
        setRecordedVideo(video.path);
      },
      onRecordingError: (error) => {
        console.error('onRecordingError', error);
      },
    });
  };

  const onStopRecording = async () => {
    await cameraRef.current.stopRecording();
    setIsRecordingVide(false);
  };

  return (
    <SafeAreaView style={StyleSheet.absoluteFill}>
      <StatusBar barStyle="dark-content"/>
      <ScrollView contentInsetAdjustmentBehavior="automatic">
        {Platform.OS === 'android' ? (
          <View style={styles.viewRecordVideo}>
            <RNCamera
              ref={cameraRef}
              style={styles.RNCamera}
              type={RNCamera.Constants.Type.front}
              flashMode={RNCamera.Constants.FlashMode.on}
              useCamera2Api={true}
              playSoundOnCapture={false}
            />
          </View>
        ) : (
          <View style={styles.viewRecordVideo}>
            {devices.front && (
              <Camera
                ref={cameraRef}
                style={styles.recordVideo}
                device={devices.front}
                video={true}
                audio={true}
                isActive={true}
                preset="high"
              />
            )}
          </View>
        )}

        <View style={styles.viewButtons}>
          {isRecordingVide ? (
            <Pressable onPress={onStopRecording}>
              <Text style={styles.buttonText}>Stop record</Text>
            </Pressable>
          ) : (
            <Pressable onPress={onStartRecording}>
              <Text style={styles.buttonText}>Start record</Text>
            </Pressable>
          )}
        </View>

        <View style={styles.viewPlayer}>
          {recordedVideo && (
            <VLCPlayer
              style={styles.player}
              resizeMode="contain"
              repeat={true}
              source={{uri: recordedVideo}}
            />
          )}
        </View>
        <View style={styles.viewButtons}>
          <Pressable
            onPress={() => setRecordedVideo(null)}>
            <Text style={styles.buttonText}>Remove video</Text>
          </Pressable>
        </View>
      </ScrollView>
    </SafeAreaView>
  );
};

const styles = StyleSheet.create({
  viewPlayer: {
    width: '100%',
    height: 220,
  },
  player: {
    flex: 1,
  },

  viewRecordVideo: {
    width: '100%',
    height: 300,
  },
  recordVideo: {
    flex: 1,
  },

  viewButtons: {
    flex: 1,
    flexDirection: 'row',
    alignItems: 'center',
    justifyContent: 'center',
    width: '100%',
    height: 50,
    backgroundColor: `black`
  },
  buttonText: {
    padding: 10,
    fontSize: 20,
    color: 'white',
    borderWidth: 1,
    borderColor: 'white',
    borderRadius: 8,
  },
});

export default AppVisionCamera;
```

### What happened instead?

When I playback recorded video with `preset="iframe-1280x720 or iframe-960x540"` I have stretched video.
You can see a bug on this [video](https://drive.google.com/file/d/1V9iGkFxD5ffzm7rKL0XGD55NeXhe2gVF/view)

### Device

iPhone 7 (IOS 15.0)

### VisionCamera Version

2.8.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

HorbachAndrii, 2021-09-23:
update:
When I added props `fps={24}` video doesn't stretched anymore

mrousavy, 2021-09-24:
Are you sure that the video is not just streched because your `<Video>` component is stretched? What `resizeMode` did you pass to the `<Video>` component?

HorbachAndrii, 2021-09-24:
yes, I'm sure. 
I did download the recorded video and watched it on my computer and the video also was be stretched.

mrousavy, 2021-09-27:
Does it work if you don't use the `preset`?

HorbachAndrii, 2021-09-28:
If I don't use preset, recording video works well. I mean recorded video will not be stretched.

dylan-westbury, 2022-05-26:
For me preset seems to work fine with mp4, however when adding videoCodec as h264 (so it can be viewed in chrome) is when the video stretches. 

Any ideas if it can be resolved? 

We need to record in mp4 not mov because of a service we use, however chrome will only show videos in h264
   
   ```const cameraOptions = {
      flash: "off",
      fileType: "mp4",
      onRecordingFinished,
      onRecordingError,
    };

    if (availableCodes.includes("h264")) {
      cameraOptions.videoCodec = "h264"; // video stretches if codec is h264
    }```

     ```<Camera
        ref={camera}
        style={StyleSheet.absoluteFill}
        device={device}
        preset="iframe-1280x720"
        orientation="portrait"
        onError={onCameraError}
        video
        audio
        isActive={isActive}
      />```

HorbachAndrii, 2022-05-26:
@dylan-westbury 

For works well

```
const newVideoPath = await new Promise((resolve, reject) => {
  cameraRef.current.startRecording({
    fileType: `mp4`,
    videoCodec: Platform.OS === `android` ? undefined : `h264`,
    onRecordingFinished: (video) => {
      resolve(video.path);
    },
    onRecordingError: (error) => {
      reject(error);
    },
  });
});
```

```
<Camera
  style={styles.camera}
  ref={cameraRef}
  device={device}
  video={true}
  audio={true}
  isActive={true}
  fps={25}
  preset="iframe-960x540"
  orientation="portrait"
  onInitialized={() => setCameraIsReady(true)}
  onError={(error) => {
    console.log(`Camera onError`, error);
  }}
/>
```

TexxUkr, 2023-05-15:
have "stretched" video on iphone 12 mini iOS 16.4.1 in case if recording uses h264 codec and camera preset is medium or low. In case if no preset video is ok.

` cameraRef.current.startRecording({
          // flash: flash,
          videoCodec: 'h264',
          fileType: 'mp4',
          onRecordingError: (error) => {
            console.error('### Recording failed!', error)
          },
          onRecordingFinished: (video) => {
            console.log(`Recording successfully finished! ${video.path}`)
            onMediaCaptured(video)
          },`

`    <CameraView
          ref={cameraRef}
          device={device}
          isActive={true}
          onInitialized={onInitialized}
          onError={(err) => {
            console.log('### error:', err)
          }}
          enableZoomGesture={false}
          photo={true}
          video={true}
          audio={true}
          orientation="portrait"
          preset="medium"
        />`

closed 2023-09-01 by mrousavy

mrousavy, 2024-01-11:
Hey!

After 8 hours of debugging, I finally found the culprit! I fixed the preview stretching issue in this PR: https://github.com/mrousavy/react-native-vision-camera/pull/2377

If you appreciate my dedication and free support, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) so I can keep providing fixes and building out new features for my open-source libraries in my free time.