margelo/react-native-vision-camera #454: RNReanimated/Scheduler.h not found🐛

yahyamokaied, 2021-09-24:
### What were you trying to do?

error while building the app (iOS)

### What happened instead?

RNReanimated/Scheduler.h not found in VisionCameraScheduler.h

### Device

Xcode

### VisionCamera Version

2.8.1

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-09-24:
* Which react-native version are you using?
* Which react-native-reanimated version are you using?

yahyamokaied, 2021-09-24:
"react-native-reanimated": "^2.2.2",
  "react": "17.0.2",
    "react-native": "0.65.1",
    "react-native-screens": "^3.7.2",

yahyamokaied, 2021-09-24:
I have tried to use same your     "react-native-reanimated": "^2.3.0-alpha.3",
but got the same issue.

mrousavy, 2021-09-27:
Can you open Xcode and hit <kbd>⌘</kbd> + <kbd>⇧</kbd> + <kbd>O</kbd> (Command + Shift + O), then enter `Scheduler` and let me know what Xcode finds? 

I assume you did run `pod install` correctly and Reanimated works inside your App, right?

adhip94, 2021-09-27:
I get this.
[image]

My react-native-reanimated version is quite low (1.13.3). Rest all dependencies are up to date.

mrousavy, 2021-09-27:
Oh yeah, VisionCamera requires react-native-reanimated 2.2.0 or higher. I'm working on making Reanimated an optional dependency here: https://github.com/mrousavy/react-native-vision-camera/pull/412 but haven't found the time to finish that yet.

adhip94, 2021-09-28:
@mrousavy I finally upgraded react-native-reanimated to v2.2.2, and now while building in Xcode I get this error
[image]

mrousavy, 2021-09-28:
Hmm, is that with Hermes enabled?

adhip94, 2021-09-28:
Yes, Hermes is enabled. I did manage to resolve the issue by adding [this patch](https://github.com/mrousavy/react-native-vision-camera/issues/390#issuecomment-912819523).

mrousavy, 2021-09-28:
Cool! Can you shoot me a PR to fix this issue once and for all, instead of patching the library yourself?

rodgomesc, 2021-10-08:
weird, i'm not facing this problem anymore after last ra-alpha-3

yahyamokaied, 2021-10-15:
<img width="1130" alt="Screenshot 2021-10-15 at 11 54 48" src="https://user-images.githubusercontent.com/50097658/137469392-f256fba8-3290-43d4-a5a9-de65a2761d5f.png"
[image]
>

coolvasanth, 2021-10-18:
@yahyamokaied were you able to resolve this issue ?

adhip94, 2021-10-18:
Try upgrading react-native-reanimated to v2.2.3 and react-native-vision-camera to v2.9.3, clear pod and Xcode caches and build again.

coolvasanth, 2021-10-18:
@adhip94 I updated the react-native-reanimated and react-native-vision-camera  to the plugin you suggested. Later deleted the Pods folder and pod.lock file from iOS folder and did pod install. Afterwards went to Xcode and cleaned the build and rebuild it. Still no luck, can you help me, where I might have made a mistake ?

yahyamokaied, 2021-10-18:
I did all setps many times but didn’t worked but **Finally I'm close to find the issue**,
 there is something missing during reanimated installing steps
 :( they have repo called playground, I cloned that repo and added this package and it worked well
  so you can investigate the difference between playground and yours to find the issue.

closed 2021-10-18 by yahyamokaied

AbdurakhmonZiyodov, 2022-11-11:
hello everyone, has anyone been able to fix this error? I also get the same error and I can't fix it

AbdurakhmonZiyodov, 2022-11-11:
I'm installing pod and it gives me this error at the end
[image]

AbdurakhmonZiyodov, 2022-11-11:
my package.json file
[image]

Please help me