margelo/react-native-vision-camera #485: iPhone 11 doesn't record video with H.264 codec 🐛

HorbachAndrii, 2021-10-04:
### What were you trying to do?

I tried to record video with H.264 codec

### Reproduceable Code

```tsx
<Camera
            ref={cameraRef}
            device={devices.front}
            video={true}
            audio={true}
            isActive={true}
            preset="iframe-960x540"
            fps={25}
            onInitialized={() => setCameraReady(true)}
          />
```

### What happened instead?

When I tried to record a video with H.264 codec, I got HEVC video codec.

### Device

iPhone 11 (IOS 15.0.1)

### VisionCamera Version

2.8.0.

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-10-04:
You can use `format={...}` to pick a format that records in H.264 codec. (I think I added those fields, if not, this is a feature request.)

HorbachAndrii, 2021-10-04:
I didn't find line about codec(
https://github.com/mrousavy/react-native-vision-camera/blob/c627febde5c51336be058172d0ed4d19ea47d697/src/CameraDevice.ts#L109

HorbachAndrii, 2021-10-04:
If you add the ability to control video I will be very happy.

HorbachAndrii, 2021-10-05:
Could you add the ability to record video with h264 codec in the next release?
I very much needed this feature.

mrousavy, 2021-10-05:
Right, I didn't include this in the format. I'm currently very busy with moving + work, so I don't know when I can build this feature. You can place a bug bounty to resolve it faster 😄

PR #645 (2021-12-30): feat: [ios] video codec option

## What

For IOS only, adds an optional `videoCodec` parameter to the `startRecording` options to specify the codec to record with. Requires IOS 11 (see commit a09201c2bc88eb6c11417046824cac88d9338069 vs 17b960603298e19d45e10933fe6dbf1da6ee00a6 .... please advise if not suitable)

## Changes
- handle user provided video codec in `CameraView+RecordVideo.swift`
- adds an optional `videoCodec` parameter to the `RecordVideoOptions` interface
- adds a `getAvailableVideoCodecs` method which takes a `fileType` parameter (see: https://stackoverflow.com/questions/4684384 [... cut]
closed 2021-12-30 by mrousavy