margelo/react-native-vision-camera #486: Use volume buttons to make photo / start video recoding✨

pke, 2021-10-04:
### What feature or enhancement are you suggesting?

Device default Camera apps allow users to use the volume buttons to make photos or start video recording.
It would be nice if this plugin would support this too.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

There is currently no workaround for this feature.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-10-05:
Hi!

> There is currently no workaround for this feature.

That's technically not correct. You can just use a react native library that detects when the volume down (or up) button is pressed, and call `takePhoto()` in that handler.

jqn, 2021-12-15:
@pke but how do you handle the volume button changing the volume level? Last time I tried I had to keep track of the volume levels and set it back to the original level when the camera screen unmounts.

mrousavy, 2023-09-01:
You'd have to handle those using events, this is not something the Camera should do imo

closed 2023-09-01 by mrousavy