margelo/react-native-vision-camera #488: 🐛   Native Module 'CameraView' was null!

manuelbieh, 2021-10-05:
### What were you trying to do?

Hey, first of all, although I've worked with React for over 5 years on a daily basis I am still pretty new to React Native, so please bear with me if I've missed out on a step or if I'm just use it incorrectly. 

I've just tried to use this wonderful package in a completely new and bare Expo project. Once I call `Camera.getAvailableCameraDevices()` I get the error above. 

One of the reasons might be that I'm using the most recent versions of basically everything. I'm using an iPhone 13 with iOS 15 to test on, Expo Go, Expo 42, XCode 13 on an arm64 Mac. Not sure about that. Anyway, I had to follow the steps from the troubleshooting guide and had to set `SWIFT_VERSION` to `5.2` manually. Which I did, but didn't help.

### Reproduceable Code

```tsx
The code I'm using is as minimal as it can be:

expo init react-native-app
cd react-native-app
expo install react-native-vision-camera
npx pod-install
expo prebuild
expo start
```

My App.tsx looks as simple as that:
```
import React, { useEffect } from 'react';
import { Button, View } from 'react-native';
import { Camera } from 'react-native-vision-camera';

export default function App() {
  useEffect(() => {
      const getDevices = async () => {
          const devices = await Camera.getAvailableCameraDevices();
          console.log(devices);
      };

      getDevices();
  }, []);

  return (
      <View>
          <Button title='Record video' onPress={() => {}} />
      </View>
  );
}
```
```

### What happened instead?

Once I open the app in Expo Go I get the following error:
```
Camera: Native Module 'CameraView' was null! Did you run pod install?
```
Yes, I did.

### Relevant log output

```shell
Camera: Native Module 'CameraView' was null! Did you run pod install?
at node_modules/react-native-vision-camera/src/Camera.tsx:40:26 in <global>
at node_modules/react-native-vision-camera/src/index.ts:1:0 in <global>
at App.tsx:3:0 in <global>
at index.js:4:0 in <global>
at http://192.168.178.193:19000/index.bundle?platform=ios&dev=true&hot=false&minify=false:121154:3 in global code

[Unhandled promise rejection: TypeError: null is not an object (evaluating 'CameraModule.getAvailableCameraDevices')]
at node_modules/react-native-vision-camera/src/Camera.tsx:257:31 in getAvailableCameraDevices
at node_modules/react-native-vision-camera/src/Camera.tsx:255:2 in getAvailableCameraDevices
at App.tsx:12:34 in getDevices
at App.tsx:11:27 in getDevices
at App.tsx:16:18 in useEffect$argument_0
at node_modules/react-native/Libraries/ReactNative/renderApplication.js:54:4 in renderApplication
at node_modules/react-native/Libraries/ReactNative/AppRegistry.js:117:25 in runnables.appKey.run
at node_modules/react-native/Libraries/ReactNative/AppRegistry.js:202:4 in runApplication
```
```

### Device

iPhone 13

### VisionCamera Version

2.8.3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-10-05:
Hey! So in Expo Go you cannot use _native_ libraries, and VisionCamera is a native library.

You can either use Expo EAS (with the prebuild command) or a bare react-native project to use VisionCamera.

closed 2021-12-13 by mrousavy

gregdelaun, 2023-03-18:
same also on IOS

dreese33, 2023-04-05:
Also seeing this on iOS