margelo/react-native-vision-camera #496: 🐛  iOS capture photo throws an error when enableAutoDistortionCorrection is enabled

kidculli, 2021-10-07:
### What were you trying to do?

Attempted to take a picture on IOS device with autoDistortionCorrection enabled

### Reproduceable Code

```tsx
<Camera
          style={{width: '100%', height: '100%'}}
          ref={camera}
          device={device as CameraDevice}
          preset="photo"
          isActive={true}
          photo={true}
        />

<FAB
        style={{position: 'absolute', margin: 16, right: 0, bottom: 0}}
        small
        onPress={() => {
       
            camera.current
              ?.takePhoto({
                qualityPrioritization: 'quality',
                skipMetadata: true,
                flash: 'off',
                enableAutoDistortionCorrection: true,
                enableAutoStabilization: true,
              })
              .then(image => {
                setPictureData(image);
                handleImageProcessing(image);
              })
              .finally(() => setIsProcessing(false));
          }
        }}
      />
```

### What happened instead?

Error is thrown from this [line](https://github.com/mrousavy/react-native-vision-camera/blob/8d24e344c9252a6a500e5f63a39539e0ff24fa11/ios/CameraView%2BTakePhoto.swift#L106)      

```
Exception	NSException *	"*** -[AVCapturePhotoOutput capturePhotoWithSettings:delegate:] settings.autoContentAwareDistortionCorrectionEnabled may not be set to YES unless self.contentAwareDistortionCorrectionEnabled is YES"	0x0000000280304120
```

### Relevant log output

```shell
flipper: FlipperClient::addPlugin Inspector
flipper: FlipperClient::addPlugin Preferences
flipper: FlipperClient::addPlugin React
flipper: FlipperClient::addPlugin Network
2021-10-07 18:21:00.108916-0400 FOO[1269:493267] [native] Running application FOO ({
    initialProps =     {
    };
    rootTag = 1;
})
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1007 18:21:00.110905 1872719872 Inspector.cpp:126] Inspector::Inspector transitioning to initial state RunningDetached
2021-10-07 18:21:00.136113-0400 FOO[1269:493590] FrameProcessorBindings: Creating Runtime Manager...
2021-10-07 18:21:00.136464-0400 FOO[1269:493590] FrameProcessorBindings: Runtime Manager created!
2021-10-07 18:21:00.136486-0400 FOO[1269:493590] FrameProcessorBindings: Installing Frame Processor plugins...
2021-10-07 18:21:00.136504-0400 FOO[1269:493590] FrameProcessorBindings: Frame Processor plugins installed!
2021-10-07 18:21:00.136550-0400 FOO[1269:493596] FrameProcessorBindings: Installing Frame Processor Bindings for Bridge...
2021-10-07 18:21:00.136574-0400 FOO[1269:493596] FrameProcessorBindings: Installing global functions...
2021-10-07 18:21:00.136603-0400 FOO[1269:493596] FrameProcessorBindings: Finished installing bindings.
I1007 18:21:04.714180 1872719872 Inspector.cpp:490] received didPause for reason: 0 in state: RunningDetached
2021-10-07 18:21:05.034038-0400 FOO[1269:493596] [javascript] Running "FOO" with {"rootTag":1,"initialProps":{}}
2021-10-07 18:21:05.160937-0400 FOO[1269:493588] [connection] nw_socket_handle_socket_event [C5:1] Socket SO_ERROR [61: Connection refused]
2021-10-07 18:21:05.161171-0400 FOO[1269:493592] [connection] nw_connection_get_connected_socket [C5] Client called nw_connection_get_connected_socket on unconnected nw_connection
2021-10-07 18:21:05.161271-0400 FOO[1269:493592] TCP Conn 0x283b9c630 Failed : error 0:61 [61]
2021-10-07 18:21:05.326671-0400 FOO[1269:493267] [native] [GESTURE HANDLER] Initialize gesture handler for root view <RCTRootContentView: 0x105d15f10; reactTag: 1; frame = (0 0; 414 896); gestureRecognizers = <NSArray: 0x2802ad680>; layer = <CALayer: 0x280aa23a0>>
2021-10-07 18:21:09.968471-0400 FOO[1269:493267] [native] VisionCamera.didSetProps(_:): Updating 10 prop(s)...
2021-10-07 18:21:09.968604-0400 FOO[1269:493588] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2021-10-07 18:21:09.968617-0400 FOO[1269:493588] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:6...
2021-10-07 18:21:09.968655-0400 FOO[1269:493588] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2021-10-07 18:21:09.978813-0400 FOO[1269:493588] [native] VisionCamera.configureCaptureSession(): Adding Photo output...
2021-10-07 18:21:09.982331-0400 FOO[1269:493588] [native] VisionCamera.invokeOnInitialized(): Camera initialized!
2021-10-07 18:21:09.982412-0400 FOO[1269:493588] [native] VisionCamera.configureCaptureSession(): Session successfully configured!
2021-10-07 18:21:09.982984-0400 FOO[1269:493588] [native] VisionCamera.configureFormat(): Configuring Format...
2021-10-07 18:21:09.982995-0400 FOO[1269:493588] [native] VisionCamera.configureDevice(): Configuring Device...
2021-10-07 18:21:09.983002-0400 FOO[1269:493588] [native] VisionCamera.configureDevice(): Device successfully configured!
2021-10-07 18:21:09.984223-0400 FOO[1269:493588] [native] VisionCamera.didSetProps(_:): Starting Session...
2021-10-07 18:21:10.706210-0400 FOO[1269:493588] [native] VisionCamera.didSetProps(_:): Started Session!
2021-10-07 18:21:34.506429-0400 FOO[1269:493590] [native] VisionCamera.takePhoto(options:promise:): Capturing photo...
```

### Device

iPhone 11 Pro Max  (iOS 15.0)

### VisionCamera Version

2.8.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

kidculli, 2021-10-12:
> Hi, thanks for the detailed bug report!

My pleasure, thanks for the fantastic module :)

almost, 2022-05-27:
Did you find any workarounds for this? Taking photos fails on iOS for me with this error and I don't think I have any particularly odd settings.

kidculli, 2022-05-27:
> Did you find any workarounds for this? Taking photos fails on iOS for me with this error and I don't think I have any particularly odd settings.

@almost I ended up using react-native-camera instead.

closed 2023-09-01 by mrousavy