margelo/react-native-vision-camera #501: 🐛 Can't run bootstrap on Linux (Ubuntu 21.04)

sedrik, 2021-10-11:
### What were you trying to do?

Try the example app

### Reproduceable Code

```tsx
On a Linux host.
Clone repo
run yarn bootstrap in repo dir.
```

### What happened instead?

After bundler has installed some dependencies a script that assumes that you are running on MacOS fails.

### Relevant log output

```shell
...
Fetching podspec for `glog` from `../node_modules/react-native/third-party-podspecs/glog.podspec`
Downloading dependencies
Installing DoubleConversion (1.1.6)
Installing FBLazyVector (0.66.0)
Installing FBReactNativeSpec (0.66.0)
Installing RCT-Folly (2021.06.28.00-v2)
Installing RCTRequired (0.66.0)
Installing RCTTypeSafety (0.66.0)
Installing RNGestureHandler (1.10.3)
Installing RNReanimated (2.3.0-beta.2)
Installing RNScreens (3.8.0)
Installing RNStaticSafeAreaInsets (2.1.1)
Installing RNVectorIcons (8.1.0)
Installing React (0.66.0)
Installing React-Core (0.66.0)
Installing React-CoreModules (0.66.0)
Installing React-RCTActionSheet (0.66.0)
Installing React-RCTAnimation (0.66.0)
Installing React-RCTBlob (0.66.0)
Installing React-RCTImage (0.66.0)
Installing React-RCTLinking (0.66.0)
Installing React-RCTNetwork (0.66.0)
Installing React-RCTSettings (0.66.0)
Installing React-RCTText (0.66.0)
Installing React-RCTVibration (0.66.0)
Installing React-callinvoker (0.66.0)
Installing React-cxxreact (0.66.0)
Installing React-jsi (0.66.0)
Installing React-jsiexecutor (0.66.0)
Installing React-jsinspector (0.66.0)
Installing React-logger (0.66.0)
Installing React-perflogger (0.66.0)
Installing React-runtimeexecutor (0.66.0)
Installing ReactCommon (0.66.0)
Installing VisionCamera (2.9.0)
Installing Yoga (1.14.0)
Installing boost (1.76.0)
Installing fmt (6.2.1)
Installing glog (0.3.5)
[!] /usr/bin/bash -c
set -e
#!/bin/bash
# Copyright (c) Facebook, Inc. and its affiliates.
#
# This source code is licensed under the MIT license found in the
# LICENSE file in the root directory of this source tree.

set -e

PLATFORM_NAME="${PLATFORM_NAME:-iphoneos}"
CURRENT_ARCH="${CURRENT_ARCH}"

if [ -z "$CURRENT_ARCH" ] || [ "$CURRENT_ARCH" == "undefined_arch" ]; then
    # Xcode 10 beta sets CURRENT_ARCH to "undefined_arch", this leads to incorrect linker arg.
    # it's better to rely on platform name as fallback because architecture differs between simulator and device

    if [[ "$PLATFORM_NAME" == *"simulator"* ]]; then
        CURRENT_ARCH="x86_64"
    else
        CURRENT_ARCH="arm64"
    fi
fi

export CC="$(xcrun -find -sdk $PLATFORM_NAME cc) -arch $CURRENT_ARCH -isysroot $(xcrun -sdk $PLATFORM_NAME --show-sdk-path)"
export CXX="$CC"

# Remove automake symlink if it exists
if [ -h "test-driver" ]; then
    rm test-driver
fi

./configure --host arm-apple-darwin

cat << EOF >> src/config.h
/* Add in so we have Apple Target Conditionals */
#ifdef __APPLE__
#include <TargetConditionals.h>
#include <Availability.h>
#endif

/* Special configuration for ucontext */
#undef HAVE_UCONTEXT_H
#undef PC_FROM_UCONTEXT
#if defined(__x86_64__)
#define PC_FROM_UCONTEXT uc_mcontext->__ss.__rip
#elif defined(__i386__)
#define PC_FROM_UCONTEXT uc_mcontext->__ss.__eip
#endif
EOF

# Prepare exported header include
EXPORTED_INCLUDE_DIR="exported/glog"
mkdir -p exported/glog
cp -f src/glog/log_severity.h "$EXPORTED_INCLUDE_DIR/"
cp -f src/glog/logging.h "$EXPORTED_INCLUDE_DIR/"
cp -f src/glog/raw_logging.h "$EXPORTED_INCLUDE_DIR/"
cp -f src/glog/stl_logging.h "$EXPORTED_INCLUDE_DIR/"
cp -f src/glog/vlog_is_on.h "$EXPORTED_INCLUDE_DIR/"

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for arm-apple-darwin-strip... no
checking for strip... strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for arm-apple-darwin-gcc...  -arch arm64 -isysroot
checking whether the C compiler works... no
/usr/bin/bash: line 25: xcrun: command not found
/usr/bin/bash: line 25: xcrun: command not found
/home/sedrik/Library/Caches/CocoaPods/Pods/External/glog/2263bd123499e5b93b5efe24871be317-53372/missing: Unknown `--is-lightweight' option
Try `/home/sedrik/Library/Caches/CocoaPods/Pods/External/glog/2263bd123499e5b93b5efe24871be317-53372/missing --help' for more information
configure: WARNING: 'missing' script is too old or missing
configure: error: in `/home/sedrik/Library/Caches/CocoaPods/Pods/External/glog/2263bd123499e5b93b5efe24871be317-53372':
configure: error: C compiler cannot create executables
See `config.log' for more details
```

### Device

Ubuntu 21.04

### VisionCamera Version

master branch

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-10-11:
Hi!

Thanks for the bug report. 
That looks like an issue with `pod install` (or a subtask in a CocoaPod, such as one from React Native), not with VisionCamera. If you're on Linux, you can skip the `yarn pods` (`pod install`) task anyways as you're not using iOS. :)

closed 2021-10-11 by mrousavy

sedrik, 2021-10-11:
The example README tells me to run `yarn bootstrap` in the root of react-native-vision-camera and that fails with the above error. I get that I should skip the pods for ios but I am but a simple-minded user who follows instructions :)