margelo/react-native-vision-camera #522: Where is scanQRCodes function ❓

thanhnh-implvn, 2021-10-15:
### Question

I cant use scanQRCodes function

### What I tried

```
import { scanQRCodes, QrCode } from 'vision-camera-qrcode-scanner'

import AppLoading from '@src/features/AppLoading'

interface Props {
  postingJobId: number
}

const RNCameraScanQR = ({ postingJobId }: Props) => {
  // const dispatch = useDispatch()
  // let checker = false
  // useFocusEffect(
  //   useCallback(() => {
  //     checker = false
  //   }, [])
  // )

  const devices = useCameraDevices()
  const device = devices.back
  const format = useCameraFormat(device)
  const [permissons, setPermissons] = useState(false)
  const [qrCodes, setQrCodes] = useState<QrCode[]>([])

  const requestPermissons = async () => {
    const newCameraPermission = await Camera.requestCameraPermission().catch((error) =>
      console.warn('camera error:', error)
    )
    if (newCameraPermission === 'authorized') {
      setPermissons(true)
    }
  }
  useEffect(() => {
    if (!permissons) {
      requestPermissons()
    }
  }, [permissons])

  useEffect(() => {
    ;(async () => {
      await Camera.getAvailableCameraDevices()
    })()
  }, [])

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'

    const qrcode = scanQRCodes(frame)
    console.log('Processing Frame:', qrcode)

    setQrCodes(qrcode)
    runOnJS(setQrCodes)(qrcode)
    // runOnUI(setQrCodes)(qrcode)
    console.log('qrCodes', qrCodes)
  }, [])

```

### VisionCamera Version

2.9.3 

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

rodgomesc, 2021-10-15:
this function does not exist in the library itself, you need to install a plugin to be able to scan qrcodes,

https://www.npmjs.com/package/vision-camera-qrcode-scanner

closed 2021-10-16 by mrousavy

lutfi-haslab, 2022-07-07:
i am facing same issue, this is how i resolve based on example vision-camera-code-scanner
`  
import {
  BarcodeFormat,
  scanBarcodes
} from "vision-camera-code-scanner";
import { runOnJS } from "react-native-reanimated";
const onQRCodeDetected = useCallback((qrCode) => {
    console.log(qrCode)
  }, [])

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    const qrCodes = scanBarcodes(frame, [BarcodeFormat.QR_CODE], { checkInverted: true });
    if (qrCodes.length > 0) {
      runOnJS(onQRCodeDetected)(qrCodes[0])
    }
  }, [onQRCodeDetected])`

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

Try upgrading to V3 if you can, it has a much more stable yet flexible foundation.

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.