margelo/react-native-vision-camera #524: RCTBridge required dispatch_sync to load RCTDevLoadingView. This may lead to deadlocks🐛

haibert, 2021-10-16:
### What were you trying to do?

Installed VisionCamera, and I started to get this warning, it was harmless in development but causes the app to crash in production. 

### Reproduceable Code

```tsx
https://github.com/haibert/sh43
```

### What happened instead?

I only  get a warning when using the custom development client, but the app is fully crashing when downloaded from TestFlight.  

### Relevant log output

```shell
RCTBridge required dispatch_sync to load RCTDevLoadingView. This may lead to deadlocks
at node_modules/react-native/Libraries/Utilities/RCTLog.js:34:8 in logIfNoNativeHook
```

### Device

iPhone 12 iOS 14

### VisionCamera Version

2.9.3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-10-16:
Huh, this seems unrelated to VisionCamera 🤔 Is the error gone when you uninstall VisionCamera?

haibert, 2021-10-20:
So this was happening from lottie of all things.. go figure. Weirdest part is I stopped the crashes by simply commenting out the lottie Icons in my bottom navigation tab bar, and when I un-commented them the app did not crash again... makes a world of sense :D

haibert, 2021-10-20:
After applying process of elimination by commenting out chunks of my app I was able to figure out that the crash was occurring when the bottom tab navigation would mount. I noticed that I have lottie views as my bottom tab button icons. I commented them out and oddly enough the app stopped crashing. What's even more strange is that when I uncommented them the app continued to work fine and I could not reproduce the crash that has been giving me a headache for 3 days.