margelo/react-native-vision-camera #526: ✨ AR - Record and Capture with AR objects

MadCoderme, 2021-10-16:
### What feature or enhancement are you suggesting?

I think it would be great if AR functionalities were available here. As we have in Viro, we can't actually record or capture images with the 3D objects or texts. If something similar can be added in this library, it will help a lot.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Currently, there are no alternatives for this, we can display images on the screen based on frame processor recognition, but we can't record or capture it

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-10-16:
Could you describe what you mean by "AR functionalities"? What's Viro?

> we can display images on the screen based on frame processor recognition, but we can't record or capture it

Yes you can, you can write a Frame Processor Plugin that also records the given `Frame`. Of course this is not part of VisionCamera itself, because VisionCamera doesn't know what you use to display "images on the screen".

MadCoderme, 2021-10-16:
By AR functionalities, I mean, integration of AR in this library. So we can show AR texts, objects and so on.

The method I mentioned is actually that useful to present AR. It is locked to screen, not to the scene.

Viro React is the best option right now for AR and VR in react native.

mrousavy, 2021-10-16:
I see. Sounds very interesting, and I believe a fast, nice to use, cross-platform AR API would be a game-changer.

However, the scope of this feature request is so huge that I won't be able to have enough free-time to work on it. If you, or anyone else reading this comment, is interested in this becoming reality, consider a collaboration with me and my app development/consulting agency. Contact me at me@mrousavy.com

MadCoderme, 2021-10-16:
Cool! Android uses ARCore library and IOS uses ARKit library. That can be a starting point.

Angelk90, 2022-01-09:
@MadCoderme : Are there any news?

Angelk90, 2022-01-10:
@MadCoderme : Are you testing on IOS or Android at the moment?

thomas-coldwell, 2022-10-22:
Marking this as a duplicate of https://github.com/mrousavy/react-native-vision-camera/issues/724. We are actively working on adding support for write-back frame processors in Vision Camera V3 (the in-progress PR for this functionality can be found at https://github.com/mrousavy/react-native-vision-camera/pull/1198).

Color filters or custom shaders can then be applied to the camera texture or the camera texture could be used as an unlit background in a 3D engine such as [Filament](https://github.com/google/filament)

closed 2022-10-22 by thomas-coldwell