margelo/react-native-vision-camera #542: Screen transitions broken in Example App on Android🐛

ldstein, 2021-10-19:
### What were you trying to do?

On Android, if the Permissions Screen is displayed and permission is granted to Camera and Microphone, the view will fail to transition to the Camera Screen:

I believe it is caused by an incompatibility between `@react-navigation/native` and `react-native-reanimated@2.3.0-beta.2`.

This is not a new bug. The example App screen transitions haven't worked on Android since ~2.7.0 after the move away from [wix/react-native-navigation](https://github.com/wix/react-native-navigation).

Worth keeping an eye on https://github.com/software-mansion/react-native-reanimated/issues/2460 as it looks like there is activity on resolving this.

[image]

### Reproduceable Code

```tsx
https://github.com/mrousavy/react-native-vision-camera/tree/7206a794a9e1a2da4def7fdf4d8a70c9eb7f43ff/example
```

### What happened instead?

Permission Screen displayed instead of Camera Screen

### Relevant log output

```shell
N/A
```

### Device

Pixel 3, Pixel 3XL, Android 11

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-10-19:
Thanks for the bug report!

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy