margelo/react-native-vision-camera #563: ❓ Android simulator Instant runtime crash but builds fine

pentatonictritones, 2021-10-31:
### Question

Hi,
Thank you for this package. It looks great.

Installing this for the first time, went through the tutorial and got reanimated. However, my app instantly crashes on launch now for Android simulator.

I went through the troubleshooting guide multiple times and this was the only thing that showed up in logcat.

How am I supposed to fix this? Thank you!

> 
> 2021-10-31 22:32:00.118 6278-6365/com.writingchecker E/AndroidRuntime: FATAL EXCEPTION: pool-7-thread-1
>     Process: com.writingchecker, PID: 6278
>     java.lang.UnsatisfiedLinkError: dlopen failed: library "libVisionCamera.so" not found
>         at java.lang.Runtime.loadLibrary0(Runtime.java:1077)
>         at java.lang.Runtime.loadLibrary0(Runtime.java:998)
>         at java.lang.System.loadLibrary(System.java:1656)
>         at com.mrousavy.camera.frameprocessor.FrameProcessorRuntimeManager.<clinit>(FrameProcessorRuntimeManager.kt:23)
>         at com.mrousavy.camera.CameraViewManager._init_$lambda-1(CameraViewManager.kt:20)
>         at com.mrousavy.camera.CameraViewManager.$r8$lambda$rPWatBJL5ihquxQ0qI4k0TW2orQ(Unknown Source:0)
>         at com.mrousavy.camera.CameraViewManager$$ExternalSyntheticLambda1.run(Unknown Source:4)
>         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
>         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
>         at java.lang.Thread.run(Thread.java:920)

### What I tried

Works fine on IOS simulator!

I looked at the gradle files for the example and tried them but got different errors.

Went through the troubleshooting guide multiple times.

Also, I tried building the example app but I have some unrelated issues with that (my computer) so I am not sure if it works or not on my computer.

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2021-11-01:
Did you try cleaning the gradle cache and rebuild everything?

pentatonictritones, 2021-11-01:
> Did you try cleaning the gradle cache and rebuild everything?

Yes I ran ./gradlew clean and rebuilt multiple times now.

This is definitely an issue on my end then right? Do you have any idea what could be causing this? Incompatibility with another module maybe?

Thanks!

pentatonictritones, 2021-11-01:
> Did you try cleaning the gradle cache and rebuild everything?

I was looking at the example build.gradles just to see the difference and I noticed that they included a couple of things that were different from mine.

The ones I found so far (Related to vision-camera) are:

```
dependencies {
    implementation "androidx.camera:camera-core:1.1.0-alpha08" 
in app/build.gradle 
```

and 

```
        @Override
        protected List<ReactPackage> getPackages() {
          
          @SuppressWarnings("UnnecessaryLocalVariable")
          List<ReactPackage> packages = new PackageList(this).getPackages();
          // Packages that cannot be autolinked yet can be added manually here, for example:
          packages.add(new CameraPackage());
          return packages;
          
        }
```

in Mainapplication.java.

Is there supposed to be a separate linking process that I am missing?

pentatonictritones, 2021-11-01:
By the way, the example application works fine! I just tried it.

So it's gotta be something to do with my project... I'd love if yo have any intuition/general directions to chase down.

suniahk, 2021-11-01:
A few debug questions:
- Did you install reanimated per the instructions on the reanimated docs?  There are a few specific steps for both Android and iOS, so it might be worth double-checking.  Found here for reference: https://docs.swmansion.com/react-native-reanimated/docs/fundamentals/installation/
- Did you try running on an actual device?  Based on the error I have a feeling that this won't matter, but always good to know.
- What version of React Native are you using? 
- The code you posted from the example deals with Frame processors.  Are you trying to use FP's?  If not, have you tried running the app with them disabled?

pentatonictritones, 2021-11-01:
> A few debug questions:
>
> * Did you install reanimated per the instructions on the reanimated docs?  There are a few specific steps for both Android and iOS, so it might be worth double-checking. [...]

Hi, 

Yes, I'll check the other points you raised.

About the frame processors, I am not looking to use them at the moment. How do I run the app with it disabled before launch? Because my code doesn't use anything with frame processors at all.

suniahk, 2021-11-01:
There's a blurb in the docs about how to disable them for iOS, found here: https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors#disabling-frame-processors

For Android, I assume that adding `VISION_CAMERA_DISABLE_FRAME_PROCESSORS=1` to /android/gradle.properties would have a similar effect, but I don't know for sure.  @mrousavy would have a better idea than I.

mrousavy, 2021-11-01:
Disabling Frame Processors does unfortunately not work yet on Android, see https://github.com/mrousavy/react-native-vision-camera/pull/412

pentatonictritones, 2021-11-02:
Thank you,

I tried out all of the above and it doesn't work haha....
It seems like
` java.lang.UnsatisfiedLinkError: dlopen failed: library "libVisionCamera.so" not found`

and specifically
`      System.loadLibrary("VisionCamera")`

isn't working.
I'll try it on a completely clean install and see if that works. I'm rebuilding my app anyways so clean install might be what it takes....

Thanks!

pentatonictritones, 2021-11-04:
Guys... I found the issue.

I ran a new project to narrow down where the error was coming from and it seems like there is some dependency issue when using older versions of React and ReactNative modules.

I was using React 0.65.1 and it wasn't working 
`Execution failed for task ':react-native-vision-camera:generateJsonModelDebug'.
> /Users/sungeunchoi/ReactNative/App/node_modules/react-native-vision-camera/android/CMakeLists.txt : C/C++ debug|armeabi-v7a : CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
  Please set them or make sure they are set and tested correctly in the CMake files:
  JSI_LIB
      linked by target "VisionCamera" in directory /Users/sungeunchoi/ReactNative/App/node_modules/react-native-vision-camera/android`

But I changed it to 0.66.1 and now everything works and builds perfectly!

Maybe one to consider for troubleshooting? Thanks for the help!

closed 2021-11-04 by pentatonictritones

ngxhuyhoang, 2024-10-31:
> > Disabling Frame Processors does unfortunately not work yet on Android, see #412
> 
> Hi, I got issue relate with this, how do I fix it. Please let me know, thanks Here's my issue:
> 
> ```
>  Fatal Exception: java.lang.UnsatisfiedLinkError: dlopen failed: library "libVisionCamera.so" not found
>        at java.lang.Runtime.loadLibrary0(Runtime.java:1087)
>        at java.lang.Runtime.loadLibrary0(Runtime.java:1008)
>        at java.lang.System.loadLibrary(System.java:1664)
>        at com.mrousavy.camera.react.CameraViewModule.<clinit>(CameraViewModule.kt:48)
>        at com.mrousavy.camera.react.CameraPackage.createNativeModules(CameraPackage.kt:11)
>        at com.facebook.react.ReactPackageHelper.getNativeModuleIterator(ReactPackageHelper.java:43)
>        at com.facebook.react.NativeModuleRegistryBuilder.processPackage(NativeModuleRegistryBuilder.java:42)
>        at com.facebook.react.ReactInstanceManager.processPackage(ReactInstanceManager.java:1458)
>        at com.facebook.react.ReactInstanceManager.processPackages(ReactInstanceManager.java:1429)
>        at com.facebook.react.ReactInstanceManager.createReactContext(ReactInstanceManager.java:1331)
>        at com.facebook.react.ReactInstanceManager.lambda$runCreateReactContextOnNewThread$2(ReactInstanceManager.java:1101)
>        at com.facebook.react.ReactInstanceManager.$r8$lambda$AwGS8CysOZmWJw3kRVARHQvw9Ew()
>        at com.facebook.react.ReactInstanceManager$$ExternalSyntheticLambda5.run(:4)
>        at java.lang.Thread.run(Thread.java:923)
>         
> ```

Same issue

egmepara, 2024-12-10:
Same Issue I am using RN v0.71.16 and Vision Camera 3.9.,2

koleCar, 2024-12-19:
RN version 0.75.3 vision-camera version 4.5.2 same issue