margelo/react-native-vision-camera #568: Cannot find module '@babel/preset-typescript' 🐛

anutting, 2021-11-02:
### What were you trying to do?

I am setting up my project to use react-native-vision-camera

### What happened instead?

The app bundle is not able to be built through the metro bundler:

```
error: node_modules/react-native-vision-camera/src/hooks/useFrameProcessor.ts: /node_modules/react-native-vision-camera/src/hooks/useFrameProcessor.ts: Cannot find module '@babel/preset-typescript'
```

I had to install a few other babel plugins just to get this far.

I received the same error for the follow plugins:

```
    '@babel/plugin-transform-shorthand-properties',
    '@babel/plugin-transform-arrow-functions',
    '@babel/plugin-transform-template-literals',
```

My project is not TypeScript and I should not be required to install these plugins to get this module working.

### Device

Pixel 4 XL

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

iSaCode, 2021-12-12:
I'm also encountering this issue.

error: FaceDetection.tsx: FaceDetection.tsx: Cannot find module '@babel/preset-typescript'
Require stack:
- \node_modules\@babel\core\lib\config\files\plugins.js
- \node_modules\@babel\core\lib\config\files\index.js
- \node_modules\@babel\core\lib\index.js
- \react-native\node_modules\metro-transform-worker\src\index.js
- \react-native\node_modules\metro\src\DeltaBundler\Worker.js
- \react-native\node_modules\jest-worker\build\workers\processChild.js
 BUNDLE  ./index.js

mrousavy, 2021-12-13:
Did you try cleaning your cache as mentioned in the Troubleshooting Guide? 

```
npx react-native start --reset-cache
```

Also, is babel installed correctly?

nandorojo, 2021-12-15:
So weird, I got this same issue after upgrading Reanimated to `2.3.0`.

Clearing the cache didn't help for me. Can't seem to figure out why Reanimated is causing this, but maybe it's a similar reason to vision camera?

mrousavy, 2022-01-02:
Closed due to inactivity

closed 2022-01-02 by mrousavy

tiavina-mika, 2022-01-05:
I'm facing the same issue with `react-native-gesture-handler. have you found a solution?

nandorojo, 2022-01-06:
`yarn add @babel/preset-typescript` fixed it for me with reanimated.

tiavina-mika, 2022-01-06:
@nandorojo thanks, it worked for me with `react-native-gesture-handler`

MuhammadAbdullah54321, 2022-03-02:
I had to pod install after doing `yarn add @babel/preset-typescript`