margelo/react-native-vision-camera #569: 🐛 Android - Taking picture with automatic flashlight

Hajan39, 2021-11-02:
### What were you trying to do?

When I want to take the first picture in the darkness, the automatic flash is not working on Android. The second image is OK, the flash is flashing.

### Reproduceable Code

```tsx
import { Camera } from 'react-native-vision-camera';

const cameraRef = useRef<Camera>(null);
const takePicture = () => {
    if (allowedNext) {
      setAllowedNext(false);

      cameraRef.current
        ?.takePhoto({
          flash: 'auto',
          qualityPrioritization: 'speed',
        }).then(x=> console.log(x));
}r
return(
<>
{device ? (
        <Camera
          ref={cameraRef}
          style={{
            opacity: takingPicture ? 1 : 0,
            ...wh,
            backgroundColor: '#f8f9fa',
          }}
          device={device}
          isActive={true}
          preset={'photo'}
          zoom={currentZoom}
          onError={onError}
          onTouchEnd={x => device?.supportsFocus && onTouch(x.nativeEvent)}
          torch={device.hasTorch && torch ? 'on' : 'off'}
          hdr={true}
          onInitialized={cameraReady}
          enableZoomGesture={true}
          photo={true}
        />
      ) : (
        <Text>No device found</Text>
      )}
        <TouchableOpacity onPress={props.takePicture}  />
</>)
```

### What happened instead?

The first image is without flash and the second image is with flash. The same darkness, the same spot, 2s later.

### Device

Samsung s10e, Huawei P30lite

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Hajan39, 2021-11-19:
@mrousavy  any update on this?

mrousavy, 2021-11-20:
Hey! I haven't had time to look into this yet.

Hajan39, 2022-01-21:
Hello, would it be possible to let me know if there is a plan to fix this, please? I always have the first picture black and others are fine.

mrousavy, 2022-01-21:
Hey again! I still haven't had time to investigate this, I am currently really busy with the agency I'm running. You can hire us to fix it, or put a bugbounty on this issue, but I don't really have any free time at the moment.

Also, I don't have neither of those devices and for me (Huawei P10) the flash correctly fires.

Maybe it is also a native CameraX library issue and you could try replicate it in a native Android app, and if it is, post an issue in the Google Issue Tracker for CameraX.

Hajan39, 2022-01-22:
To be precise, it is only the first attempt. The flash is not working only for the first picture when you open an app in complete darkness. Others are fine, but only the first one is not working.

Hajan39, 2022-01-26:
@mrousavy I think I found a solution - it seems that this code is not working correctly in the CameraView+TakePhoto.kt:

`imageCapture!!.flashMode = when (flashMode) {
      "on" -> ImageCapture.FLASH_MODE_ON
      "off" -> ImageCapture.FLASH_MODE_OFF
      "auto" -> ImageCapture.FLASH_MODE_AUTO
      else -> throw InvalidTypeScriptUnionError("flash", flashMode ?: "(null)")
    }`

It looks like the flashMode should be set when creating the imageCapture !or! it has to be rebuilt somehow - sorry, I am not a Kotlin developer, but I am trying my best.
I have tested Android app and the flash is working great, just because it is set directly to the ImageCapture.Builder().

I have changed your code in CameraView.kt this:

`
val flashMode = "auto"
val modeFlash = when (flashMode) {
        "on" -> ImageCapture.FLASH_MODE_ON
        "off" -> ImageCapture.FLASH_MODE_OFF
        "auto" -> ImageCapture.FLASH_MODE_AUTO
        else -> throw InvalidTypeScriptUnionError("flash", flashMode ?: "(null)")
      }
  imageCaptureBuilder.setFlashMode(modeFlash)`

And it is working! I have no answer to the problem, but this is a solution. Can you please help me add flashMode as a parameter in the <Camera flashMode="auto" /> object? 
Thank you.

TasigurIshtar, 2022-01-26:
Hello, 
I can confirm same behavior on Xiaomi Mi9t. 
Flash on Auto & dark room. 
First shot is without flash, the others are with flash.

Hajan39, 2022-01-26:
@mrousavy Please have a look at this change - it is working correctly 
https://github.com/mrousavy/react-native-vision-camera/compare/main...Hajan39:main

Hajan39, 2022-01-26:
OK, but as you can see, this is not working correctly on more devices. I think you will find more devices ... 
AFAIK this has a late initialization, and it changes on the second attempt. Every time you set an "auto" flash mode, the first image is black. I am not sure if this is the problem of CameraX, but this approach seems to be wrong.

alexstanbury, 2022-03-30:
FWIW I'm seeing the same issue on a Xiaomi Mi9

2sem, 2023-05-19:
some device flash not working...
os problem?

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy