margelo/react-native-vision-camera #570: 🐛 App crashes on view unmount

heroic, 2021-11-04:
### What were you trying to do?

When the camera view is open, and the android activity is being destroyed, unmounting the camera causes a crash.

### What happened instead?

App crashed.

### Relevant log output

```shell
Error
Exception in HostFunction: com.mrousavy.camera.e0: [system/view-not-found] The given view (ID 2537) was not found in the view manager.

    (native) unsetFrameProcessor
    index.android.bundle:1:1558254 unsetFrameProcessor
    index.android.bundle:1:1558456 componentWillUnmount
    index.android.bundle:1:112490 commitUnmount
    index.android.bundle:1:114515 unmountHostComponents
    index.android.bundle:1:122731 commitRootImpl
    index.android.bundle:1:246972 anonymous
    index.android.bundle:1:88119 runWithPriority
    index.android.bundle:1:121900 commitRoot
    index.android.bundle:1:118763 performSyncWorkOnRoot
    index.android.bundle:1:88534 anonymous
    index.android.bundle:1:246972 anonymous
    index.android.bundle:1:88119 runWithPriority
    index.android.bundle:1:88352 flushSyncCallbackQueueImpl
    index.android.bundle:1:88246 flushSyncCallbackQueue
    index.android.bundle:1:116398 scheduleUpdateOnFiber
    index.android.bundle:1:127771 updateContainer
    index.android.bundle:1:127993 unmountComponentAtNode
    index.android.bundle:1:137487 anonymous
    index.android.bundle:1:452190 unmountApplicationComponentAtRootTag
    (native) apply
    index.android.bundle:1:64407 __callFunction
    index.android.bundle:1:62896 anonymous
    index.android.bundle:1:63853 __guard
    index.android.bundle:1:62854 callFunctionReturnFlushedQueue
```
```

### Device

ZenFone Max Pro M1 (ZB602KL) (WW) / Max Pro M1 (ZB601KL) (IN) (ASUS_X00TD)

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

rusakovic, 2021-11-06:
this issue is discussed here: https://github.com/mrousavy/react-native-vision-camera/issues/500

heroic, 2021-11-07:
@rusakovic that's a different error. That one was fixed in 2.9.1 or 2.9.2. This issue happens after that fix. Notice the stacktrace now says error was in unsetFrameProcessor, but the earlier one was in setFrameProcessor.

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy