margelo/react-native-vision-camera #571: maxDuration props✨

Ppang0405, 2021-11-05:
### What feature or enhancement are you suggesting?

 I think it would be great if we have maxDuration props.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I mean using JS Timer is not good enough.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-11-05:
> I mean using JS Timer is not good enough.
Why?

Ppang0405, 2021-11-07:
Because the JS timer does not guarantee the exact delay [somewhere around 300ms -1000ms missing/redundency]. I need high precision [around 2-5ms is ok].

mrousavy, 2021-11-08:
Where did you get the info from that a Timer has a 1000ms inaccuracy?

Ppang0405, 2021-11-11:
I got it from my company project. 
Close this issues because I have already fork and do my own custom.

closed 2021-11-11 by Ppang0405