margelo/react-native-vision-camera #572: ✨ Possibility to have `orientation` props.

geoffreyLB, 2021-11-05:
### What feature or enhancement are you suggesting?

I think it would be great to have as `react-natice-camera` a props `orientation` to pass in `takePhoto`. 
Actually if you have your app lock in `portrait`, you can't just turn your phone and take a photo in landscape mode. 

Switched from `react-native-camera` to this repository and I have to admit this feature miss a lot. 

Let me know what you think.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

I used to `unlock-portrait-mode` on the app when taking picture and unfortunately the transition on IOS device is not smooth at all, the previous screen drools on camera screen.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-11-08:
Hi!
I don't really get what `orientation` should do, why do you need that feature? If you want to take a landscape picture, either rotate your phone to landscape (don't lock to portrait) or rotate the picture afterwards?

jzxchiang1, 2021-12-05:
+1, this feature would be helpful.

> Hi! I don't really get what `orientation` should do, why do you need that feature? If you want to take a landscape picture, either rotate your phone to landscape (don't lock to portrait) or rotate the picture afterwards?

Well, sometimes we have to lock to portrait if we don't want the UI to get rotated. For example, that's what the Camera app does on Android (instead of rotating the entire UI, it just rotates a couple of buttons). But even though the UI is locked, we still want to take a photo in landscape mode if the device is rotated.

mrousavy, 2022-01-04:
Okay that's interesting. This would be easy for takePhoto, although it comes with a performance penalty on Android. For Video Recordings, it's a bit more complicated. I can see how I can achieve it on iOS, but no idea how that would work for Android's CameraX Recorder class.

mrousavy, 2022-01-04:
Implemented in #715! Hope this works for you guys :) If you appreciate the work I do in my free time, please consider sponsoring me on GitHub, it would mean a lot to me. 🖤

closed 2022-02-11 by geoffreyLB