margelo/react-native-vision-camera #583: 🐛 Torch stays on after recording a video

mikefogg, 2021-11-11:
### What were you trying to do?

Record a video without the torch remaining on afterward.

### Reproduceable Code

```tsx
# Reproducable in the example app
```

### What happened instead?

When recording a video with flash `on`, it doesn't seem to turn the torch back off again after the recording is complete. The torch will stay on indefinitely until I either:

1. Teardown the `<Camera>` component by setting `isActive` to false
2. Disable the flash (set to `off`) and then record another video

### Device

iPhone 13 Pro – iOS 15.2

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-11-11:
How does this behave on Android?

mrousavy, 2021-11-11:
Does https://github.com/mrousavy/react-native-vision-camera/pull/584 fix your issue?

mikefogg, 2021-11-11:
@mrousavy I was only able to try it for a second, but it did! Shut right off :) (Also unfortunately I'm not setup on Android). Thanks so much for looking at that quickly!

hungdev, 2021-12-05:
Im getting same issue

PR #584 (2021-12-10): fix: Fix torch staying on after recording a video

## What

* Resets the torch to the state before after recording a video

## Changes

## Tested on

## Related issues

* Fixes #583
closed 2021-12-10 by mrousavy