margelo/react-native-vision-camera #587: 🐛

chiubaka, 2021-11-15:
### What were you trying to do?

Integrate some of the community frame processor plugins into my app. I'm specifically interested in `vision-camera-qrcode-scanner`, as my app makes heavy use of QR code scanning.

Notably, this seems related to #350 and #376, both of which seem to have been resolved by a Hermes-specific fix. I am running on RN 0.66.3 _without Hermes enabled_ as I've encountered several blockers making it difficult for me to get Hermes working properly at this time. (Specifically, lack of full Intl support and poor performance in polyfills I've tried to use to circumvent the issue.)

I have upgraded to `react-native-reanimated` v2.3.0-beta.3.

### Reproduceable Code

```tsx
// ReceiptScanner.tsx
import "react-native-reanimated";

import React, { useState } from "react";
import { LayoutChangeEvent, StyleSheet } from "react-native";
import { runOnJS } from "react-native-reanimated";
import Svg, { Rect } from "react-native-svg";
import {
  Camera,
  useCameraDevices,
  useFrameProcessor,
} from "react-native-vision-camera";
import { QrCode, scanQRCodes } from "vision-camera-qrcode-scanner";

import { useMountEffect } from "../hooks";

export const ReceiptScanner = () => {
  const cameras = useCameraDevices();
  const backCamera = cameras.back;

  const [cameraPermitted, setCameraPermitted] = useState(false);
  const [qrCodes, setQrCodes] = useState<QrCode[]>([]);

  const [frameHeight, setFrameHeight] = useState<number>();
  const [frameWidth, setFrameWidth] = useState<number>();

  const [layoutHeight, setLayoutHeight] = useState<number>();
  const [layoutWidth, setLayoutWidth] = useState<number>();

  useMountEffect(() => {
    const ensureCameraPermissions = async () => {
      let cameraPermission = await Camera.getCameraPermissionStatus();

      if (cameraPermission === "not-determined") {
        cameraPermission = await Camera.requestCameraPermission();
      }

      setCameraPermitted(cameraPermission === "authorized");
    };

    void ensureCameraPermissions();
  });

  const frameProcessor = useFrameProcessor((frame) => {
    "worklet";

    // Frame processing seems locked in landscape mode, so frame.width > frame.height
    // Flip this values so we can work in portrait mode coordinates
    if (frame.width !== frameHeight) {
      runOnJS(setFrameHeight)(frame.width);
    }

    if (frame.height !== frameWidth) {
      runOnJS(setFrameWidth)(frame.height);
    }

    const scannedQrCodes = scanQRCodes(frame);
    runOnJS(setQrCodes)(scannedQrCodes);
  }, []);
[... 1684 lines cut ...]
2021-11-15 10:59:40.711 22590-23042/com.chiubaka.Midana.dev A/libc: Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 23042 (pool-17-thread-), pid 22590 (baka.Midana.dev)
2021-11-15 10:59:41.823 24638-24638/? A/DEBUG: Cmdline: com.chiubaka.Midana.dev
2021-11-15 10:59:41.823 24638-24638/? A/DEBUG: pid: 22590, tid: 23042, name: pool-17-thread-  >>> com.chiubaka.Midana.dev <<<
2021-11-15 10:59:41.823 24638-24638/? A/DEBUG:       #11 pc 0000000000019098  /data/app/~~mJL--us2rnI3kygQVDeeqQ==/com.chiubaka.Midana.dev-a3XYOgKvFxS96O66uGBH4A==/lib/arm64/libfbjni.so (facebook::jni::isObjectRefType(_jobject*, jobjectRefType)+76) (BuildId: 5b9237c4cf8ff477d69b3a6401b060d9b7f17a75)
2021-11-15 10:59:41.823 24638-24638/? A/DEBUG:       #12 pc 00000000000594d4  /data/app/~~mJL--us2rnI3kygQVDeeqQ==/com.chiubaka.Midana.dev-a3XYOgKvFxS96O66uGBH4A==/lib/arm64/libVisionCamera.so (facebook::jni::LocalReferenceAllocator::verifyReference(_jobject*) const+32) (BuildId: e19355c1efab96316499dc73d5be9969429b5eb9)
2021-11-15 10:59:41.823 24638-24638/? A/DEBUG:       #13 pc 000000000005954c  /data/app/~~mJL--us2rnI3kygQVDeeqQ==/com.chiubaka.Midana.dev-a3XYOgKvFxS96O66uGBH4A==/lib/arm64/libVisionCamera.so (facebook::jni::LocalReferenceAllocator::deleteReference(_jobject*) const+80) (BuildId: e19355c1efab96316499dc73d5be9969429b5eb9)
2021-11-15 10:59:41.823 24638-24638/? A/DEBUG:       #14 pc 000000000006b824  /data/app/~~mJL--us2rnI3kygQVDeeqQ==/com.chiubaka.Midana.dev-a3XYOgKvFxS96O66uGBH4A==/lib/arm64/libVisionCamera.so (facebook::jni::base_owned_ref<vision::JImageProxy, facebook::jni::LocalReferenceAllocator>::reset(facebook::jni::detail::JTypeFor<vision::JImageProxy, facebook::jni::JObject, void>::_javaobject*)+144) (BuildId: e19355c1efab96316499dc73d5be9969429b5eb9)
2021-11-15 10:59:41.823 24638-24638/? A/DEBUG:       #15 pc 000000000006b76c  /data/app/~~mJL--us2rnI3kygQVDeeqQ==/com.chiubaka.Midana.dev-a3XYOgKvFxS96O66uGBH4A==/lib/arm64/libVisionCamera.so (facebook::jni::base_owned_ref<vision::JImageProxy, facebook::jni::LocalReferenceAllocator>::reset()+28) (BuildId: e19355c1efab96316499dc73d5be9969429b5eb9)
2021-11-15 10:59:41.873 1781-24654/? W/ActivityTaskManager:   Force finishing activity com.chiubaka.Midana.dev/com.chiubaka.midana.MainActivity
```

### Device

Android Pixel 3 running Android 12

### VisionCamera Version

2.9.3

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

chiubaka, 2021-11-15:
I filed https://github.com/rodgomesc/vision-camera-qrcode-scanner/issues/16 over at @rodgomesc's `vision-camera-qrcode-scanner` because I wasn't sure if this was specific to that plugin.

However, after testing with @mrousavy's `vision-camera-image-labeler`, I've found that the app tends to crash after awhile with a similar error. I _think_ it works a little bit longer with `vision-camera-image-labeler`, but that's just a feeling.

Here's the log output when using `vision-camera-image-labeler`:
```
2021-11-15 17:47:24.793 9108-9325/com.chiubaka.Midana.dev A/baka.Midana.de: java_vm_ext.cc:579] JNI DETECTED ERROR IN APPLICATION: JNI ERROR (app bug): jobject is an invalid local reference: 0x15 (deleted reference at index 1 in a table of size 1)
    java_vm_ext.cc:579]     in call to GetObjectRefType
    java_vm_ext.cc:579]     from void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy)
2021-11-15 17:47:25.086 9108-9325/com.chiubaka.Midana.dev A/baka.Midana.de: runtime.cc:669] Runtime aborting...
    runtime.cc:669] Dumping all threads without mutator lock held
    runtime.cc:669] All threads:
    runtime.cc:669] DALVIK THREADS (69):
    runtime.cc:669] "pool-17-thread-1" prio=5 tid=77 Runnable
    runtime.cc:669]   | group="" sCount=0 ucsCount=0 flags=0 obj=0x165063f0 self=0x7d9a0e7bf0
    runtime.cc:669]   | sysTid=9325 nice=0 cgrp=default sched=0/0 handle=0x7b2bde3cb0
    runtime.cc:669]   | state=R schedstat=( 4616202011 127027893 2428 ) utm=438 stm=22 core=4 HZ=100
    runtime.cc:669]   | stack=0x7b2bce0000-0x7b2bce2000 stackSize=1039KB
    runtime.cc:669]   | held mutexes= "abort lock" "mutator lock"(shared held)
    runtime.cc:669]   native: #00 pc 000000000055f938  /apex/com.android.art/lib64/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+140)
    runtime.cc:669]   native: #01 pc 0000000000676358  /apex/com.android.art/lib64/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool, BacktraceMap*, bool) const+360)
    runtime.cc:669]   native: #02 pc 0000000000694034  /apex/com.android.art/lib64/libart.so (art::DumpCheckpoint::Run(art::Thread*)+920)
    runtime.cc:669]   native: #03 pc 000000000068db58  /apex/com.android.art/lib64/libart.so (art::ThreadList::RunCheckpoint(art::Closure*, art::Closure*)+520)
    runtime.cc:669]   native: #04 pc 000000000068cd6c  /apex/com.android.art/lib64/libart.so (art::ThreadList::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool)+1464)
    runtime.cc:669]   native: #05 pc 0000000000626e50  /apex/com.android.art/lib64/libart.so (art::Runtime::Abort(char const*)+2164)
    runtime.cc:669]   native: #06 pc 000000000001595c  /apex/com.android.art/lib64/libbase.so (android::base::SetAborter(std::__1::function<void (char const*)>&&)::$_3::__invoke(char const*)+76)
    runtime.cc:669]   native: #07 pc 0000000000014f8c  /apex/com.android.art/lib64/libbase.so (android::base::LogMessage::~LogMessage()+364)
    runtime.cc:669]   native: #08 pc 00000000004501fc  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+2516)
    runtime.cc:669]   native: #09 pc 0000000000450274  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbortV(char const*, char const*, std::__va_list)+108)
    runtime.cc:669]   native: #10 pc 00000000004424bc  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::AbortF(char const*, ...)+140)
    runtime.cc:669]   native: #11 pc 0000000000441a30  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::CheckInstance(art::ScopedObjectAccess&, art::(anonymous namespace)::ScopedCheck::InstanceKind, _jobject*, bool)+1116)
    runtime.cc:669]   native: #12 pc 000000000044028c  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::CheckPossibleHeapValue(art::ScopedObjectAccess&, char, art::(anonymous namespace)::JniValueType)+568)
    runtime.cc:669]   native: #13 pc 000000000043f870  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::Check(art::ScopedObjectAccess&, bool, char const*, art::(anonymous namespace)::JniValueType*)+644)
    runtime.cc:669]   native: #14 pc 000000000043ef58  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::GetObjectRefType(_JNIEnv*, _jobject*) (.llvm.11195161549560735479)+656)
    runtime.cc:669]   native: #15 pc 0000000000019098  /data/app/~~yXtM5iqD-_hLZuqHiRV80A==/com.chiubaka.Midana.dev-m5xkJPUE7KaASBmizEfmqg==/lib/arm64/libfbjni.so (facebook::jni::isObjectRefType(_jobject*, jobjectRefType)+76)
    runtime.cc:669]   native: #16 pc 00000000000594d4  /data/app/~~yXtM5iqD-_hLZuqHiRV80A==/com.chiubaka.Midana.dev-m5xkJPUE7KaASBmizEfmqg==/lib/arm64/libVisionCamera.so (facebook::jni::LocalReferenceAllocator::verifyReference(_jobject*) const+32)
2021-11-15 17:47:25.086 9108-9325/com.chiubaka.Midana.dev A/baka.Midana.de: runtime.cc:669]   native: #17 pc 000000000005954c  /data/app/~~yXtM5iqD-_hLZuqHiRV80A==/com.chiubaka.Midana.dev-m5xkJPUE7KaASBmizEfmqg==/lib/arm64/libVisionCamera.so (facebook::jni::LocalReferenceAllocator::deleteReference(_jobject*) const+80)
[... cut ...] 9)
2021-11-15 17:47:26.217 11869-11869/? A/DEBUG:       #15 pc 000000000006b76c  /data/app/~~yXtM5iqD-_hLZuqHiRV80A==/com.chiubaka.Midana.dev-m5xkJPUE7KaASBmizEfmqg==/lib/arm64/libVisionCamera.so (facebook::jni::base_owned_ref<vision::JImageProxy, facebook::jni::LocalReferenceAllocator>::reset()+28) (BuildId: e19355c1efab96316499dc73d5be9969429b5eb9)
2021-11-15 17:47:26.268 1781-11889/? W/ActivityTaskManager:   Force finishing activity com.chiubaka.Midana.dev/com.chiubaka.midana.MainActivity
```

chiubaka, 2021-11-15:
I'm a little mystified by the errors here haha 😅. But more than happy to fetch whatever other information is needed! Also happy to try to create a repro project if that would help. Hoping that maybe this is just me missing something obvious.

rodgomesc, 2021-11-19:
I don't know if you will be able to run this lib with hermes disabled, hermes is mandatory in react native reanimated setup, and to use frameprocessors, you need to have reanimated configured, one thing leads to another

chiubaka, 2021-11-19:
Yeah, I'm aware of the hermes requirement in the `react-native-reanimated` docs. Though, frame processors and `react-native-reanimated` in general do seem to work without `hermes` otherwise. @mrousavy had [suggested that I turn off `hermes`](https://github.com/mrousavy/react-native-vision-camera/issues/292#issuecomment-885597030) in a comment awhile back, so not totally sure if it's meant to be a strict requirement here.

mrousavy, 2021-11-19:
Nor reanimated or VisionCamera require Hermes. Both work fine with JSC, and I use JSC in one of my apps. (I think even Colorwaver)

chiubaka, 2021-11-20:
Sweet, that's good to know. So there's hope, yet haha!

It feels to me like a couple of immediate next steps on this issue to gather more debugging info are:
1. Set up a clean repo with my current dependency versions, and see if switching `hermes` on vs off triggers this issue on my device.
2. Potentially try to run Colorwaver or one of @mrousavy's other JSC projects on my device and see if I run into this issue. If I don't, maybe see what I have to change to trigger the issue.

I'm tied up with a couple of other things in the moment--decided to stick to `react-native-camera` on Android until I can get to the bottom of this, so have been making a series of PRs to fix bugs over there--but can potentially play with this later this week.

mrousavy, 2021-11-22:
Might be interesting to debug this, maybe I am misusing a fbjni reference type somewhere.

alexstanbury, 2021-11-26:
I'm having the same issue, tried with the `vision-camera-image-labeler` and also with my own custom mlkit labelling plugin. It runs ok for a minute or so then just crashes with [this log](https://gist.github.com/alexstanbury/b33105e92511b27c47d9a1deced2cab5)

FWIW it will crash even if I don't move the device at all and have the camera lens face down.

I've created a repo [here](https://github.com/alexstanbury/frame-processor-crash) that demonstrates it. 

Using Pixel 6 running Android 12
```
"dependencies": {
    "react": "17.0.2",
    "react-native": "0.66.3",
    "react-native-reanimated": "^2.2.4",
    "react-native-vision-camera": "^2.9.4",
    "vision-camera-image-labeler": "^0.1.6"
  }
```

chiubaka, 2021-11-26:
Notably, @alexstanbury's repo has Hermes enabled for Android. So maybe this issue isn't isolated to JSC like I thought?

suniahk, 2021-12-17:
I've had some success in mitigating this issue, although I'm still confused about what exactly is happening (moreso I don't really understand JNI's side of things, but I digress).  It seems like the memory allocated to the returned object from our callback isn't being deallocated in a reasonable timeframe.  I think there's more to it than just allocation and deallocation though, for reasons I'll explain below.  My solution was to always return null, and instead pass the processed data back to the React Native app using an EventEmitter.  Since handling my return values this way, my app hasn't crashed once due to this particular issue.

My use case was that I was grabbing the frames, converting them to base64, and returning those ~2kb strings back.  Memory usage would rise and drop, but always at a steady overall increase until the app crashed and that error popped up.  I honestly don't know if there's a better way to do what I want, this was just the one that I found that works and maintains performance.  This is also on Android specifically as I haven't implemented the native code on the iOS side yet.
- Convert the frame into an nv21 YUVImage
- Process THAT image into a scaledBitmap. (for rotation purposes, as you can't apply a matrix to other image types and the workarounds that people have made for YUVImage have horrible performance)
- Check the ImageProxy.getImage().getInfo() for rotation/orientation.  If it doesn't match our current screen orientation, apply a rotation to the scaledBitmap to match. (captured frames can sometimes be rotated due to camera/screen orientation mismatch)
- compress the image to a jpeg (I'm sending the frame out via WebRTC, which is being handled within the app).

As far as my limited debugging efforts go, I'm not particularly well versed in C++ so I'm not sure how helpful it will be but I'll add my efforts to the pile in case it helps.
- First up, my environment:
  - android/build.gradle (all the boilerplate files have been mostly updated to 66.4, although there are still a few holdouts due to compatibility issues a la gradle 7/removal of jcenter)
```buildscript {
    ext {
        buildToolsVersion = "31.0.0"
        minSdkVersion = 23
        compileSdkVersion = 31
        targetSdkVersion = 31
        ndkVersion = "20.1.5948944"
        googlePlayServicesAuthVersion = "16.0.1"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:4.1.2")
        classpath 'com.google.gms:google-services:4.1.0'
    }
}

allprojects {
    repositories {
        mavenLocal()
        google()
        maven {
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            url("$rootDir/../node_modules/jsc-android/dist")
        }

        google()
        jcenter()
        maven { url 'https://www.jitpack.io' }
    }
}```
  - Packages (truncated to assumed relevant packages only)
``` 
    "@react-navigation/bottom-tabs": "^6.0.9",
    "@react-navigation/drawer": "^6.1.8",
    "@react-navigation/native": "^6.0.6",
    "@react-navigation/native-stack": "^6.2.5",
    "@react-navigation/stack": "^6.0.11",
    "expo": "^43.0.4",
    "expo-background-fetch": "~10.0.3",
    "expo-constants": "~12.1.3",
    "appcenter": "4.3.0",
    "appcenter-analytics": "4.3.0",
    "appcenter-crashes": "4.3.0",
    "react": "17.0.2",
    "react-i18next": "^11.14.3",
    "react-native": "0.66.4",
    "react-native-fast-image": "^8.5.11",
    "react-native-reanimated": "^2.3.0",
    "react-native-safe-area-context": "^3.3.2",
    "react-native-screens": "^3.10.1",
    "react-native-vision-camera": "^2.10.0",
```
- First, the bug happens on both JSC and Hermes.  I did multiple cleans/rebuilds to verify that nothing was being unexpectedly cached.  I did see additional/different errors on JSC, although, although nothing particularly helpful.  It didn't look like the "Frame Processor threw an error!" log from CameraView.cpp.
- Based on the stacktrace, my assumption currently is that if there isn't enough memory to allocate for the app, the frameproccessor doesn't return at all or returns something like an EoM) and so `JSIJNIConversion::convertJNIObjectToJSIValue` can't convert the value.  You'd think it would be a nullptr but the app crashes around the time it wants to check that so there's definitely something else going on here.
- I tried this with all sorts of combinations of image formts/processing code. The crash happened significantly slower without Bitmaps but in all cases, still happened.  The only thing that worked to mitigate the crash was to remove my return and replace it with `return null`.
- I tried some basic memory management in my callback; no luck (clearing bytebuffers/arrays closing images, etc).  The issue in my case was absolutely instigated by the amount of memory that bitmaps use, and handling bitmaps  in Android is just one giant pain in the butt.
- Just for kicks, I tried to call image.close() (on the callback param).  Surprisingly, nothing happened (not even the errors I was expecting)

If anyone has any questions, I'm happy to answer them.

mrousavy, 2021-12-21:
couple of pointers for someone that wants to debug this (or maybe my future self):

---

Make sure nothing here leaks and everything is safely collected:

https://github.com/mrousavy/react-native-vision-camera/blob/b10ca27daabeb18ec3652ffc64fbbe0fecb298f5/android/src/main/cpp/FrameProcessorRuntimeManager.cpp#L255-L269

☝️ this is probably the root cause of the problem. 99% sure.

---

Not sure if that should be a JNI `alias_ref` or better a `local_ref`?

https://github.com/mrousavy/react-native-vision-camera/blob/b10ca27daabeb18ec3652ffc64fbbe0fecb298f5/android/src/main/cpp/CameraView.cpp#L31

---

This HostObject Shared Pointer should be auto-collected, make sure that there is no strong reference held to that after I'm done so that the JS runtime can delete it:

https://github.com/mrousavy/react-native-vision-camera/blob/b10ca27daabeb18ec3652ffc64fbbe0fecb298f5/android/src/main/cpp/FrameProcessorRuntimeManager.cpp#L140

---

Make sure this is safe as Hermes calls the destructor on a separate thread, whereas JSC calls it on the same thread:

https://github.com/mrousavy/react-native-vision-camera/blob/b10ca27daabeb18ec3652ffc64fbbe0fecb298f5/android/src/main/cpp/FrameHostObject.cpp#L23

FloweryK, 2021-12-28:
@mrousavy Hi. Not sure whether this is the cause or not. but I checked line by line and found that `count` is always 1 so that the for loop below is always skipped. Is this OK?

https://github.com/mrousavy/react-native-vision-camera/blob/b10ca27daabeb18ec3652ffc64fbbe0fecb298f5/android/src/main/cpp/FrameProcessorRuntimeManager.cpp#L261-L263

mrousavy, 2021-12-30:
Yes, that just means you have no additional parameters.

mrousavy, 2021-12-30:
Just to clarify - does the memory leak/crash only happen when you use parameters for your frame processor? Or does it also happen when you don't have any parameters? cc @FloweryK @suniahk @alexstanbury @chiubaka

alexstanbury, 2021-12-30:
@mrousavy it happens with no parameters. As per my comment above, my repro repo doesn't have any parameters passed to the image labeller plugin 

See https://github.com/alexstanbury/frame-processor-crash/blob/main/App.js

mrousavy, 2021-12-30:
Okay then it must be the return values - I'm having a hard time trying to figure out how to use Android's C++ profilers to find memory leaks but I'll fix it soon. I'm putting a bug bounty on this internally in my Agency.

mrousavy, 2022-01-02:
Please test if #689 fixes the issue for you guys. 🙏

closed 2022-01-02 by mrousavy

chiubaka, 2022-01-03:
Amazing, thank you! Will give this a try sometime this week.

FloweryK, 2022-01-03:
@mrousavy Manual fixing with #689 from version 2.10.0 works perfectly. No crash 👍

chiubaka, 2022-01-11:
This is working awesome for me! Big improvement. Really appreciate the work here.

mrousavy, 2022-01-11:
Thanks Daniel! Glad you appreciate my work