margelo/react-native-vision-camera #588: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.🐛

arn3342, 2021-11-15:
### What were you trying to do?

I have a simple RN project which utilises different libraries, one of them being the `react-native-vision-camera`. I am using the latest version `2.9.3`. While building the project for android (both with `run-android` command or directly from android studio), i am getting a series of errors, whereas it was supposed to build without any issues.

### Reproduceable Code

```tsx
Not exactly sure, but can be reproduced in any blank app.
```

### What happened instead?

When building on Android, with the following gradle settings:

> buildToolsVersion = "30.0.2"
>  minSdkVersion = 21
>  compileSdkVersion = 31
>  targetSdkVersion = 31
>  ndkVersion = "24.0.7856742"

... the build fails with CMAKE errors stating that _some of the variables used by the project are set to NOTFOUND_

### Relevant log output

```shell
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
FOLLY_JSON_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
JSI_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
JS_ENGINE_LIB
 linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
REACT_NATIVE_JNI_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
REANIMATED_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
```

### Device

On Android (Build fails on both M1 + Windows machine)

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mAdutskevich, 2021-11-16:
have react-native: 0.64.0, M1 and the same error with JSI_LIB

JackStudd, 2021-11-17:
Same here with RN 0.64.2

swizes, 2021-11-17:
Same error with RN 0.65.1 and react-native-vision-camera 2.9.3

```
  buildToolsVersion = "30.0.2"
  minSdkVersion = 21
  compileSdkVersion = 30
  targetSdkVersion = 30
  ndkVersion = "20.1.5948944"
  playServicesVersion = "17.0.0" //google maps
  androidMapsUtilsVersion = "2.2.6" //google maps
  //Only using Android Support libraries
  supportLibVersion = "28.0.0"
```

JackStudd, 2021-11-18:
I got it working by reading related issues #388, #338 and #513.
First, I deleted my `node_modules` folder. Then, I ran a `yarn install`.
* I deleted my `node_modules` folder
* I ran `yarn install`
* `cd android`
* `./gradlew clean`
* `./gradlew assembleDebug`
* I was then confronted with the same issue in #513, so I followed this : https://github.com/mrousavy/react-native-vision-camera/issues/513#issuecomment-940857627
* It all worked! Sometimes sync fails in Android Studio, displaying no error, but within a terminal it works just fine.

Sewb21, 2021-12-07:
I am experiencing this also, any updates as to when an official fix may get published?

mrousavy, 2021-12-08:
Let me quickly explain the problems:

## Problem A

JSI_LIB (`libjsi.so`) is only available in React Native 0.65 (or 66?) and higher. VisionCamera relies on that, so either upgrade to React Native 0.66 or higher, or create a PR that adds support for older RN versions.

To add support for older RN versions, remove [this](https://github.com/mrousavy/react-native-vision-camera/blob/a54ff5782cf223cfe4bfff29c636c1dfbb1dc0ec/android/CMakeLists.txt#L90-L95) and [this](https://github.com/mrousavy/react-native-vision-camera/blob/a54ff5782cf223cfe4bfff29c636c1dfbb1dc0ec/android/CMakeLists.txt#L128), and add `jsi.cpp` below [this line](https://github.com/mrousavy/react-native-vision-camera/blob/a54ff5782cf223cfe4bfff29c636c1dfbb1dc0ec/android/CMakeLists.txt#L24). (find the full path by browsing node modules)

## Problem B

Your Gradle Version is 7.0.0 or higher. In that case, we _might_ need to change this line:

https://github.com/mrousavy/react-native-vision-camera/blob/a54ff5782cf223cfe4bfff29c636c1dfbb1dc0ec/android/build.gradle#L469

to something like this:

```gradle
    if (task.name.contains('externalNative') || task.name.contains('CMake')) {
```

but I haven't tested this.

I am rolling VisionCamera in 4 production apps for clients of my app development agency (Margelo), and it works on all 4 apps. So it surely must be the older RN versions, since all of those apps are RN 0.65 and higher. I try to maintain backwards compatibility as good as I can, but I'm a single developer with 5 big open source projects to maintain, CEO at an agency and app developer - I don't have enough time to maintain support for _everything_. 

I would greatly appreciate PRs that fix this, and am also willing to test if a PR works in all 4 of my production apps. If it does work and does not introduce a lot of complexity just for having older RN version support, I am willing to merge it.

willcray, 2021-12-08:
Upgrading from react-native 0.64.2 -> 0.66.3 and following the guide from @JackStudd worked for me!

Sewb21, 2021-12-15:
@mrousavy I am using the package in a project that uses Expo SDK 43 so I cannot upgrade the version of React Native, unfortunately

Sewb21, 2021-12-15:
I am using Expo SDK 43 so upgrading React Native versions isn't an option, unfortunately. The fixes for adding older support doesn't work either, I will keep an eye on the issue. 
Are there any older, stable versions of the package that work with React Native 0.64?

Hirbod, 2021-12-20:
This error is still happening for me on 2.10.0 (Expo SDK 44, Custom Dev Client)

sandlands, 2021-12-24:
@Hirbod I finally managed to build version 2.9.4 for Android using `eas build` (Expo SDK 44, Dev Client). See [this comment](https://github.com/mrousavy/react-native-vision-camera/issues/546#issuecomment-1000849122).

williamOld, 2021-12-27:
I should've commented this earlier. I am on RN `0.66.2` so the assumption that my RN being of an old version is causing this issue is wrong! I did try the workoarounds mentioned in other threads but nothing seem to work properly. Sometimes even if the build succeeds, the application crashes on launch on Android (somehow it works on iOS, but i only tested on emulator)

Sewb21, 2022-01-05:
I still get this error when I refresh Gradle dependencies before building, I am using ejected Expo SDK 43 and React Native Vision v2.11.2

jhoanborges, 2022-03-17:
same issue

faustaleonardo, 2022-05-20:
Same issue here. 
```
"react": "17.0.2",
"react-native": "0.65.1",
"react-native-reanimated": "^2.2.0",
"react-native-vision-camera": "2.10.0",
```

dittmarconsulting, 2022-05-26:
Same issue

```
"react": "^17.0.2",
"react-native": "^0.67.0",
"react-native-reanimated": "^2.8.0",
"react-native-vision-camera": "2.13.1",
```

maurorusso002, 2022-06-06:
Same issue

    "react": "^17.0.2",
    "react-native": "0.68.2",
    "react-native-reanimated": "2.5.0",
    "react-native-vision-camera": "^2.13.3",

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
linked by target "VisionCamera" in directory C:/Users/mauro.a.russo/Documents/projects/react-native/COIMA/node_modules/react-native-vision-camera/android
FOLLY_JSON_LIB
linked by target "VisionCamera" in directory C:/Users/mauro.a.russo/Documents/projects/react-native/COIMA/node_modules/react-native-vision-camera/android
JSI_LIB
linked by target "VisionCamera" in directory C:/Users/mauro.a.russo/Documents/projects/react-native/COIMA/node_modules/react-native-vision-camera/android
JS_ENGINE_LIB
linked by target "VisionCamera" in directory C:/Users/mauro.a.russo/Documents/projects/react-native/COIMA/node_modules/react-native-vision-camera/android
REACT_NATIVE_JNI_LIB
linked by target "VisionCamera" in directory C:/Users/mauro.a.russo/Documents/projects/react-native/COIMA/node_modules/react-native-vision-camera/android
REANIMATED_LIB
linked by target "VisionCamera" in directory C:/Users/mauro.a.russo/Documents/projects/react-native/COIMA/node_modules/react-native-vision-camera/android

zienna, 2022-06-08:
Same issue with RN 0.68.2
Any update @mrousavy ?

dittmarconsulting, 2022-06-10:
Can we please reopen this issue @mrousavy? There are now 8 people with the same issue.

dittmarconsulting, 2022-06-15:
Thank you @mrousavy 🙏

PankajPunia, 2022-06-16:
I was facing the same issue and I was using react-native-vision-camera 2.13.3, I upgraded it 2.13.5 and rebuild the project works good for me.

lundjrl, 2022-06-20:
Can confirm with @PankajPunia. Upgrading to version 2.13.5 fixed this issue for me.

ApWin77, 2022-06-22:
What’s your react-native version @PankajPunia ? 
I did as you but it doesn’t work

PankajPunia, 2022-06-24:
@ApWin77  
"react": "17.0.2",
"react-native": "0.67.4"

tgreco, 2022-07-02:
Getting this issue with 2.13.5 and RN 0.68.1

alexdarh, 2022-07-05:
Getting this issue with 2.13.5 and RN 0.68.1

arthwood, 2022-07-10:
NOTE: This step (and hence the issue) doesn't happen if you [disable frame processors](https://mrousavy.com/react-native-vision-camera/docs/guides/frame-processors/#disabling-frame-processors).

arthwood, 2022-07-10:
A possible solution can be upgrading CMake as per warning that I noticed during gradle build:
> An old version of CMake is being used that cannot automatically detect
> compiler attributes.  Compiler identification is being bypassed.  Some
> values may be wrong or missing.  Update to CMake 3.19 or newer to use
> CMake's built-in compiler identification.

However, Android Studio allows to install CMake 3.18.1 at most.

tgreco, 2022-07-17:
> disable frame processors

The biggest selling point of this library is the frame processor 🙃

twcyao, 2022-08-20:
same issue, i created a new project to reproduce it.

```
  "dependencies": {
    "react": "17.0.2",
    "react-native": "0.68.2",
    "react-native-reanimated": "2.9.1",
    "react-native-vision-camera": "2.14.1"
  },
```

sync error in android studio

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
FOLLY_JSON_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
JSI_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
JS_ENGINE_LIB
 linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
REACT_NATIVE_JNI_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
REANIMATED_LIB
linked by target "VisionCamera" in directory /Users/myuser/Documents/RN Projects/proj-v2/node_modules/react-native-vision-camera/android
```

but i have some finding

do `./gradlew build` from command line seems build VisionCamera successfully, then `run` app in android studio works.

dittmarconsulting, 2022-09-12:
Upgraded `react-native-vision-camera` but still the same error. Why can't that be fixed?
  
  ```
  "dependencies": {
    "react": "17.0.2",
    "react-native": "0.67.0,
    "react-native-reanimated": "2.8.0",
    "react-native-vision-camera": "2.14.1"
  },
```

HorbachAndrii, 2022-09-13:
I fixed it after update 
```
...
buildscript {
    ext {
        buildToolsVersion = "32.0.0"
        compileSdkVersion = 32
        targetSdkVersion = 32
        ....
```

archcorsair, 2022-09-24:
> I fixed it after update
> 
> ```
> ...
> buildscript {
>     ext {
>         buildToolsVersion = "32.0.0"
>         compileSdkVersion = 32
>         targetSdkVersion = 32
>         ....
> ```

This did not work for me.

sheikx1221, 2022-09-24:
Not for vision camera because I am not using it
expo-av and expo-core--modules also have the same issue

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
    linked by target "expo-av" in directory E:/CoolApp/coolconnect-app-driver/node_modules/expo-av/android
JSI_LIB
    linked by target "expo-av" in directory E:/CoolApp/coolconnect-app-driver/node_modules/expo-av/android
REACT_NATIVE_JNI_LIB
    linked by target "expo-av" in directory E:/CoolApp/coolconnect-app-driver/node_modules/expo-av/android

CMake Generate step failed.  Build files cannot be regenerated correctly.
```

I think when we use hermes as js-engine, it excludes these libraries if you see this line
```
    if (enableHermes) {
        //noinspection GradleDynamicVersion
        implementation("com.facebook.react:hermes-engine:+") { // From node_modules
            exclude group:'com.facebook.fbjni'
        }
    } else {
        implementation jscFlavor
    }
```

but if removed this, it throws duplicate libraries error, it means they are loaded but not available or path is not set in CMake.

expo run:android and expo prebuild did not made a CMake.txt file for me

Windows 10
Node 16.17.0
React-Native 0.69.5
Expo 46
expo-av 12.0.4

HorbachAndrii, 2022-09-26:
> > I fixed it after update
> > ```
> > ...
> > buildscript {
> >     ext {
> >         buildToolsVersion = "32.0.0"
> >         compileSdkVersion = 32
> >         targetSdkVersion = 32
> >         ....
> > ```
> 
> This did not work for me.
> 
> **Edit:** This worked for me (I believe the order is important)
> 
> 1. Run `./gradlew clean` in the `/android` directory
> 2. Nuke your `node_modules` and install deps fresh again
> 3. `npx react-native run-android	`

What is your ndkVersion ?
I use next 
```
if (System.properties['os.arch'] == "aarch64") {
    // For M1 Users we need to use the NDK 24 which added support for aarch64
    ndkVersion = "24.0.8215888"
} else {
    // Otherwise we default to the side-by-side NDK version from AGP.
    ndkVersion = "21.4.7075529"
}
```

namti, 2022-10-04:
"File" -> "Invalidate Caches..." works

hengkx, 2022-10-20:
0.70.3 same error

alfredarcifa, 2022-10-25:
> 0.70.3 same error

Me too, on the .70.3

```
CMake Warning (dev) in CMakeLists.txt:
No project() command is present.  The top-level CMakeLists.txt file must
contain a literal, direct call to the project() command.  Add a line of
code such as....

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
FBJNI_LIB
linked by target "VisionCamera" in directory /Users/user/Projects/bennet/node_modules/react-native-vision-camera/android
FOLLY_LIB
linked by target "VisionCamera" in directory /Users/user/Projects/bennet/node_modules/react-native-vision-camera/android
JSI_LIB
linked by target "VisionCamera" in directory /Users/user/Projects/bennet/node_modules/react-native-vision-camera/android
REACT_NATIVE_JNI_LIB
linked by target "VisionCamera" in directory /Users/user/Projects/bennet/node_modules/react-native-vision-camera/android
REANIMATED_LIB
linked by target "VisionCamera" in directory /Users/user/Projects/bennet/node_modules/react-native-vision-camera/android
```

Sewb21, 2022-11-03:
Still experiencing this vision 2.15.1, react native 0.69.5

prateekjain98, 2022-11-05:
> Not for vision camera because I am not using it expo-av and expo-core--modules also have the same issue
>
> ```
> CMake Error: The following variables are used in this project, but they are set to [...]

I am also facing issue in expo. Did you get any solution for it?

andreasheick-isobar, 2022-11-07:
Using version 2.15.2, react native 0.70.5 and still getting this issue. 

On react native 0.70.4 it worked with these steps:
- cd to the android folder and run `./gradlew clean`
- nuke node_modules and reinstall
- Android Studio -> Build -> Clean Project
- Android Studio -> File -> Invalidate Caches / Restart

But now I can't fix it with this method anymore

tgreco, 2022-12-06:
Using the example app out of the box I am running into this issue.

martin-gudmundson, 2023-01-02:
Updating Android Studio's CMake version to 3.22.1 via the SDK Manager and then clean and rebuild resolved the issue for me.

2sem, 2023-01-27:
Same here
0.63.4 -> 0.63.5
gradle plugin 4.1.2 -> 4.2.2
gradle 6.5 -> 6.7.1

bohdan-sharkz, 2023-02-03:
> Using version 2.15.2, react native 0.70.5 and still getting this issue.
> 
> On react native 0.70.4 it worked with these steps:
> 
> * cd to the android folder and run `./gradlew clean`
> * nuke node_modules and reinstall
> * Android Studio -> Build -> Clean Project
> * Android Studio -> File -> Invalidate Caches / Restart
> 
> But now I can't fix it with this method anymore.
> 
> EDIT: I managed to get rid of the error by running `./gradlew assembleDebug` and then could run the app in Android Studio and sync gradle once again.

I'm surprised this comment doesn't have any thumb-ups as only `./gradlew assembleDebug` solved my issue and i've tried all of the above comments solutions

akinlekan28, 2023-02-21:
> Updating Android Studio's CMake version to 3.22.1 via the SDK Manager and then clean and rebuild resolved the issue for me.

I can confirm this worked for me

closed 2023-08-18 by mrousavy

TassainRasool, 2024-04-29:
react-native version =0.68.5 
I have updated the react-native-reanimated: 2.17.0 to react-native-reanimated: 3.0.0-rc.0 now my project build is getting failed I tried so many solutions none of them worked .
help me in this

flyskywhy, 2024-08-12:
[Add support for react-native 0.71 to fix `CMake Error: The following …](https://github.com/flyskywhy/playtorch/commit/ce55e974dc4e29622039b04ecf6efc62ab205c6b)