margelo/react-native-vision-camera #594: ✨ Jest Mock

pke, 2021-11-18:
### What feature or enhancement are you suggesting?

Would be nice to be able to write jest tests that involve the camera component and its hooks.
react-native-camera brought its own mock (which was quite strange implemented) but I think a simple mock that is just stubbing everything as `jest.fn` could be placed inside `jest/mock.ts`

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

Workaround would be that every developer writes her own mock for this library.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

Alaa-Ben, 2023-02-17:
Any news about this ?
The mocking section shows how to make a mock for simulators.
What about unit testing only ?

FranciscoPDNeto, 2023-07-06:
Hello everyone, any updates about this?

mrousavy, 2023-09-30:
Already implemented in V3

closed 2023-09-30 by mrousavy