margelo/react-native-vision-camera #595: 🐛 Android phones not reporting all existing cameras

viljark, 2021-11-18:
### What were you trying to do?

I want to use ultrawide and telephoto cameras that exist on my [Samsung Galaxy A12](https://www.gsmarena.com/samsung_galaxy_a12-10604.php) (it has quad camera setup) and [Oneplus 7T](https://www.gsmarena.com/oneplus_7t-9816.php) (triple camera setup)

### Reproduceable Code

```tsx
useEffect(() => {
		Camera.getAvailableCameraDevices().then((devices) => {
			console.log(JSON.stringify(devices.map((d) => ({ [d.position]: d.devices }))))
		})

	}, [devices])
```

### What happened instead?

`Camera.getAvailableCameraDevices()` for Samsung Galaxy A12 reports it only has the following cameras:
`[{"back":["wide-angle-camera"]},{"front":["wide-angle-camera"]},{"front":["wide-angle-camera"]}]`

A12 back cameras:
48 MP, f/2.0, 26mm (wide), 
AF5 MP, f/2.2, 123˚ (ultrawide)
2 MP, f/2.4, (macro)
2 MP, f/2.4, (depth)

Oneplus 7T also only reports single back wide-angle-camera, while it has these back cameras:
48 MP, f/1.6, 26mm (wide), 1/2.0", 0.8µm, PDAF, OIS
12 MP, f/2.2, 51mm (telephoto), 1.0µm, PDAF, 2x optical zoom
16 MP, f/2.2, 17mm (ultrawide), AF

### Device

Samsung Galaxy A12, Oneplus 7T

### VisionCamera Version

2.9.4-expo.11

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-11-22:
Hi!

That's not yet supported by the underlying native Android CameraX library, I've created an issue in their issue tracker and they're working on it.

Yennnful, 2022-01-29:
Hi, any update on this issue?

matiaspan, 2022-03-03:
@mrousavy do you have a link to the issue you reported on CameraX? would like to give this more visibility

cristianoccazinsp, 2022-03-04:
I'm curious whether or not this has been resolved. I'm observing similar issues with the ancient react-native-camera on Samsung Galaxy S22 and this may be a good excuse to migrate libraries.

viljark, 2022-03-04:
> I'm curious whether or not this has been resolved. I'm observing similar issues with the ancient react-native-camera on Samsung Galaxy S22 and this may be a good excuse to migrate libraries.

I think this is not resolvedm, modern multi camera setups still report only single back camera.
But when zooming in and out, the camera is switched to zoom / wide angle automatically on my Pixel 6 pro, so it still can use the cameras. I'm not sure if there is a way to trigger camera switching by setting the zoom values manually.

cristianoccazinsp, 2022-03-04:
So how do you know the camera supports an ultra wide / or telephoto camera on android? If the cameras are not reported, does that mean you need to rely on min/max zoom values and the user zooming out enough to trigger the camera? I haven’t read through the docs so apologies if that’s explained there.

Does the library handle native zoom and camera switching automatically? Including tap to focus? As I’ve seen issues with native zoom support blocking other tap actions to, for example, implement tap to focus.

mrousavy, 2022-03-07:
> But when zooming in and out, the camera is switched to zoom / wide angle automatically on my Pixel 6 pro

Wait, really? This means CameraX does have multi-cam support after all?

As far as I know, they don't. And there's an open issue somewhere in the google issue tracker about this, don't have a link right now

viljark, 2022-03-07:
> Wait, really? This means CameraX does have multi-cam support after all?
>
> As far as I know, they don't. And there's an open issue somewhere in the google issue tracker about this, don't have a link [...]

Seems like it works on Pixel 6 Pro (Android 12)
https://photos.app.goo.gl/bhBRg9kWx1F7wXyr7

mrousavy, 2022-03-07:
Haha awesome, maybe they do it fully on hardware level.

rfischer20, 2022-04-14:
Is there a way to detect if it's camerax. We're trying to work on a solution to not set the devices when the device uses android camerax.

mrousavy, 2022-04-15:
What? If you don't use CameraX you don't have a camera at all?

jokertrip, 2022-07-21:
Hi, any update on this issue?

awcchungster, 2022-07-28:
I also have this problem, and it's very frustrating Android manufactures don't improve their software to support developers.

awcchungster, 2022-07-28:
> Hi!
> 
> That's not yet supported by the underlying native Android CameraX library, I've created an issue in their issue tracker and they're working on it.

Could you please link us the ticket?

davidlcorbitt, 2022-07-28:
Seeing the ticket would be super helpful for my team @mrousavy.  If possible, we can try to fork the repo and submit a PR with our best work around?

cloudmu, 2022-11-07:
> Seems like it works on Pixel 6 Pro (Android 12)

I can confirm that when zooming out, the camera cannot zoom out all the way on S20 and Note 10.

nikonhub, 2022-11-18:
New user here.

So the google ticket is [here](https://issuetracker.google.com/issues/179925896)
It's commented under the [getAvailableCameraDevices](https://github.com/mrousavy/react-native-vision-camera/blob/a65b8720bd7f2efffc5fb9061cc1e5ca5904bd27/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L168) method.

Would be helpful to know if someone managed to get the video mode work fluidly ?
I get 4 physical cameras on android. None of them get a good result, despite the stock camera app working perfectly fine.

Not sure if this issue even would fix my problem. But wouldn't it be possible to get a logical camera for back camera instead of physical ? Something more automatic, and working just out of the box.

Today letting useCameraDevices hook choose the camera for us, results in a 5 fps camera, unusable.

If on android side, to use the camera is not just plug & play, I'd be glad to get more documentation on this.

wibes, 2023-04-14:
I am facing similar issue, not able to get ultra wide angle camera. Any update on fix ?

Tiropz, 2023-07-26:
I'm facing the same issue, the only camera available on android is the "wide-angle-camera". Couldn't find a workaround

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy

mdottavi, 2023-09-19:
Hi, we are trying to integrate v3 in our project...
We tested the example app on several Android phones and found the following issues:

- OnePlus 10 Pro  (Android 13):
   starts but shows Front Cam + 1 Back wide-angle camera 
   but the phone includes 3 Back cameras (including 1 ultra-wide-angle- camera 0.6 zoom)
- OnePlus Nord2  (Android 13):
   starts but shows Front Cam + 1 Back wide-angle camera 
   but the phone includes 3 Back cameras (including 1 ultra-wide-angle- camera 0.6 zoom)
- Samsung S10+ (Android 12):
   detects Front Cam + 3 Back cameras ( 2 wide-angle camera  + 1 ultra-wide-angle camera)
   but throws an error when displaying camera frames:
    ERROR  [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR]
- Motoral One Macro (Android 10)
  detects shows Front Cam + 1 Back wide-angle camera 
  but throws an error when displaying camera frames:
 ERROR  [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: DISCONNECTED]

mrousavy, 2023-09-19:
@mdottavi can you put the output of this here?

```js
const devices = await Camera.getAvailableCameraDevices()
for (const device of devices) {
  device.formats = null // just remove the formats for logging
}
console.log(JSON.stringify(devices, null, 2))
```

Thanks! 🙏

mrousavy, 2023-09-19:
Regarding Samsung S10+ and Motoral One Macro, maybe this also has something to do with the custom format. There have been multiple reports on that.

Can you please try removing those 24 lines https://github.com/mrousavy/react-native-vision-camera/blob/18c7034b5006986a7df6da9bf799dcc745244932/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L436-L460
and see if the camera then starts?
If it does, try adding feature by feature back and see which one causes it. My guess is either fps or video stabilization, those manufacturers tend to be weird/off-platform for those things 😅

mrousavy, 2023-09-19:
Also, if you want you can [join our community discord](https://discord.gg/6CSHz2qAvA) to talk about this there (i sometimes miss github notificaitons, I get too many)

okalil, 2024-10-28:
I'm experiencing a similar issue in v4 with a Motorola device. Since it's a different library version, I opened a new issue #3262