margelo/react-native-vision-camera #601: Setting videoStabilizationMode='auto' freezes the video output on iPhone11+🐛

Adam-Beno, 2021-11-20:
### What were you trying to do?

Tried to use auto stabilization when recording videos and expected the output video to be stabilized and not cut in the middle.

### Reproduceable Code

```tsx
<Camera
  ref={cameraRef}
  device={device}
  preset={preset}
  fps={fps}
  hdr={enableHdr}
  lowLightBoost={device?.supportsLowLightBoost && enableNightMode}
  isActive={isActive}
  onInitialized={onInitialized}
  onError={onError}
  enableZoomGesture={false}
  photo={true}
  video={true}
  audio={true}
  videoStabilizationMode="auto" // Omiting this prop fixes it
/>
```

### What happened instead?

Video freezes  by the end of the recording on iPhone11,12,13. On any other devices it doesn't do that.

Here is a video where you can see the freeze:
[https://streamable.com/kx8826](https://streamable.com/kx8826)

### Device

iPhone 11

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-11-21:
That's interesting, I don't think I've encountered this before. Maybe the video recording stops correctly, but the audio continues to record?

Adam-Beno, 2021-11-21:
To be honest this took me a while to figure out at all, because it doesn't occur on **any** other device but the iPhone 11 and higher and we had the stabilization enabled since day one...

The camera component is mostly copy paste of what you have in the example app with some changes to it. I thought maybe that's where the problem is because I was doing the capture in the parent component instead of the recording button like you do.

So my thinking was, maybe I'm somehow losing the reference to the camera when the parent component re-renders, so I moved all recording stuff into the button component and logged re-renders in both components. The results were that the camera doesn't rerender at all once I start recording so there was no way I could mess up the reference obj.

There doesn't seem to be an issue with how the result is delivered either. I press the button to stop recording and the callback function gets called to deliver the file...

Eventually I just started removing props that were being passed to the camera component itself and then finally i found out that the autostabilization is at fault here.

bsord, 2022-12-07:
Sorry to revive an old thread but recently ran into this and only noticed it after turning the stabilization feature on. When any stabilization mode is defined it will add some half second to 1 second of presumably empty frames. the video appears frozen. This is also in the resulting recorded file. I also went through the trouble of making sure there was no state changes during and after capture is complete. Did you ever find a fix for this?

bsord, 2022-12-07:
Last update in case anyone else finds this issue, looks like react-native-vision-camera is using avfoundation under the hood and basically just passing a property down, not doing a lot directly with stabilization so I started googling issues similar to this for AVFoundation and came across [stack overflow article](https://stackoverflow.com/questions/55190815/avcapturesession-video-stabilization-lag) and it mentions that changing stabilization mode from cinematic to standard greatly reduced the impact. When I changed stabilization mode to standard in react-native-vision-camera, sure enough the issue basically went away. For now this is my work around but would be cool if there was a proper fix for the other cinematic modes.

mrousavy, 2022-12-12:
Hey! This is really weird.

I am currently a bit flooded with work so I can't really look into this atm (unless you wanna pay us to work on that, I can put a team member onto that issue), but here's some pointers if you wanna look into this yourself (I'd really appreciate a PR!)

Here's where I get the frame and create the timestamp:
https://github.com/mrousavy/react-native-vision-camera/blob/ae2c28a2c27a2ef6afa276b1b2d2034d2651027c/ios/CameraView%2BRecordVideo.swift#L200-L211

Here's where I write it to the actual buffer:
https://github.com/mrousavy/react-native-vision-camera/blob/ae2c28a2c27a2ef6afa276b1b2d2034d2651027c/ios/RecordingSession.swift#L129-L187

nomuppets, 2023-04-23:
So you can't use video stabilisation in IOS using this component?

mrousavy, 2023-04-25:
You can. I remember it worked for me previously. Maybe there was a regression somewhere.

mrousavy, 2023-09-06:
Does this issue still exist on VisionCamera V3? If yes, we can reopen this issue, otherwise let's mark it as finished

closed 2023-09-06 by mrousavy

spenweb, 2023-10-31:
It still exists in V3. I don't have this issue when I only use standard stabilization. I followed this advice:

> Last update in case anyone else finds this issue, looks like react-native-vision-camera is using avfoundation under the hood and basically just passing a property down, not doing a lot directly with [...]

spenweb, 2023-11-15:
@mrousavy,  I can confirm that this error is still happening in the latest version (v3.6.6).

I just pulled from the main branch and ran the example app after commenting out the `frameProcessor` and adding `videoStabilizationMode={'cinematic-extended'}` to the `ReanimatedCamera` Component.

I notice this error when I attempt to record a short video (about 1 second): 
> The capture has been stopped before any input data arrived.

Here are the logs for this:
```
Camera initialized!
state: BEGAN
calling startRecording()...
called startRecording()!
VisionCamera.startRecording(options:onVideoRecorded:onError:): Starting Video recording...
VisionCamera.startRecording(options:onVideoRecorded:onError:): File path: /private/var/mobile/Containers/Data/Application/9C442358-4876-4200-9BF1-D72D9E985492/tmp/ReactNative/6FDD366D-CB3C-4A37-B74E-F58D69665847.mov
VisionCamera.startRecording(options:onVideoRecorded:onError:): Recommended Video Settings: ["AVVideoCompressionPropertiesKey": {
    AllowFrameReordering = 1;
    AllowOpenGOP = 1;
    AverageBitRate = 47848572;
    BaseLayerFrameRate = 15;
    ExpectedFrameRate = 60;
    MaxAllowedFrameQP = 41;
    MaxKeyFrameIntervalDuration = 1;
    MinAllowedFrameQP = 15;
    MinimizeMemoryUsage = 1;
    Priority = 80;
    ProfileLevel = "HEVC_Main_AutoLevel";
    RealTime = 1;
    RelaxAverageBitRateTarget = 1;
}, "AVVideoWidthKey": 2160, "AVVideoCodecKey": hvc1, "AVVideoHeightKey": 3840]
VisionCamera.initializeVideoWriter(withSettings:pixelFormat:): Initialized Video AssetWriter.
VisionCamera.activateAudioSession(): Activating Audio Session...
VisionCamera.updateCategory(_:options:): Changing AVAudioSession category from AVAudioSessionCategorySoloAmbient -> AVAudioSessionCategoryPlayAndRecord
<0x123108de0> Gesture: System gesture gate timed out.
state: ACTIVE
state: END
calling stopRecording()...
VisionCamera.initializeAudioWriter(withSettings:): Initialized Audio AssetWriter.
VisionCamera.startAssetWriter(): Starting Asset Writer(s)...
VisionCamera.finish(): Finishing Recording with AssetWriter status "writing"...
VisionCamera.startRecording(options:onVideoRecorded:onError:): RecordingSession finished with status failed.
VisionCamera.startRecording(options:onVideoRecorded:onError:): RecordingSession Error 1: Error Domain=capture/aborted Code=1 "Stopped Recording Session too early, no frames have been recorded!" UserInfo={NSLocalizedDescription=Stopped Recording Session too early, no frames have been recorded!}
VisionCamera.deactivateAudioSession(): Deactivating Audio Session...
VisionCamera.deinit: Cancelling AssetWriter...
'Recording failed!', { code: 'capture/aborted',
  cause: null,
  message: 'capture/aborted: The capture has been stopped before any input data arrived.' }
stopped recording video!
called stopRecording()!
 ```
 
Frames are being dropped for about a second at the end of the video (while audio is continued to be heard). Here are the logs for that (nothing stands out to me, but maybe for you):
 ```
 state: BEGAN
calling startRecording()...
called startRecording()!
VisionCamera.startRecording(options:onVideoRecorded:onError:): Starting Video recording...
VisionCamera.startRecording(options:onVideoRecorded:onError:): File path: /private/var/mobile/Containers/Data/Application/9C442358-4876-4200-9BF1-D72D9E985492/tmp/ReactNative/FCD14160-4B81-4296-BEBA-4FF561884E7E.mov
VisionCamera.startRecording(options:onVideoRecorded:onError:): Recommended Video Settings: ["AVVideoHeightKey": 3840, "AVVideoCodecKey": hvc1, "AVVideoWidthKey": 2160, "AVVideoCompressionPropertiesKey": {
    AllowFrameReordering = 1;
    AllowOpenGOP = 1;
    AverageBitRate = 47848572;
    BaseLayerFrameRate = 15;
    ExpectedFrameRate = 60;
    MaxAllowedFrameQP = 41;
    MaxKeyFrameIntervalDuration = 1;
    MinAllowedFrameQP = 15;
    MinimizeMemoryUsage = 1;
    Priority = 80;
    ProfileLevel = "HEVC_Main_AutoLevel";
    RealTime = 1;
    RelaxAverageBitRateTarget = 1;
}]
VisionCamera.initializeVideoWriter(withSettings:pixelFormat:): Initialized Video AssetWriter.
VisionCamera.activateAudioSession(): Activating Audio Session...
VisionCamera.updateCategory(_:options:): Changing AVAudioSession category from AVAudioSessionCategoryPlayback -> AVAudioSessionCategoryPlayAndRecord
VisionCamera.initializeAudioWriter(withSettings:): Initialized Audio AssetWriter.
VisionCamera.startAssetWriter(): Starting Asset Writer(s)...
VisionCamera.appendBuffer(_:type:timestamp:): Started RecordingSession at 590561.304804875 seconds.
<0x123108de0> Gesture: System gesture gate timed out.
state: ACTIVE
state: END
calling stopRecording()...
VisionCamera.finish(): Finishing Recording with AssetWriter status "writing"...
called stopRecording()!
VisionCamera.startRecording(options:onVideoRecorded:onError:): RecordingSession finished with status completed.
VisionCamera.deactivateAudioSession(): Deactivating Audio Session...
Recording successfully finished! file:///private/var/mobile/Containers/Data/Application/9C442358-4876-4200-9BF1-D72D9E985492/tmp/ReactNative/FCD14160-4B81-4296-BEBA-4FF561884E7E.mov
Media captured! {"path":"file:///private/var/mobile/Containers/Data/Application/9C442358-4876-4200-9BF1-D72D9E985492/tmp/ReactNative/FCD14160-4B81-4296-BEBA-4FF561884E7E.mov","duration":4.28346996}
stopped recording video!
VisionCamera.didSetProps(_:): Updating 1 props: [isActive]
VisionCamera.configure(_:): Updating Session Configuration...
Video loaded. Size: 2160x3840 (landscape, 4.304852485656738 seconds)
media has loaded.
VisionCamera.withDeviceLock(_:): Beginning CaptureDevice configuration...
VisionCamera.withDeviceLock(_:): Committed CaptureDevice configuration!
```