margelo/react-native-vision-camera #608: 🐛 useCameraDevices and Camera.getAvailableCameraDevices issues

SeamusY, 2021-11-23:
### What were you trying to do?

I have attempted both available methods:
`const devices = useCameraDevices('wide-angle-camera');` returns 

{"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}

phoneDevice returns an entire object, which with all the expected things that I wanted. However, upon passing this to the following:
```
<ReanimatedCamera 
                        ref={camera}
                        photo
                        device={phoneDevice}
                        isActive
                        style={{width: "100%", height: "80%"}}
                    />
```
It would break, I have also tried using standard "Camera" but it would fail and say the following:
`Error: Objects are not valid as a React child (found: object with keys {_U, _V, _W, _X}). If you meant to render a collection of children, use an array instead.`

### Reproduceable Code

```tsx
import { ConsoleLogger } from "@aws-amplify/core";
import { Center, Container, NativeBaseProvider, Image, Button, Text } from "native-base";
import React, { useState, useRef, useEffect } from "react";
import { StyleSheet, TouchableOpacity } from "react-native";
import Reanimated from 'react-native-reanimated';
import { Navigation } from "react-native-navigation";
import { Camera, useCameraDevices } from "react-native-vision-camera";

export const Scanner = (props) => {
    const camera = useRef(null)
    const [data, setData] = useState({})
    const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera);
    // const devices = Camera.getAvailableCameraDevices()
    const devices = useCameraDevices('wide-angle-camera');
    const device = devices.back
    // const deviceList = devices.filter((device) => device.position === "back")
    // const phoneDevice = [...deviceList].shift()
    console.log(devices);
    const handleOnPress = async () => {
        const photo = await camera.current.takePhoto()
        setData(photo)
        console.log(data)
      }

    const styles = StyleSheet.create({
        button: {
            alignSelf: "center",
            alignItems: "center",
            borderColor: "red",
            borderWidth: 2,
            marginTop: 16,
            height: 32,
            flex: 1
        },
        buttonText: {
            backgroundColor: "rgba(245, 252, 255, 0.7)",
            fontSize: 32,
        }
    })

    return (
        <NativeBaseProvider>
            <Center flex={1}>
            { device !== null ?
                <Container width={"100%"} height={"90%"} alignContent={"center"}>
                    <ReanimatedCamera 
                        ref={camera}
[... 61 lines cut ...]
    in AppContainer (at renderApplication.js:41)
    in Scanner(RootComponent) (at renderApplication.js:57)
 LOG  Running "LoginScreen" with {"rootTag":151,"initialProps":{"componentId":"Component1"}}
 LOG  Error happened
 LOG  Error happened

useCameraDevices('wide-angle-camera'):
LOG  {"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}
 LOG  {"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}
```

### Device

Iphone Xr, 15.1

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jpike88, 2022-01-04:
wow this just stopped me in my tracks. same exact issue. any luck?

mrousavy, 2022-01-04:
You have to use one of those returned devices.

```tsx
const devices = useCameraDevices()
const device = devices.back

return (<Camera device={device} />)
```

danielpgross, 2022-01-04:
I had the same issue, it was caused by the `Camera` component mounting before `device` had initialized (i.e. it was still `undefined`).

In your code, you could try changing:
```
            { device !== null ?
                <Container width={"100%"} height={"90%"} alignContent={"center"}>
```
to
```
            { device &&
                <Container width={"100%"} height={"90%"} alignContent={"center"}>
```

This way, you'll handle the case where device is `undefined` rather than `null`.

mrousavy, 2022-01-04:
`device != null` is better than `device !== null` or just `device` (in terms of readability and safety)

faizantariq1, 2022-01-06:
Having same issue.
After using 
const devices = useCameraDevices()
returns

{"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}

getting undefined.

mrousavy, 2022-01-07:
The function might still be loading - is it also undefined after a few seconds?

faizantariq1, 2022-01-07:
Yes it's still undefined after a few seconds.

ldxoxo, 2022-01-10:
Same problem . Undefined forever.

  ```
const devices = useCameraDevices()
  const device = devices.back

return (
 {device && <Camera
      style={{height:"100%", width:"100%"}}
      device={device}
      isActive={true}
    />}
)
```

"react": "17.0.2",
    "react-native": "0.66.4",
    "react-native-gesture-handler": "^2.1.0",
    "react-native-reanimated": "^2.3.1",
    "react-native-vision-camera": "^2.11.2"

mrousavy, 2022-01-10:
On which device?

faizantariq1, 2022-01-10:
i'm using infinix hot 11.

SeamusY, 2022-01-10:
> On which phone?

As the issue mentioned, Iphone XR for me Ios 15.1. Ill attempt another round of debugging today, ill get back to you about the potential solutions on github.

Let me know your thoughts, ill help debug/ get some directions today.

mrousavy, 2022-01-10:
Hm, weird. I'd like to see what the native code actually does, it surely must have some devices. Could you put a breakpoint in Xcode here: https://github.com/mrousavy/react-native-vision-camera/blob/233094d18e179b534bb15fa3418830b535694267/ios/CameraViewManager.swift#L106

and see what `devices` is?

SeamusY, 2022-01-10:
> You have to use one of those returned devices.
> 
> ```tsx
> const devices = useCameraDevices()
> const device = devices.back
> 
> return (<Camera device={device} />)
> ```

This works now, I just testing the binding of the camera photos right now.

faizantariq1, 2022-01-11:
@SeamusY did you check on android device?

SeamusY, 2022-01-11:
Couldn’t get it running currently, as for some reason android studio is showing some configuration issues after installing back the react native vision camera. Will test it soon, tomorrow after i fix it. But if you have any information on “debugManifest” issue let me know.

faizantariq1, 2022-01-11:
i don't know the exact problem you're facing.
but this link might help you: https://stackoverflow.com/questions/2409923/what-do-i-have-to-add-to-the-manifest-to-debug-an-android-application-on-an-actu

SeamusY, 2022-01-11:
@faizantariq1 

VisionCamera: Building with JSC...
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
VisionCamera: Frame Processors are disabled because REA v2 does not exist. C++ part will not be built.
WARNING:: Please remove usages of `jcenter()` Maven repository from your build scripts and migrate your build to other Maven repositories.
This repository is deprecated and it will be shut down in the future.
See http://developer.android.com/r/tools/jcenter-end-of-service for more information.
Currently detected usages in: project ':amazon-cognito-identity-js', project ':react-native-async-storage_async-storage', project ':react-native-camera', ...
WARNING:: The specified Android SDK Build Tools version (29.0.3) is ignored, as it is below the minimum supported version (30.0.2) for Android Gradle Plugin 4.2.1.
Android SDK Build Tools 30.0.2 will be used.
To suppress this warning, remove "buildToolsVersion '29.0.3'" from your build.gradle file, as each version of the Android Gradle Plugin now has a default version of the build tools.

> Task :app:processDebugMainManifest FAILED

SeamusY, 2022-01-11:
@faizantariq1 yeah, I just tested it on android seems to be working fine. Samsung Galaxy Note 5 with Android 7

export const ScannerVision = (props) => {
    const camera = useRef(null)
    const [data, setData] = useState({})
    const [allowedCamera, setAllowedCamera] = useState(false)
    const [cameraPermission, setCameraPermission] = useState("");

    const devices = useCameraDevices();
    const device = devices.back

    return (
        <NativeBaseProvider>
            <Center flex={1} px="0.5" width={"100%"} padding={16}>
                { (cameraPermission && device) ?
                    <Container style={{ width: "100%" }}>
                        <Camera
                            ref={camera}
                            photo
                            isActive
                            device={device}
                            style={{ height: "70%", width: "100%" }}
                        />
                        <Button onPress={handleOnPress}>Take Photo</Button>
                    </Container> 
                    : <Text>You must accept camera permission</Text>
                }
            </Center>
        </NativeBaseProvider>
    )
}

faizantariq1, 2022-01-11:
@SeamusY related to warnings. basically, jcenter() is going to depreciate in the future. that's why they want it to remove. May be they are using jcenter() in the library.

and related to the code of camera. I'm already taking the permission of the camera. I'll try again and see if it's working or not.

SeamusY, 2022-01-12:
@faizantariq1 if you can, try to debugging this file "CameraView.h" it seems to be the one that returns to you the devices.

I did however, find that the android for some reason defaulted to a permission as denied, not sure if it's a setting or something. But just be sure to check, it might be the reason for blocking.... but it is a guess work on my end.

Heyrio, 2022-01-27:
running ios 15.2 not sure what more to do.
[image]

ashrazdev, 2022-11-30:
useCameraDevices() 

return 

{
   "back": "undefined",
   "external": "undefined",
   "front": "undefined",
   "unspecified": "undefined"
} 

on some devices especially on 
Samsung A32, infinix X657B  both have android version 11,

current config:

    "react-native": "0.69.0",
    "react-native-vision-camera": "^2.15.2",

Please help am stuck a whole day

venura-atlasopen, 2023-01-05:
@SeamusY @ashrazdev did you find a solution? Thanks in advance

jp-23, 2023-01-05:
@mrousavy I am currently seeing this issue.  Can we re-open this issue?  Or I can create a new one if necessary.

I just tried implementing using the below example from the guide, and `device` is always **undefined** and `devices` always looks like this:
```
{"back": undefined, "external": undefined, "front": undefined, "unspecified": undefined}
```

```
function App() {
  const devices = useCameraDevices()
  const device = devices.back

  if (device == null) return <LoadingView />
  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  )
}
```

I've also tried this with `getAvailableCameraDevices` method and it always returns an empty array (`[]`).
```
const devices = await Camera.getAvailableCameraDevices()
```

Here is my setup:
- iPhone simulator (iPhone 14 Pro) with iOS 16.0 on mac 12.6.1
- react-native 0.70.3
- react-native-vision-camera 2.15.2

mrousavy, 2023-01-06:
VisionCamera does not work on an iOS simulator - apple limitation.

criticalmiind, 2023-02-22:
try this code:

```js
import React, { useState, useRef } from 'react';
import { StyleSheet, Text, View, TouchableOpacity, Button } from 'react-native';
import { Camera, CameraRecordingQuality, useCameraDevices } from 'react-native-vision-camera';
// import * as RNVisionCamera from 'react-native-vision-camera';

const VideoCamRecorder = () => {
    const [isRecording, setIsRecording] = useState(false);
    const cameraRef = useRef(null);

    const startRecording = async () => {
        console.log("1234567")
        setIsRecording(true);
        try {
            const options = {
                quality: CameraRecordingQuality['720p'],
                videoBitrate: 12000000, // 12Mbps
                audioBitrate: 256000, // 256Kbps
                fps: 30,
                codec: 'H264',
                enableAudio: true,
            };
            const recording = await cameraRef.current?.startRecording(options);
            console.log('Recording started');
        } catch (error) {
            console.warn(error);
        }
    };

    const stopRecording = async () => {
        setIsRecording(false);
        try {
            const recording = await cameraRef.current?.stopRecording();
            console.log('Recording stopped', recording?.uri);
        } catch (error) {
            console.warn(error);
        }
    };

    const devices = useCameraDevices('wide-angle-camera')
    const device = devices.back
    // const device = devices.front

    return (
        <View style={styles.container}>
            { device && <Camera
                ref={cameraRef}
                device={device}
                isActive={true}
                // style={styles.preview}
                style={StyleSheet.absoluteFill}
            /> }
            {/* <Camera style={styles.preview} /> */}
            <View style={styles.controlsContainer}>
                {isRecording ? (
                    <TouchableOpacity
                        style={styles.button}
                        onPress={stopRecording}>
                        <Text style={styles.buttonText}>Stop Recording</Text>
                    </TouchableOpacity>
                ) : (
                    <TouchableOpacity
                        style={styles.button}
                        // onPress={startRecording}
                        onPress={() => {
                            console.log("ertyui")
                        }}>
                        <Text style={styles.buttonText}>Start Recording</Text>
                    </TouchableOpacity>
                )}
            </View>
        </View>
    );
};

const styles = StyleSheet.create({
    container: {
        flex: 1,
        backgroundColor: '#000',
    },
    preview: {
        flex: 1,
        justifyContent: 'flex-end',
        alignItems: 'center',
    },
    controlsContainer: {
        flexDirection: 'row',
        alignItems: 'center',
        justifyContent: 'center',
        padding: 20,
        zIndex: 1000,
        position: 'absolute',
        bottom:20,
        width:'100%'
    },
    button: {
        backgroundColor: '#fff',
        borderRadius: 50,
        paddingVertical: 15,
        paddingHorizontal: 25,
        alignItems: 'center',
        justifyContent: 'center'
    },
    buttonText: {
        fontWeight: 'bold',
        fontSize: 18,
    },
});

export default VideoCamRecorder;

```

FatjonRrapaj, 2023-04-05:
set isActive={true} in Camera props
` <Camera
          isActive={true}
          device={device}
          style={{flex: 1}}
        />`

Mihir-Gajera1, 2023-08-17:
Getting the same error. please help.

Charles-Johnson, 2023-08-23:
People are suggesting
```js
const devices = useCameraDevices('wide-angle-camera')
const device = devices.back
```

but that won't work on all devices e.g. Samsung Galaxy A14 only has an ultra-wide angle camera at the back

We're resorting to this code
```js
const devices = useCameraDevices("wide-angle-camera");
const fallbackDevices = useCameraDevices("telephoto-camera");
const ultraWide = useCameraDevices("ultra-wide-angle-camera");
const device = devices.back || fallbackDevices.back || ultraWide.back;
```
to avoid device specific issues

mrousavy, 2023-08-24:
@Charles-Johnson two things:

1. You can also just use `useCameraDevices()` without any arguments, then do the filtering yourself.
2. It seems like a bug that Samsung Galaxy A14 only has an ultra wide on the back. Can you confirm this happens even with VisionCamera V3 RC? What's the focal lengths there?

SameerAther, 2023-10-05:
@mrousavy I can confirm that i am getting same error in v3.1.0. I am getting two errors
First One: Cannot read property 'CameraDevices' of undefined
Second One: Cannot read property 'getConstants' of null

This is my code:

`export const VideoCall = ({route, navigation}: any) => {
  const devices = useCameraDevices('wide-angle-camera');
  const device = devices.back;
  const [hasPermission, setHasPermission] = useState(false);
  const cameraRef = useRef<Camera>(null);
  const [flip, setFlip] = useState(false);
  // const device = devices.front;
  // let cameraDevice = flip ? devices.front : devices.back;
  const [isCameraActive, setIsCameraActive] = useState(true);
  const [isModalVisible, setIsModalVisible] = useState(true);

  const toggleCamera = () => {
    if (isCameraActive) {
      setFlip(!flip);
    }
  };

  useEffect(() => {
    const handleBackPress = () => {
      navigation.navigate('ChatDetails', {username: route.params.username});
      return true;
    };
    const backHandler = BackHandler.addEventListener(
      'hardwareBackPress',
      handleBackPress,
    );
    return () => {
      backHandler.remove();
    };
  }, [navigation]);

  const toggleCameraComponent = () => {
    setIsCameraActive(!isCameraActive);
  };
  useEffect(() => {
    Camera.requestCameraPermission().then(p =>
      setHasPermission(p === 'granted'),
    );
  }, []);

  const panResponder = React.useRef(
    PanResponder.create({
      onStartShouldSetPanResponder: () => true,
      onPanResponderMove: (evt, gestureState) => {
        const swipeThreshold = 100;
        if (gestureState.dy < -swipeThreshold) {
          setIsModalVisible(true);
        }
      },
      onPanResponderRelease: () => {},
      onPanResponderTerminate: () => {},
    }),
  ).current;

  return (
    <View style={styles.container} {...panResponder.panHandlers}>
      {device && (
        <Camera
          ref={cameraRef}
          device={device}
          isActive={isCameraActive}
          style={isCameraActive && StyleSheet.absoluteFill}
        />
      )}
      <View
        style={[
          styles.avatarContainer,
          !isCameraActive && styles.avatarContainerCameraInactive,
        ]}>
        <Avatar.Text
          size={!isCameraActive ? 100 : 40}
          label={route.params.username[0]}
          style={{
            backgroundColor: '#5e01a9',
            borderWidth: 1,
            borderColor: 'white',
          }}
        />
        <View
          style={
            !isCameraActive && {justifyContent: 'center', alignItems: 'center'}
          }>
          <Text style={styles.name}>{route.params.username}</Text>
          <Text style={styles.status}>
            @{route.params.username.replace(/\s+/g, '').toLowerCase()}
          </Text>
        </View>
      </View>
      {isModalVisible && (
        <VideoCallModal
          onPressTopLine={() => setIsModalVisible(false)}
          toggleCamera={toggleCamera}
          toggleCameraComponent={toggleCameraComponent}
          isCameraActive={isCameraActive}
          username={route.params.username}
        />
      )}
    </View>
  );
};`

I even tried the documentation code but still the error persist. 

My setup:
Xiaomi Redmi Note 10S
Android V13

mrousavy, 2023-10-05:
@SameerAther 

1. You did not include any logs or stacktrace. How do you expect me to help you?
2. You use V2 code. `useCameraDevices('wide-angle-camera')` is not part of V3. This is V2 API.
3. "I even tried the documentation code" - I don't believe you, the documentation never mentions this V2 API.

ermohitpy, 2024-02-09:
I am getting the same issue , It works fine for android but in IOS , the device.id is undefined and when I debug the issue I find 
const devices = useCameraDevices()
this code gives me :
{ back: undefined,
  external: undefined,
  front: undefined,
  unspecified: undefined }
  My configs are:
  react-native : 0.70.6
  vision-camera: 2.15.4
  device : iphone 14 simulator

amir-ibrahim, 2024-09-04:
Still facing this issue in RN 0.73.9 did anyone find a solution?

jedi24garcia, 2024-09-15:
Has anyone actually solved this issue?

nickmccomb, 2025-02-19:
useCameraDevice('back')