margelo/react-native-vision-camera #609: Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug' 🐛

sijav, 2021-11-23:
### What were you trying to do?

I can't build for debug

### What happened instead?

I'm getting the folowing errors:
error: redefinition of 'JHashMap'
error: redefinition of 'create'

### Relevant log output

```shell
Build VisionCamera_armeabi-v7a
ninja: Entering directory `C:\Some\Path\Here\ReactNativeProject\node_modules\react-native-vision-camera\android\.cxx\cmake\debug\armeabi-v7a'
In file included from ../../../../src/main/cpp/java-bindings/JHashMap.cpp:5:
../../../../src/main/cpp/java-bindings/JHashMap.h:16:8: error: redefinition of 'JHashMap'
../../../../build/fbjni-0.3.0-headers.jar\fbjni/detail/Iterator.h:170:8: note: previous definition is here
../../../../src/main/cpp/java-bindings/JHashMap.cpp:15:43: error: redefinition of 'create'
../../../../build/fbjni-0.3.0-headers.jar\fbjni/detail/Iterator-inl.h:230:42: note: previous definition is here
In file included from ../../../../src/main/cpp/JSIJNIConversion.cpp:25:
../../../../src/main/cpp/java-bindings/JHashMap.h:16:8: error: redefinition of 'JHashMap'
../../../../build/fbjni-0.3.0-headers.jar\fbjni/detail/Iterator.h:170:8: note: previous definition is here
```

### Device

Xiaomi K20Pro

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Baer05, 2021-11-23:
same problem -.-

timh1004, 2021-11-23:
I also have this exact issue! A week ago it was working properly.

sijav, 2021-11-23:
I guess it's because of the new fbjni module that added JHashMap in the new version perhaps?
I'll try to remove them and see if it resolve anything

sijav, 2021-11-23:
yup worked:
remove both JHashMap.h and JHashMap.cpp from node_modules/react-native-vision-camera/android/src/main/cpp/java-binding
from node_modules/react-native-vision-camera/android/src/main/cpp/JSIJNIConversion.cpp remove the following line 25:
`#include "java-bindings/JHashMap.h"`
from node_modules/react-native-vision-camera/android/CMakeLists.txt remove the following line 24:
`        src/main/cpp/java-bindings/JHashMap.cpp`
looking forward for a fix

mrousavy, 2021-11-23:
> I guess it's because of the new fbjni module that added JHashMap in the new version perhaps?

yes, I waited until they released a new version with that. I'll remove mine

PR #610 (2021-11-23): fix: Fix duplicate `JHashMap` definition by moving to custom namespace
Technically `JHashMap` is duplicated now, but in separate namespaces. If I were to remove my `JHashMap` (and `JArrayList`) definitions, the user is forced to use fbjni v3.

fixes #609
closed 2021-11-23 by mrousavy