margelo/react-native-vision-camera #611: 🐛 Undefined symbols for architecture x86_64 when building for Simulator and Undefined symbols for architecture arm64 when building for iOS Device

rishijoshi-ibm, 2021-11-23:
### What were you trying to do?

Building iOS Application
Followed the [Troubleshooting Guide](https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting) and kept `SWIFT_VERSION` as `5.2` and also removed explicit reference to `Swift-5.0` from `LIBRARY_SEARCH_PATH`

### Reproduceable Code

```tsx
xcodebuild -workspace "./TestApp.xcworkspace" -scheme "TestApp" -sdk "iphonesimulator" -destination "generic/platform=iOS Simulator" -configuration "Release" -derivedDataPath ./build ONLY_ACTIVE_ARCH=NO
```

### What happened instead?

Build Failed

### Relevant log output

```shell
❌  ld: symbol(s) not found for architecture x86_64
❌  clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

### Device

Iphone Simulator (IOS 14.5)

### VisionCamera Version

1.0.10

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-11-24:
There's no actual log output here. Please paste the actual errors.

rishijoshi-ibm, 2021-11-25:
@mrousavy Please find the error logs:

```
/Users/*****/Projects/validations/TestApp/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift:16:17: error: type 'AVCaptureColorSpace' has no member 'HLG_BT2020'
        self = .HLG_BT2020
               ~^~~~~~~~~~
/Users/*****/Projects/validations/TestApp/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift:34:11: error: type 'AVCaptureColorSpace' has no member 'HLG_BT2020'
    case .HLG_BT2020:
         ~^~~~~~~~~~
/Users/*****/Projects/validations/TestApp/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+matchesFilter.swift:67:36: error: cannot find 'supportedColorSpaces' in scope
      let allColorSpacesIncluded = supportedColorSpaces.allSatisfy { avColorSpaces.contains($0) }
                                   ^~~~~~~~~~~~~~~~~~~~
/Users/*****/Projects/validations/TestApp/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+toDictionary.swift:34:22: error: cannot find 'supportedColorSpaces' in scope
      "colorSpaces": supportedColorSpaces.map(\.descriptor),
                     ^~~~~~~~~~~~~~~~~~~~
/Users/*****/Projects/validations/TestApp/node_modules/react-native-vision-camera/ios/CameraView+TakePhoto.swift:79:23: error: value of type 'AVCapturePhotoSettings' has no member 'isAutoContentAwareDistortionCorrectionEnabled'
        photoSettings.isAutoContentAwareDistortionCorrectionEnabled = enableAutoDistortionCorrection
        ~~~~~~~~~~~~~ ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

CompileSwiftSources normal arm64 com.apple.xcode.tools.swift.compiler (in target 'VisionCamera' from project 'Pods')

```

rishijoshi-ibm, 2021-12-01:
@mrousavy did you go through the logs and understand what the issue is all about?

mrousavy, 2021-12-01:
Yeah, looks like either your installation is messed up or you have an older Swift version.

Clean your ios/Pods folder and reinstall them, and take a look at the Troubleshooting guide in the documentation.

rishijoshi-ibm, 2021-12-03:
I tried clean install and also updated the Swift version to 5.2, still it's giving same issue. Can you give me some pointers as to in what direction shall I look into?

shuzootani, 2022-01-11:
@rishijoshi-ibm I'm not quite sure if this works for your problem, but you could check out the troubleshooting section.
https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting#ios

mrousavy, 2023-09-30:
This should be resolved now with V3! :)

closed 2023-09-30 by mrousavy