margelo/react-native-vision-camera #613: Constantly getting random crashes every now and then🐛

sijav, 2021-11-26:
### What were you trying to do?

After some time camera opening and processing QR codes I'm getting the following issues.

### Reproduceable Code

```tsx
I'm also using QRCode as a frame processor
```

### What happened instead?

Fatal signal 7 (SIGBUS), code 1 (BUS_ADRALN), fault addr 0x7b3e7be02e in tid 32375 (pool-34-thread-), pid 32010 (my.app.id)

### Relevant log output

```shell
2021-11-26 06:00:55.020 32010-32375/? A/libc: Fatal signal 7 (SIGBUS), code 1 (BUS_ADRALN), fault addr 0x7b3e7be02e in tid 32375 (pool-34-thread-), pid 32010 (my.app.id)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #00 pc 00000000000f16d0  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #01 pc 00000000000eb8cc  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #02 pc 00000000000e826c  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #03 pc 00000000000eacf0  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #04 pc 000000000005b180  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #05 pc 000000000005a040  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #06 pc 00000000000260d8  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #07 pc 00000000000152c8  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libVisionCamera.so (vision::JSIJNIConversion::convertJNIObjectToJSIValue(facebook::jsi::Runtime&, facebook::jni::basic_strong_ref<_jobject*, facebook::jni::LocalReferenceAllocator> const&)+728) (BuildId: 330ff3b65273e029876e71de45e53cc16f8d7f4b)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #08 pc 0000000000015440  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libVisionCamera.so (vision::JSIJNIConversion::convertJNIObjectToJSIValue(facebook::jsi::Runtime&, facebook::jni::basic_strong_ref<_jobject*, facebook::jni::LocalReferenceAllocator> const&)+1104) (BuildId: 330ff3b65273e029876e71de45e53cc16f8d7f4b)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #09 pc 000000000001d80c  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libVisionCamera.so (BuildId: 330ff3b65273e029876e71de45e53cc16f8d7f4b)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #10 pc 00000000000303bc  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #11 pc 000000000002fd5c  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #12 pc 00000000000363a4  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #13 pc 000000000004f000  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.388 32526-32526/? A/DEBUG:       #14 pc 0000000000050974  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #15 pc 0000000000050130  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #16 pc 0000000000036608  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #17 pc 0000000000026950  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #18 pc 00000000000b9f6c  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (facebook::jsi::Function::call(facebook::jsi::Runtime&, facebook::jsi::Value const*, unsigned long) const+104) (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #19 pc 00000000000d7114  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #20 pc 00000000000d6ec4  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #21 pc 00000000000d6e08  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #22 pc 00000000000d6d68  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #23 pc 00000000000d5c30  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #24 pc 00000000000303bc  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #25 pc 000000000002fd5c  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #26 pc 00000000000363a4  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #27 pc 000000000004f000  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #28 pc 0000000000050974  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #29 pc 0000000000050130  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #30 pc 0000000000036608  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #31 pc 0000000000026950  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libhermes.so (BuildId: eddee6deb8a97571592812907782e8a8e580c0c8)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #32 pc 00000000000b9f6c  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (facebook::jsi::Function::call(facebook::jsi::Runtime&, facebook::jsi::Value const*, unsigned long) const+104) (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #33 pc 00000000000d7114  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #34 pc 00000000000d6ec4  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #35 pc 00000000000d6e08  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
2021-11-26 06:00:55.389 32526-32526/? A/DEBUG:       #36 pc 00000000000d6d68  /data/app/DATA_HASH/my.app.id-MY_APP_SIG/lib/arm64/libreanimated.so (BuildId: 82beb6674b1fcf7c178099fb1b3c8eab556ca2d3)
[... cut ...]
Android Miui 12.5 Xiaomi K20 pro

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Mohamed-kassim, 2022-03-16:
@mrousavy the app is also crashing for me using vision camera face detector as a processor
actually, it crashes in only two cases
1- saving the code to trigger a fast refresh
2- navigating back from the camera to the previous screen 
our hander have a long process done on catched face, I noticed that when I decrease the logic code inside the handler the crash doesn't happen every time  I need to repeatedly trigger fast refresh to crash
I tried the issue with and without flipper and it happens
this is my trace log
`*** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***

Build fingerprint: 'POCO/surya_global/surya:11/RKQ1.200826.002/V12.5.5.0.RJGMIXM:user/release-keys'
Revision: '0'
ABI: 'arm64'
Timestamp: 2022-03-17 01:27:55+0200
pid: 7297, tid: 7391, name: mqt_js  >>> com.hayasolutions.attendlab <<<
uid: 10636
signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x1e
Cause: null pointer dereference
    x0  000000000000000e  x1  b400006fafc119c0  x2  000000000000229b  x3  0000000000000001
    x4  0000006fa331c8a0  x5  0000006f9df6689d  x6  0000000000000000  x7  0000006f9f19fe79
    x8  d644f1108486d3e3  x9  0000000000000000  x10 000000000000005f  x11 0000000000000000
    x12 000000000000003c  x13 0000000000000007  x14 0000006f9f19c994  x15 0000000000000002
    x16 0000000000000004  x17 0000000000000000  x18 0000006fa0e2e000  x19 0000006fa331c8a0
    x20 000000000000229b  x21 b400006fafc119c0  x22 b400006f8f4e4030  x23 b400006fafc119c0
    x24 0000006fa3320000  x25 0000000000000000  x26 0000000000000000  x27 0000006fa331c900
    x28 b400006fafc119d0  x29 0000006fa331c810
    lr  0000006fc3019680  sp  0000006fa331c7d0  pc  0000006fc2ffadd0  pst 0000000080000000
backtrace:
      #00 pc 000000000004edd0  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #01 pc 000000000006d67c  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #02 pc 000000000006e15c  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #03 pc 0000000000027c1c  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #04 pc 00000000000c9e10  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libreanimated.so (offset 0x30ab000) (facebook::jsi::Object::getProperty(facebook::jsi::Runtime&, facebook::jsi::String const&) const+52) (BuildId: beda31e6b53a653c7ea41497a9312003fca670bc)
      #05 pc 00000000000c60b8  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libreanimated.so (offset 0x30ab000) (facebook::jsi::Object::getProperty(facebook::jsi::Runtime&, char const*) const+96) (BuildId: beda31e6b53a653c7ea41497a9312003fca670bc)
      #06 pc 00000000000efa14  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libreanimated.so (offset 0x30ab000) (reanimated::ShareableValue::adapt(facebook::jsi::Runtime&, facebook::jsi::Value const&, reanimated::ValueType)+108) (BuildId: beda31e6b53a653c7ea41497a9312003fca670bc)
      #07 pc 00000000000f0b00  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libreanimated.so (offset 0x30ab000) (reanimated::ShareableValue::adapt(facebook::jsi::Runtime&, facebook::jsi::Value const&, reanimated::RuntimeManager*, reanimated::ValueType)+236) (BuildId: beda31e6b53a653c7ea41497a9312003fca670bc)
      #08 pc 00000000000e37e8  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libreanimated.so (offset 0x30ab000) (reanimated::MutableValue::set(facebook::jsi::Runtime&, facebook::jsi::PropNameID const&, facebook::jsi::Value const&)+1184) (BuildId: beda31e6b53a653c7ea41497a9312003fca670bc)
      #09 pc 0000000000030824  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #10 pc 000000000006f0bc  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #11 pc 0000000000057d34  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #12 pc 0000000000055f8c  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #13 pc 000000000003aa1c  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4b9c00df73)
      #14 pc 00000000000390c8  /data/app/~~H5-oBC-CWpLXuYwPYlYoyQ==/com.hayasolutions.attendlab-8K-1dHbSkSLsIgU70pS8GQ==/base.apk!libhermes.so (offset 0x1eca000) (BuildId: d6a741d92f2e596cdaa865206992ce4 [... cut ...] t*)+460) (BuildId: 8da8188d8b3a217a92b42486afb82894)
      #61 pc 00000000005a4f10  /apex/com.android.art/lib64/libart.so (art::Thread::CreateCallback(void*)+1308) (BuildId: 8da8188d8b3a217a92b42486afb82894)
      #62 pc 00000000000eb868  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+64) (BuildId: e679588de88514d3f5c28f04a93ffc0e)
      #63 pc 000000000008ba88  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: e679588de88514d3f5c28f04a93ffc0e)`

mrousavy, 2022-03-17:
huh, okay those are unfortunately not symbolicated but it looks like a frame processor access error. I'll try to figure this out soon once I have some free time..

HyopeR, 2022-04-14:
Will there be any regulation on this bug related to frame processors? Is there a workaround that we can use temporarily? Thanks for your hard work. @mrousavy

mrousavy, 2023-08-18:
Hey! I've rewritten the entire Android codebase of VisionCamera from CameraX to Camera2 in the efforts of [✨ VisionCamera V3](https://github.com/mrousavy/react-native-vision-camera/issues/1376).

I just now completed the Camera2 rewrite and I believe the core structure is running, but there might be some edge cases to iron out. Can you try and test the PR #1674 for me to see if you can still reproduce this issue here?

Here's an instruction on how you can test that: https://github.com/mrousavy/react-native-vision-camera/pull/1674#issuecomment-1684104217

If the issue cannot be reproduced with that version/PR anymore, then hoorayy, I fixed it! 🎉 
Otherwise please let me know and I'll keep this issue open to keep track of it.

Thank you!

closed 2023-09-01 by mrousavy