margelo/react-native-vision-camera #614: 🐛 Task :react-native-vision-camera:externalNativeBuildDebug failed on android build after upgrade to React-Native 0.66

stephraja, 2021-11-26:
### What were you trying to do?

After upgrading my application from React-Native 0.65 to 0.66, the build android is failing on executing react-native-vision-camera actions

I've looked for a similar issue into this repo but there is no one that looks like it instead of #609 which is already fixed and applied with the 2.9.4 version that I'm using

I am not familiar with Android build and debug that is why I am posting this issue if some people knows how to resolve or understand it and can explain me

I have tried to downgrade the ndk-version from 21.4.7075529 to 20.1.5948944 to see if the problems comes from this but it did not solve the problem

The iOS build works with no problem btw

### What happened instead?

The android installation to my device crashes with the following error :

### Relevant log output

```shell
Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'.
> Build command failed.
  Error while executing process /Users/up2it/Library/Android/sdk/cmake/3.10.2.4988404/bin/ninja with arguments {-C <project-path>/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a VisionCamera}
  ninja: Entering directory `<project-path>/node_modules/react-native-vision-camera/android/.cxx/cmake/debug/armeabi-v7a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/CameraView.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  [6/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  [7/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  [8/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  [9/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o
  [10/10] Linking CXX shared library ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libVisionCamera.so 
  : && /Users/up2it/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=armv7-none-linux-androideabi21 --gcc-toolchain=/Users/up2it/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/up2it/Library/Android/sdk/ndk/21.4.7075529/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -DFOLLY_NO_CONFIG=1 -DFOLLY_HAVE_CLOCK_GETTIME=1 -DFOLLY_HAVE_MEMRCHR=1 -DFOLLY_USE_LIBCPP=1 -DFOLLY_MOBILE=1 -DON_ANDROID -DONANDROID -DFOR_HERMES=true -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -Wl,--build-id -Wl,--fatal-warnings -Wl,--exclude-libs,libunwind.a -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cmake/debug/obj/armeabi-v7a/libVisionCamera.so CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/CameraView.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o -L<project-path>/node_modules/react-native-vision-camera/android/build/third-party-ndk/hermes/jni/armeabi-v7a  -L<project-path>/node_modules/react-native-vision-camera/android/build/react-native-reanimated-66-hermes.aar/jni/armeabi-v7a  -L<project-path>/node_modules/react-native-vision-camera/android/build/react-native-0.66.3.aar/jni/armeabi-v7a -llog -lhermes -lreanimated -lreactnativejni -lfbjni -lfolly_json -landroid -latomic -lm && :
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJSIValueToJNIObject(facebook::jsi::Runtime&, facebook::jsi::Value const&): error: undefined reference to 'facebook::jsi::Value::asObject(facebook::jsi::Runtime&) const &'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJSIValueToJNIObject(facebook::jsi::Runtime&, facebook::jsi::Value const&): error: undefined reference to 'facebook::jsi::dynamicFromValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJSIValueToJNIObject(facebook::jsi::Runtime&, facebook::jsi::Value const&): error: undefined reference to 'facebook::jsi::dynamicFromValue(facebook::jsi::Runtime&, facebook::jsi::Value const&)'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJSIValueToJNIObject(facebook::jsi::Runtime&, facebook::jsi::Value const&): error: undefined reference to 'facebook::jsi::Value::toString(facebook::jsi::Runtime&) const'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJSIValueToJNIObject(facebook::jsi::Runtime&, facebook::jsi::Value const&): error: undefined reference to 'typeinfo for facebook::jsi::HostObject'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJNIObjectToJSIValue(facebook::jsi::Runtime&, facebook::jni::basic_strong_ref<_jobject*, facebook::jni::LocalReferenceAllocator> const&): error: undefined reference to 'facebook::jsi::Value::~Value()'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJNIObjectToJSIValue(facebook::jsi::Runtime&, facebook::jni::basic_strong_ref<_jobject*, facebook::jni::LocalReferenceAllocator> const&): error: undefined reference to 'facebook::jsi::Value::~Value()'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJNIObjectToJSIValue(facebook::jsi::Runtime&, facebook::jni::basic_strong_ref<_jobject*, facebook::jni::LocalReferenceAllocator> const&): error: undefined reference to 'facebook::jsi::Value::~Value()'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function vision::JSIJNIConversion::convertJNIObjectToJSIValue(facebook::jsi::Runtime&, facebook::jni::basic_strong_ref<_jobject*, facebook::jni::LocalReferenceAllocator> const&): error: undefined reference to 'facebook::jsi::Value::~Value()'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function facebook::jsi::Value::Value<facebook::jsi::String>(facebook::jsi::String&&): error: undefined reference to 'facebook::jsi::Value::~Value()'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function facebook::jsi::Value::Value<facebook::jsi::Array>(facebook::jsi::Array&&): error: undefined reference to 'facebook::jsi::Value::~Value()'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function facebook::jsi::Value::Value<facebook::jsi::Object>(facebook::jsi::Object&&): error: undefined reference to 'facebook::jsi::Value::~Value()'
  CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o:JSIJNIConversion.cpp:function facebook::jsi::detail::toValue(facebook::jsi::Runtime&, facebook::jsi::Value const&): error: undefined reference to 'facebook::jsi::Value::Value(facebook::jsi::Runtime&, facebook::jsi::Value const&)'
  CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o:FrameHostObject.cpp:function vision::FrameHostObject::FrameHostObject(facebook::jni::alias_ref<facebook::jni::detail::JTypeFor<vision::JImageProxy, facebook::jni::JObject, void>::_javaobject*>): error: undefined reference to 'facebook::jsi::HostObject::~HostObject()'
[... cut ...]
Huawei P20 Lite

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jettandres, 2021-11-27:
+1 I've also been debugging this with no success

NiketanG, 2021-11-27:
+1 Same issue here, Even `gradlew clean` fails with following error
```
> Configure project :react-native-vision-camera
Warning: The 'kotlin-android-extensions' Gradle plugin is deprecated. Please use this migration guide (https://goo.gle/kotlin-android-extensions-deprecation) to start working with View Binding (https://developer.android.com/topic/libraries/view-binding) and the 'kotlin-parcelize' plugin.
```

stephraja, 2021-11-29:
I have tried to upgrade my CMake version from 3.10.2.4988404 to 3.18.1 but still have the same output.

I will continue to work on it and let you know if I find a solution

Shreyakkk, 2021-11-29:
+1 same issue

FloweryK, 2021-12-03:
same here. I also tested with version 2.9.0, and still doesn't work with either RN 0.65 and 0.66 ;(

batical, 2021-12-10:
still present in 2.10.0

stephraja, 2021-12-15:
I found the solution, I was facing the #588 issue before I upgraded React Native and I patched the CMAKELIST.txt file by removing the JSI_LIB like mentioned in #513

After I removed this patch, the Android application build completely without a problem

Hope that it will help everyone who face this issue

stephraja, 2021-12-15:
Hope that it will help everyone who face this issue

closed 2021-12-15 by stephraja

abdullahIsa, 2022-01-02:
For anyone having this issue, do same as this [https://github.com/mrousavy/react-native-vision-camera/issues/588#issuecomment-972589478](url)

geakk, 2022-02-17:
same as @sami-99

mineturtle2137, 2022-03-04:
same as @sami-99 and @geakk, RN v0.67.3 react-native-vision-camera v ^2.12.0

Details:
```
> Task :react-native-vision-camera:externalNativeBuildDebug FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.9/userguide/command_line_interface.html#sec:command_line_warnings
261 actionable tasks: 9 executed, 252 up-to-date
```

```
BUILD FAILED in 57s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:externalNativeBuildDebug'.
> Build command failed.

```

Not sure why the thread is closed as the library fails on the android build. I tried downgrading to RN 0.66 and the solutions from other issues but nothing fixes the problem