margelo/react-native-vision-camera #624: How do we change the camera orientation❓

jzxchiang1, 2021-11-30:
### Question

When testing on both an iOS device and an Android simulator with v2.8.3 of this library, I noticed the `metadata` for `takePhoto` always has `Orientation: 6`.

Even when I physically rotate my iPhone (or click the rotate right/left buttons for the Pixel emulator), the metadata always has an Orientation value of 6, which is the default 90 degree clockwise rotation of landscape mode, i.e. the upright portrait mode. My end goal is to take photos in landscape mode, and have the photos also be stored in "landscape mode".

How do I take advantage of the rotation feature that was available starting from v2.5.0? How does this library listen for orientation changes? I checked my AndroidManifest.xml and XCode settings, and I don't lock the device to portrait only... I don't even know how to do that...

Couldn't find any examples in the repo. Thanks!

### VisionCamera Version

2.8.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

jzxchiang1, 2021-11-30:
Do I have to install a library like `react-native-orientation-locker` for this rotation feature to work properly?

mrousavy, 2021-11-30:
> Do I have to install a library like react-native-orientation-locker for this rotation feature to work properly?

No.

Does your screen even rotate? If your react view actually rotates, then the camera should do too. How does this behave on iOS?

jzxchiang1, 2021-12-01:
Yeah sorry, it was a false alarm. I'm seeing the rotation now.

Is there a way to take a "rotated" photo in landscape position but have the UI remain in portrait mode? I tried locking the screen to portrait mode, but that also makes it so that the vision-camera doesn't detect landscape mode and thus takes the photo without rotation.

In other words, have vision-camera detect rotation based on device orientation instead of UI orientation.

Or do I need to handle that myself with a 3rd party image rotation library? Thanks a lot!

jzxchiang1, 2021-12-06:
I resolved it by using `react-native-orientation-locker` and listening to the device orientation.

Note that device orientation [doesn't work](https://github.com/wonday/react-native-orientation-locker/issues/141) for iOS if the user has locked the orientation from their Control Center. See my comment on that issue for how to resolve it.

mrousavy, 2022-01-04:
Implemented in #715! Hope this works for you :) If you appreciate the work I do in my free time, please consider sponsoring me on GitHub, it would mean a lot to me. 🖤

thomasttvo, 2022-05-07:
@mrousavy we can also copy the `react-native-camera`'s code that checks the gravity vector to guesstimate the orientation https://github.com/react-native-camera/react-native-camera/blob/master/ios/RN/RNSensorOrientationChecker.m

You can also use `react-native-sensors` 

```ts
import { useLayoutEffect, useRef } from 'react';
import { Platform } from 'react-native';

import { gravity } from 'react-native-sensors';

type Rotation = 'top' | 'down' | 'right' | 'left';

export const useDeviceRotationSensor = (
  callback: (rotation: Rotation, degree: number) => void,
) => {
  const callbackRef = useRef(callback);
  callbackRef.current = callback;
  useLayoutEffect(() => {
    // We use gravity sensor here because react-native-orientation
    // can't detect landscape orientation when the device's orientation is locked
    const subscription = gravity.subscribe(({ x, y }) => {
      const radian = Math.atan2(y, x);
      const degree = (radian * 180) / Math.PI;

      let rotation: Rotation = 'left';
      if (degree > -135) rotation = 'top';
      if (degree > -45) rotation = 'right';
      if (degree > 45) rotation = 'down';
      if (degree > 135) rotation = 'left';

      if (Platform.OS === 'android') {
        rotation = 'right';
        if (degree > -135) rotation = 'down';
        if (degree > -45) rotation = 'left';
        if (degree > 45) rotation = 'top';
        if (degree > 135) rotation = 'right';
      }

      callbackRef.current(rotation, degree);
    });
    return () => subscription.unsubscribe();
  }, []);
};
```

mrousavy, 2022-05-09:
That's weird. This should be detected automatically by VisionCamera

PABourdais, 2022-05-09:
wo @thomasttvo I was about to look at the old code from react-native-camera to make this feature works.
Going to try it asap

PABourdais, 2022-05-11:
Thanks @thomasttvo, I was able to fix it with `react-native-sensors`

yestay90, 2022-12-29:
@mrousavy , when taking picture on Android in portrait mode, the orientation is 0, However, iOS return orientation of 6. I think iOS orientation is wrong.

andreiwow2, 2023-01-23:
> 

I also have the same issue, my photos on iOS have orientation 6.

ajp8164, 2023-05-12:
> @mrousavy @jzxchiang1 To get around the orientation lock issue, we can copy the `react-native-camera`'s code that checks the gravity vector to guesstimate the orientation https://github.com/react-native-camera/react-native-camera/blob/master/ios/RN/RNSensorOrientationChecker.m
> 
> Alternatively if you don't wanna touch this library's code you can also use `react-native-sensors` with the hook below
> 
> ```ts
> import { useLayoutEffect, useRef } from 'react';
> import { Platform } from 'react-native';
> 
> import { gravity } from 'react-native-sensors';
> 
> type Rotation = 'top' | 'down' | 'right' | 'left';
> 
> export const useDeviceRotationSensor = (
>   callback: (rotation: Rotation, degree: number) => void,
> ) => {
>   const callbackRef = useRef(callback);
>   callbackRef.current = callback;
>   useLayoutEffect(() => {
>     // We use gravity sensor here because react-native-orientation
>     // can't detect landscape orientation when the device's orientation is locked
>     const subscription = gravity.subscribe(({ x, y }) => {
>       const radian = Math.atan2(y, x);
>       const degree = (radian * 180) / Math.PI;
> 
>       let rotation: Rotation = 'left';
>       if (degree > -135) rotation = 'top';
>       if (degree > -45) rotation = 'right';
>       if (degree > 45) rotation = 'down';
>       if (degree > 135) rotation = 'left';
> 
>       if (Platform.OS === 'android') {
>         rotation = 'right';
>         if (degree > -135) rotation = 'down';
>         if (degree > -45) rotation = 'left';
>         if (degree > 45) rotation = 'top';
>         if (degree > 135) rotation = 'right';
>       }
> 
>       callbackRef.current(rotation, degree);
>     });
>     return () => subscription.unsubscribe();
>   }, []);
> };
> ```
> 
> In your component
> 
> ```tsx
>  useDeviceRotationSensor((rotation) => {
>     // These still work when the device orientation is unlocked
>     setCameraOrientation('landscapeRight');
>     if (rotation === 'top') setCameraOrientation('portrait');
>     if (rotation === 'right') setCameraOrientation('landscapeLeft');
>     if (rotation === 'down') setCameraOrientation('portraitUpsideDown');
>     if (rotation === 'left') setCameraOrientation('landscapeRight');
>   });
> ```

This works but  the camera scene momentarily freezes while the vision camera changes orientation.

mrousavy, 2023-09-30:
Hey - I'm tracking Orientation in this feature request/issue now: https://github.com/mrousavy/react-native-vision-camera/issues/1891 ✨ 

Make sure to upvote or sponsor to support this feature, and leave a comment if you have any additional thoughts/ideas.

ingfdoaguirre, 2024-03-06:
> This works but the camera scene momentarily freezes while the vision camera changes orientation.

Hi, the problem is because every 100ms the effect is reading the orientation of the phone, so, every 100ms the camera orientation is changing.

Just create another state like "currentRotation" that saves the current orientation of the change, and only change it when it really changes, for example, from top to left.

```
const [currentRotation, setCurrentRotation] = useState('top');
.
.
.
useDeviceRotationSensor(rotation => {
    if (currentRotation !== rotation) {
      setCurrentRotation(rotation);
    }
  });
.
.
.
useEffect(() => {
    if (currentRotation === 'top' || currentRotation === 'down')
      setCameraOrientation('portrait');
    if (currentRotation === 'right') setCameraOrientation('landscape-left');
    if (currentRotation === 'left') setCameraOrientation('landscape-right');
  }, [currentRotation]);
```

matheusqfql, 2024-03-27:
This is no working for me on Ipad with camera V3

IamHungry113, 2024-04-10:
> This is not working for me on Ipad with camera V3

yes same question here

nabzmohd, 2024-05-09:
const [orientation, setOrientation] = useState('portrait');

 useEffect(() => {
    setUpdateIntervalForType(SensorTypes.accelerometer, 800); // defaults to 100ms
    const subscription = accelerometer.subscribe(
      ({x, y, z}) => {
        const absX = Math.abs(x);
        const absY = Math.abs(y);

        // Determine the orientation based on accelerometer data
        if (absX > absY) {
          if (x > 0) {
            setOrientation('landscape-left');
          } else {
            setOrientation('landscape-right');
          }
        } else {
          setOrientation('portrait');
        }
      },
      (error) => {
        console.log('The sensor is not available', error);
      },
    );

    return () => {
      setTimeout(() => {
        // If it's the last subscription to accelerometer it will stop polling in the native API
        subscription.unsubscribe();
      }, 1000);
    };
  });

nabzmohd, 2024-05-09:
// 
const [orientation, setOrientation] = useState('portrait');

useEffect(() => {
setUpdateIntervalForType(SensorTypes.accelerometer, 800); // defaults to 100ms
const subscription = accelerometer.subscribe(
({x, y, z}) => {
const absX = Math.abs(x);
const absY = Math.abs(y);

    // Determine the orientation based on accelerometer data
    if (absX > absY) {
      if (x > 0) {
        setOrientation('landscape-left');
      } else {
        setOrientation('landscape-right');
      }
    } else {
      setOrientation('portrait');
    }
  },
  (error) => {
    console.log('The sensor is not available', error);
  },
);

return () => {
  setTimeout(() => {
    // If it's the last subscription to accelerometer it will stop polling in the native API
    subscription.unsubscribe();
  }, 1000);
};

nabzmohd, 2024-05-09:
> // const [orientation, setOrientation] = useState('portrait');
>
> useEffect(() => { setUpdateIntervalForType(SensorTypes.accelerometer, 800); // defaults to 100ms const subscription = [...]

this work for me

sugaith, 2024-08-20:
from where is this `setOrientation(...)` cames from?

> > > @mrousavy @jzxchiang1 To get around the orientation lock issue, we can copy the `react-native-camera`'s code that checks the gravity vector to guesstimate the orientation https://github.com/react-native-camera/react-native-camera/blob/master/ios/RN/RNSensorOrientationChecker.m
> > > Alternatively if you don't wanna touch this library's code you can also use `react-native-sensors` with the hook below
> > > ```ts
> > > import { useLayoutEffect, useRef } from 'react';
> > > import { Platform } from 'react-native';
> > > 
> > > import { gravity } from 'react-native-sensors';
> > > 
> > > type Rotation = 'top' | 'down' | 'right' | 'left';
> > > 
> > > export const useDeviceRotationSensor = (
> > >   callback: (rotation: Rotation, degree: number) => void,
> > > ) => {
> > >   const callbackRef = useRef(callback);
> > >   callbackRef.current = callback;
> > >   useLayoutEffect(() => {
> > >     // We use gravity sensor here because react-native-orientation
> > >     // can't detect landscape orientation when the device's orientation is locked
> > >     const subscription = gravity.subscribe(({ x, y }) => {
> > >       const radian = Math.atan2(y, x);
> > >       const degree = (radian * 180) / Math.PI;
> > > 
> > >       let rotation: Rotation = 'left';
> > >       if (degree > -135) rotation = 'top';
> > >       if (degree > -45) rotation = 'right';
> > >       if (degree > 45) rotation = 'down';
> > >       if (degree > 135) rotation = 'left';
> > > 
> > >       if (Platform.OS === 'android') {
> > >         rotation = 'right';
> > >         if (degree > -135) rotation = 'down';
> > >         if (degree > -45) rotation = 'left';
> > >         if (degree > 45) rotation = 'top';
> > >         if (degree > 135) rotation = 'right';
> > >       }
> > > 
> > >       callbackRef.current(rotation, degree);
> > >     });
> > >     return () => subscription.unsubscribe();
> > >   }, []);
> > > };
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > In your component
> > > ```tsx
> > >  useDeviceRotationSensor((rotation) => {
> > >     // These still work when the device orientation is unlocked
> > >     setCameraOrientation('landscapeRight');
> > >     if (rotation === 'top') setCameraOrientation('portrait');
> > >     if (rotation === 'right') setCameraOrientation('landscapeLeft');
> > >     if (rotation === 'down') setCameraOrientation('portraitUpsideDown');
> > >     if (rotation === 'left') setCameraOrientation('landscapeRight');
> > >   });
> > > ```
> > 
> > 
> > This works but the camera scene momentarily freezes while the vision camera changes orientation.
> 
> Hi, the problem is because every 100ms the effect is reading the orientation of the phone, so, every 100ms the camera orientation is changing.
> 
> Just create another state like "currentRotation" that saves the current orientation of the change, and only change it when it really changes, for example, from top to left.
> 
> This solved the freeze problem, also I had a problem that the app was crashing, because the camera orientation was changing every 100ms.
> 
> ```
> const [currentRotation, setCurrentRotation] = useState('top');
> .
> .
> .
> useDeviceRotationSensor(rotation => {
>     if (currentRotation !== rotation) {
>       setCurrentRotation(rotation);
>     }
>   });
> .
> .
> .
> useEffect(() => {
>     if (currentRotation === 'top' || currentRotation === 'down')
>       setCameraOrientation('portrait');
>     if (currentRotation === 'right') setCameraOrientation('landscape-left');
>     if (currentRotation === 'left') setCameraOrientation('landscape-right');
>   }, [currentRotation]);
> ```

sugaith, 2024-08-20:
from where is this `setOrientation(...)` cames from?

> Hi, the problem is because every 100ms the effect is reading the orientation of the phone, so, every 100ms the camera orientation is changing.
>
> Just create another state like "currentRotation" that [...]

ingfdoaguirre, 2024-08-20:
Sorry, is just my state to manage the current orientation

` const [cameraOrientation, setCameraOrientation] = useState('portrait');`

With this state I rotate the image taken from the camera, based on portrait or landscape, with something like this when is in landscape mode:

`const rotatedImage = await manipulateAsync(
                    picture.uri,
                    [{rotate: 270}, {resize: {width: 960, height: 540}}],
                    {compress: 0.85},
                );`

> from where is this `setCameraOrientation(...)` comes from?
> 
> > > > @mrousavy @jzxchiang1 To get around the orientation lock issue, we can copy the `react-native-camera`'s code that checks the gravity vector to guesstimate the orientation https://github.com/react-native-camera/react-native-camera/blob/master/ios/RN/RNSensorOrientationChecker.m
> > > > Alternatively if you don't wanna touch this library's code you can also use `react-native-sensors` with the hook below
> > > > ```ts
> > > > import { useLayoutEffect, useRef } from 'react';
> > > > import { Platform } from 'react-native';
> > > > 
> > > > import { gravity } from 'react-native-sensors';
> > > > 
> > > > type Rotation = 'top' | 'down' | 'right' | 'left';
> > > > 
> > > > export const useDeviceRotationSensor = (
> > > >   callback: (rotation: Rotation, degree: number) => void,
> > > > ) => {
> > > >   const callbackRef = useRef(callback);
> > > >   callbackRef.current = callback;
> > > >   useLayoutEffect(() => {
> > > >     // We use gravity sensor here because react-native-orientation
> > > >     // can't detect landscape orientation when the device's orientation is locked
> > > >     const subscription = gravity.subscribe(({ x, y }) => {
> > > >       const radian = Math.atan2(y, x);
> > > >       const degree = (radian * 180) / Math.PI;
> > > > 
> > > >       let rotation: Rotation = 'left';
> > > >       if (degree > -135) rotation = 'top';
> > > >       if (degree > -45) rotation = 'right';
> > > >       if (degree > 45) rotation = 'down';
> > > >       if (degree > 135) rotation = 'left';
> > > > 
> > > >       if (Platform.OS === 'android') {
> > > >         rotation = 'right';
> > > >         if (degree > -135) rotation = 'down';
> > > >         if (degree > -45) rotation = 'left';
> > > >         if (degree > 45) rotation = 'top';
> > > >         if (degree > 135) rotation = 'right';
> > > >       }
> > > > 
> > > >       callbackRef.current(rotation, degree);
> > > >     });
> > > >     return () => subscription.unsubscribe();
> > > >   }, []);
> > > > };
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > In your component
> > > > ```tsx
> > > >  useDeviceRotationSensor((rotation) => {
> > > >     // These still work when the device orientation is unlocked
> > > >     setCameraOrientation('landscapeRight');
> > > >     if (rotation === 'top') setCameraOrientation('portrait');
> > > >     if (rotation === 'right') setCameraOrientation('landscapeLeft');
> > > >     if (rotation === 'down') setCameraOrientation('portraitUpsideDown');
> > > >     if (rotation === 'left') setCameraOrientation('landscapeRight');
> > > >   });
> > > > ```
> > > 
> > > 
> > > This works but the camera scene momentarily freezes while the vision camera changes orientation.
> > 
> > 
> > Hi, the problem is because every 100ms the effect is reading the orientation of the phone, so, every 100ms the camera orientation is changing.
> > Just create another state like "currentRotation" that saves the current orientation of the change, and only change it when it really changes, for example, from top to left.
> > This solved the freeze problem, also I had a problem that the app was crashing, because the camera orientation was changing every 100ms.
> > ```
> > const [currentRotation, setCurrentRotation] = useState('top');
> > .
> > .
> > .
> > useDeviceRotationSensor(rotation => {
> >     if (currentRotation !== rotation) {
> >       setCurrentRotation(rotation);
> >     }
> >   });
> > .
> > .
> > .
> > useEffect(() => {
> >     if (currentRotation === 'top' || currentRotation === 'down')
> >       setCameraOrientation('portrait');
> >     if (currentRotation === 'right') setCameraOrientation('landscape-left');
> >     if (currentRotation === 'left') setCameraOrientation('landscape-right');
> >   }, [currentRotation]);
> > ```