margelo/react-native-vision-camera #625: 🐛 Photo element data has no lens, equipment or location information.

mushan0x0, 2021-11-30:
### What were you trying to do?

I hope the photos I took contain more information.

### What happened instead?

Check it in the album after taking the photo directly, but I don’t see more picture information.

### Device

iPhone 11 Pro Max iOS 15

### VisionCamera Version

2.9.3

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

hungdev, 2021-12-06:
+1
i hope it will have lat long information

hungdev, 2022-01-21:
@mrousavy I hope you add location to image <3

tomgreco, 2022-03-17:
@mrousavy  +1 to lat/long

mrousavy, 2023-09-30:
Hey, lat/lng will not be part of VisionCamera, it's not a location library. But lens information is already in there!

closed 2023-09-30 by mrousavy