margelo/react-native-vision-camera #626: 🐛 frame-processor/unavailable: Frame Processors are not enabled

jack828, 2021-12-04:
### What were you trying to do?

Using this + the QR scanner library for QR detection. 

### Reproduceable Code

```tsx
import { observer } from 'mobx-react-lite'

import * as React from 'react'
import 'react-native-reanimated'
import { StyleSheet, View } from 'react-native'
import { useCameraDevices, Camera } from 'react-native-vision-camera'
import { useScanBarcodes, BarcodeFormat } from 'vision-camera-qrcode-scanner'
import { Center, Flex, Heading, Text } from 'native-base'
import { useIsFocused } from '@react-navigation/core'
import Layout from '../../components/layouts/Layout'

export const ScanScreen: React.FC = observer(({ navigation }) => {
  const [hasPermission, setHasPermission] = React.useState(false)
  const devices = useCameraDevices()
  const device = devices.back
  const isFocused = useIsFocused()
  const [frameProcessor, qrCodes] = useScanBarcodes([BarcodeFormat.ALL_FORMATS])
  const [isActive, setIsActive] = React.useState(isFocused)

  React.useEffect(() => {
    ; (async () => {
      const status = await Camera.requestCameraPermission()
      setHasPermission(status === 'authorized')
    })()
  }, [])

  React.useEffect(() => {
    if (qrCodes.length) {
      console.log(qrCodes[0])
      navigation.navigate('nftdetails', { id: qrCodes[0].displayValue })
    }
  }, [qrCodes.length])

  return (
    <Layout noScroll noPadding>
      <View style={styles.overlayContent}>
        <Center pb={16}>
          {qrCodes.map((qrCode, idx) => (
            <Text key={idx} style={styles.qrCodeTextURL}>
              {qrCode.displayValue}
            </Text>
          ))}
        </Center>
      </View>
      {device != null && hasPermission && (
        <Camera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={isFocused}
          frameProcessor={isFocused ? frameProcessor : undefined}
          frameProcessorFps={1}
        />
      )}
    </Layout>
  )
})
```

### What happened instead?

frame-processor/unavailable: Frame Processors are not enabled.

The app works perfectly after a completely clean start - and after code changes (or bundle refresh) I can't use the frameProcessors function at all.

I'm just guessing here - I disabled the `assertFrameProcessorsEnabled` call in `node_modules/react-native-vision-camera/src/Camera.tsx` and I get the error:

```
Invariant Violation: Tried to register two views with the same name CameraView, js engine: hermes
```

So, I can only guess (a proper guess) that this library perhaps is not tearing down views correctly, so next run they are not initialised correctly?

I've managed to find a fix by making the following change:
```diff
-     {device != null && hasPermission && (
+     {device != null && hasPermission && isFocused && (
```

But I felt like it could be something that needs fixing under the hood - or if anyone stumbles across this in a search.

### Device

OnePlus 7 Pro (Android 12)

### VisionCamera Version

2.9.4

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://mrousavy.github.io/react-native-vision-camera/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2021-12-06:
Interesting, does this work on iOS?

jack828, 2021-12-06:
My colleagues on iOS tell me they haven't experienced the issue, but ill update this thread if they do today.

Mohamed-kassim, 2021-12-09:
@mrousavy this also happens with me 
after navigating to screen app crashes with this error, after restarting the app it works fine

mrousavy, 2021-12-13:
Are you sure that you're not using some remote debugger such as Chrome? It makes me wonder that the JSI functions are missing. Or maybe you are using fast-refresh and that breaks?

Can you reproduce the error if you simply open the app on your phone, without any debugger attached, and navigate through screens without fast-refresh or hot-reload?

jack828, 2021-12-13:
Not using Chrome debugger - doesn't work with <insert package that I forget now> so I've been using flipper when required, but this occurs if flipper is disconnected or not.

Disabling fast refresh did not fix the issue if it already occurs.
Tried this too:
 - From a fresh app (`react-native run-android`) on device
 - Disabling fast refresh (no option for hot-reload) 
 - Clean bundle `yarn install && yarn start --reset-cache`
 - Navigate to page with camera, allow permission
 - Works fine
 - Navigate away from page
 - Error `Exception in HostFunction: com.mrousavy.camera.ViewNotFoundError: [system/view-not-found] The given view (ID 2559) was not found in the view manager.`
 - Reload app via metro bundler
 - Navigate back to camera screen -> `frame processors are not enabled`

Hope this helps somewhat!

mrousavy, 2021-12-13:
Ah, this is a race condition in the C++ view finder part. (`findViewById`, called on JS Thread through JNI, so C++ -> JS -> C++)

I have tried several times to fix this issue, but all of those times turned into long nights of debugging and after the morning light started to shine through my window I didn't want to spend more time on it. If you are interested in posting a bug bounty on the issue, let me know and me and my guys in my agency can look at this issue in greater detail to fix it once and for all.

jack828, 2021-12-13:
No worries. If you don't mind linking any relevant code, I will see if I can't obtain time myself to see what I can do.

Appreciate it all the same!

Kprashanth20, 2021-12-18:
by when this PR will be merged?

mrousavy, 2022-01-02:
Please see if #691 fixes the issue for you guys 🙏

jack828, 2022-01-04:
@mrousavy almost!

I no longer experience the original issue on reload, however navigating away from the screen that uses frame processors gives me

`com.mrousavy.camera.ViewNotfoundError: [system/view-not-found] The given view (ID 12429) was not found in the view manager`

I tried nuking node_modules & gradlew clean and no change.

mrousavy, 2022-01-04:
Hm, okay that's a separate concurrency issue. will take a look soon

Oz-on, 2022-01-13:
I'm getting this error as well

vikasacharya16, 2022-01-24:
in my system `node` version is `12`.  just installing the package and did changes in android manifest and other files given in docs.
```
VisionCamera: Frame Processors are disabled because REA v2 does not exist. C++ part will not be built.

> Task :app:checkDebugAarMetadata FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/6.9/userguide/command_line_interface.html#sec:command_line_warnings
9 actionable tasks: 2 executed, 7 up-to-date

FAILURE: Build failed with an exception.
```

mrousavy, 2022-03-30:
> that's all, one will get this error.

@vikasacharya16 the output you shared **is not an error**.

Please read up on what errors are and how to identify them, this is a simple log output. The actual error is further up.

mrousavy, 2022-03-30:
@jack828 Hey! Does [v2.13.1](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.13.1) fix the issue for you?

jeffBoom, 2022-11-17:
I re-built and re-ran the app while also restarting metro with --reset-cache and that seemed to fix my issue. I also ran npm update but it didn't appear to update much in package.json, just some deps of deps.

emaxedon, 2022-11-24:
I'm getting this same error running in debub on Android emulator...

[image]

- "react-native-vision-camera": "2.15.2",
- "react-native-reanimated": "2.13.0",
-  "vision-camera-code-scanner": "0.2.0",
- "react-native": "0.68.2",
- "react": "17.0.2",

`import React, { useEffect, useState } from 'react'
import { View, Alert, StyleSheet } from 'react-native'

import styles from './styles'

import { Camera, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera'
import { BarcodeFormat, scanBarcodes, Barcode } from 'vision-camera-code-scanner'
import { runOnJS } from 'react-native-reanimated'

const QRReader: React.FC = () => {
  const [loading, setLoading] = useState<boolean>(false)
  const devices = useCameraDevices()
  const device = devices.back
  const [hasPermission, setHasPermission] = useState<boolean>(false)
  const [barcodes, setBarcodes] = useState<Barcode[]>()

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'

    const detectedBarcodes = scanBarcodes(frame, [BarcodeFormat.QR_CODE], { checkInverted: true })
    runOnJS(setBarcodes)(detectedBarcodes)
  }, [])

  useEffect(() => {
    ;(async () => {
      const status = await Camera.requestCameraPermission()
      setHasPermission(status === 'authorized')
    })()
  }, [])

  useEffect(() => {
    if (barcodes && barcodes.length) {
      console.log('barcodes', barcodes);
    }
  }, [barcodes])

  return (
    <View style={styles.container}>
      {device != null && hasPermission && (
        <>
          <Camera
            style={StyleSheet.absoluteFill}
            isActive={true}
            device={device}
            frameProcessor={frameProcessor}
            frameProcessorFps={1}
          />
        </>
      )}
    </View>
  )
}

export default QRReader
`

emaxedon, 2022-11-24:
> I'm getting this same error running in debug on Android emulator...
> 
> [image]
> 
> * "react-native-vision-camera": "2.15.2",
> * "react-native-reanimated": "2.13.0",
> * "vision-camera-code-scanner": "0.2.0",
> * "react-native": "0.68.2",
> * "react": "17.0.2",
> 
> ```
> import 'react-native-reanimated'
> import React, { useEffect, useState } from 'react'
> import { View, Alert, StyleSheet } from 'react-native'
> 
> import styles from './styles'
> 
> import { Camera, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera'
> import { BarcodeFormat, scanBarcodes, Barcode } from 'vision-camera-code-scanner'
> import { runOnJS } from 'react-native-reanimated'
> 
> const QRReader: React.FC = () => {
>   const [loading, setLoading] = useState<boolean>(false)
>   const devices = useCameraDevices()
>   const device = devices.back
>   const [hasPermission, setHasPermission] = useState<boolean>(false)
>   const [barcodes, setBarcodes] = useState<Barcode[]>()
> 
>   const frameProcessor = useFrameProcessor((frame) => {
>     'worklet'
> 
>     const detectedBarcodes = scanBarcodes(frame, [BarcodeFormat.QR_CODE], { checkInverted: true })
>     runOnJS(setBarcodes)(detectedBarcodes)
>   }, [])
> 
>   useEffect(() => {
>     ;(async () => {
>       const status = await Camera.requestCameraPermission()
>       setHasPermission(status === 'authorized')
>     })()
>   }, [])
> 
>   useEffect(() => {
>     if (barcodes && barcodes.length) {
>       console.log('barcodes', barcodes);
>     }
>   }, [barcodes])
> 
>   return (
>     <View style={styles.container}>
>       {device != null && hasPermission && (
>         <>
>           <Camera
>             style={StyleSheet.absoluteFill}
>             isActive={true}
>             device={device}
>             frameProcessor={frameProcessor}
>             frameProcessorFps={1}
>           />
>         </>
>       )}
>     </View>
>   )
> }
> 
> export default QRReader
> ```

After upgrade to "react-native": "0.70.6", the problem is still happening...

I can't get away from this "frame-processor/unavailable: Frame Processors are not enabled" error.

While building, it seems to log that "frame processors are enabled!"

[image]

So I'm not sure why it is throwing this error. I cannot use the "frameProcessor" prop on Camera.

emaxedon, 2022-11-24:
Finally I was able to resolve this by upgrading my emulator SDK version to 31.

Ankushpandey-ti, 2022-11-29:
Hey @emaxedon  I am facing same issue on the development build on my device (One Plus 9r). 
Surprisingly its working on my emulator . 
How do i solve this problem?
 PS - I am relatively new to react-native

emaxedon, 2022-11-30:
> Hey @emaxedon I am facing same issue on the development build on my device (One Plus 9r). Surprisingly its working on my emulator . How do i solve this problem? PS - I am relatively new to react-native
> 
> My camerascreen code - `import { Button, StyleSheet, Text, View ,TouchableOpacity} from "react-native"; import { StatusBar } from "expo-status-bar"; import * as MediaLibrary from "expo-media-library"; import * as Permissions from "expo-permissions"; import { useRef, useState, useEffect } from "react"; import styles from "../styles"; import { Ionicons } from "@expo/vector-icons"; import { useIsFocused } from "@react-navigation/native"; import {useCameraDevices,Camera,CameraPermissionStatus,frameRateIncluded,useFrameProcessor} from 'react-native-vision-camera'; import { examplePlugin } from './frame-Processors/ExamplePlugin'; import {useFrameCallback} from 'react-native-reanimated';
> 
> export default function CameraScreen({ navigation }) { const [cameraDevice, setCameraDevice] = useState('back'); const [recording, setRecording] = useState(false); const [flash, setFlash] = useState('off'); const [cameraPermission, setCameraPermission] = useState(); const [microphonePermission, setMicrophonePermission] = useState()
> 
> const cameraRef = useRef(Camera) const isFocused = useIsFocused(); const devices = useCameraDevices();
> 
> var device = devices[cameraDevice];
> 
> const frameProcessor = useFrameProcessor((frame) => { 'worklet'; const values = examplePlugin(frame); console.log(`Return Values: ${JSON.stringify(values)}`); }, []);
> 
> useEffect(() => { Camera.getCameraPermissionStatus().then(setCameraPermission); Camera.getMicrophonePermissionStatus().then(setMicrophonePermission); if (cameraPermission!='authorized'){ Camera.requestCameraPermission().then(setCameraPermission); } if (microphonePermission!='authorized'){ Camera.requestMicrophonePermission().then(setMicrophonePermission); } }, [setCameraPermission,setMicrophonePermission]);
> 
> console.log(`Re-rendering Navigator. Camera: ${cameraPermission} | Microphone: ${microphonePermission}`);
> 
> if (!device) { return null; }
> 
> async function toggleCameraType(){ await setCameraDevice(cameraDevice => (cameraDevice=='back'?'front':'back')) device = devices[cameraDevice]; }
> 
> function startRecording(){ try { if (cameraRef.current == null) throw new Error('Camera ref is null!');
> 
> ```
>   console.log('calling startRecording()...');
>   cameraRef.current.startRecording({
>     flash: flash,
>     onRecordingError: (error) => {
>       console.error('Recording failed!', error);
>       stopRecording();
>     },
>     onRecordingFinished: (video) => {
>       console.log(`Recording successfully finished! ${video.path}`);
>       navigation.navigate('ReelPreview',{videoURI:video.path});
>       stopRecording();
>     },
>   });
>   // TODO: wait until startRecording returns to actually find out if the recording has successfully started
>   console.log('called startRecording()!');
>   setRecording(true);
> } catch (e) {
>   console.error('failed to start recording!', e);
> }
> // alert('Starting Recording')
> // console.log("Lets Record")
> // cameraRef.current?.startRecording({
> //   flash: 'on',
> //   onRecoringFinished: (video) => {alert(video);console.log(video)},
> //   onRecordingError: (error) => {alert(error);console.log(error)}
> // })
> ```
> 
> } async function stopRecording (){ console.log('Recording Stopped!') await cameraRef.current?.stopRecording() } function manageFlash(){ if(!device.hasFlash){ alert('Camera does not has flash'); } console.log(flash) setFlash(flash => (flash==='on'?flash='off':flash='on')) }
> 
> return ( {/* {isFocused && hasPermission && ( */} <Camera ref = {cameraRef} photo={true} isActive={isFocused} device={device} whiteBalance="auto" type="back" style={StyleSheet.absoluteFill} video ={true} audio ={true} enableZoomGesture={true} frameProcessor={device.supportsParallelVideoProcessing ? frameProcessor : undefined} frameProcessorFps={60}
> 
> ```
>     />
>     <View style={{ flex: 9 }} />
>       <View
>         style={{
>           flex: 1,
>           justifyContent: "space-around",
>           alignItems: "center",
>           backgroundColor: "#1c1c1c",
>           flexDirection: 'row'
>         }}
>         >
>         <Ionicons
>           name={recording ? "stop-circle" : "play-circle"}
>           size={60}
[... 32 lines cut ...]
> Example plugin -
> 
> `/* global __example_plugin __example_plugin_swift */ import type { Frame } from 'react-native-vision-camera';
> 
> declare let _WORKLET: true | undefined;
> 
> export function examplePluginSwift(frame: Frame): string[] { 'worklet'; if (!_WORKLET) throw new Error('examplePluginSwift must be called from a frame processor!');
> 
> // @ts-expect-error because this function is dynamically injected by VisionCamera return __example_plugin_swift(frame, 'hello!', 'parameter2', true, 42, { test: 0, second: 'test' }, ['another test', 5]); }
> 
> export function examplePlugin(frame: Frame): string[] { 'worklet'; if (!_WORKLET) throw new Error('examplePlugin must be called from a frame processor!');
> 
> // @ts-expect-error because this function is dynamically injected by VisionCamera return __example_plugin(frame, 'hello!', 'parameter2', true, 42, { test: 0, second: 'test' }, ['another test', 5]); }`
> 
> Bable.config.json -
> 
[... cut ...] sisAbstractAnalyzer(ImageAnalysisAbstractAnalyzer.java:283) androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run(Unknown Source:14) java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) java.lang.Thread.run(Thread.java:923)
> 
> [image]

Hey, @Ankushpandey-ti - you need to ensure that your device is running Google Android API 31 or greater (Android 12 or greater).

Ankushpandey-ti, 2022-11-30:
Hey , 
@emaxedon  , i tried the development build of my app on android 13, still getting same error.
Anyidea why is it so?

emaxedon, 2022-11-30:
@Ankushpandey-ti In this case, it is unlikely that it is tied to the android SDK version. I see in your error logs that "**Frame processors are not enabled**". This will definitely cause issues.

I also notice you are not directly importing the entire "react-native-reanimated" module...

```
import {useFrameCallback} from 'react-native-reanimated';
```

should be

```
import 'react-native-reanimated';
import {useFrameCallback} from 'react-native-reanimated';
```

Ankushpandey-ti, 2022-11-30:
Hey ,@emaxedon  
I did a rebuild for android with no changes now the error is gone but the development app keeps crashing on starting the camera without any log or error. 
It some times gives a pop up the APP keeps stopping. 
One more thing that i noticed is if i keep the isActive = {false} , then the app doesn't closes. 
It closes after  i turn the isActive ={true}

Also its working fine on emulator as it was earlier.

DeepikaSharma5, 2022-12-27:
Any solution for this? @mrousavy

Ankushpandey-ti, 2022-12-27:
> Any solution for this? @mrousavy

@DeepikaSharma5  The solution that worked for me was.
1. I updated my android device to Andorid 13 as suggested by @emaxedon  in above comments.
2. Insted of directly using true false for the isActive tag is used isFocuses.

Here is the sample code for better understanding

Not working code :-
`<Container>
        {isFocused && <Camera
          ref={cameraRef}
          photo={true}
          isActive={true}
          device={device}
          style={StyleSheet.absoluteFill}
          video={true}
          audio={true}
          enableZoomGesture={true}
        />}
      </Container>
 `

Working Code part :-
 ` const isFocused = useIsFocused();
     <Container>
        {isFocused && <Camera
          ref={cameraRef}
          photo={true}
          isActive={isFocused}
          device={device}
          style={StyleSheet.absoluteFill}
          video={true}
          audio={true}
          enableZoomGesture={true}
        />}
      </Container>
      `

allangsilva, 2023-01-10:
In my case what solved this was add the Processor plugin class package to the `MainApplication.java` file in `getPackages` method.

Like:

```
@Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
         ...
         new VisionCameraFaceDetectorPluginPackage() // -> Frame processor 
      );
    }
```

Bamorem, 2023-01-15:
I got the same issue but on IOS. And i follow all the steps on https://mrousavy.com/react-native-vision-camera/docs/guides/troubleshooting/

do someone has some some idea ?

in my Podfile i have 
platform :ios, '13'
and ...
 if pod.name.eql?('vision-camera-code-scanner') || pod.name.eql?('VisionCamera')
      def pod.build_type;
        Pod::BuildType.static_library
      end
    end

hassanharoon321, 2023-02-08:
Just close debugger mode on android

mguerrero-exe, 2023-02-16:
> ¿Estás seguro de que no estás usando algún depurador remoto como Chrome? Me pregunto si faltan las funciones JSI. ¿O tal vez estás usando una actualización rápida y eso se rompe?
> 
> ¿Puede reproducir el error si simplemente abre la aplicación en su teléfono, sin ningún depurador adjunto, y navega por las pantallas sin una actualización rápida o una recarga en caliente?

I stopped debugging by chrome and instantly the error went away so it worked for me

icanas, 2023-03-13:
I have finally solve this issue.
For today the latest version of  react-native-reanimated is "3.0.2", this is not compatible with latest version of react-native-camera.
Using version `"react-native-reanimated": "^2.10.0"` solved the issue for me after deleting yarn.lock file and then `yarn install` again.

KirillBorunov, 2023-03-26:
> I have finally solve this issue. For today the latest version of react-native-reanimated is "3.0.2", this is not compatible with latest version of react-native-camera. Using version [...]

Solved the issue for me too!
I installed the 2.14.4 version.
The site says: "If you want to use Frame Processors, you need to install react-native-reanimated 2.2.0 or higher."
3.0.2 is higher than 2.2.0 but not works, maybe you can fix this info on the site?

nguyenduy2392, 2023-05-04:
> Tôi cuối cùng đã giải quyết vấn đề này. Hiện tại, phiên bản mới nhất của react-native-reanimated là "3.0.2", phiên bản này không tương thích với phiên bản mới nhất của react-native-camera. Việc sử dụng phiên bản `"react-native-reanimated": "^2.10.0"`đã giải quyết vấn đề cho tôi sau khi xóa tệp yarn.lock rồi `yarn install`xóa lại.

my version is ^2.9.0. when update to 2.10.0 it work fine

Yves97, 2023-05-04:
> I have finally solve this issue. For today the latest version of react-native-reanimated is "3.0.2", this is not compatible with latest version of react-native-camera. Using version [...]

thank you, its work for me

opeyemijj, 2023-07-08:
If anyone is facing this issue, it is most likely because you are using a package like Firebase that requires   use_frameworks! :linkage => :static. You need specify the build types for RNReanimated and Vision camera. Add the block below to your pod file and it should work.

pre_install do |installer|
    installer.pod_targets.each do |pod|
      if pod.name.eql?('RNReanimated') || pod.name.eql?('VisionCamera')
        def pod.build_type
          Pod::BuildType.static_library
        end
      end
    end
  end

Marcos-Botene, 2023-08-11:
> 
It worked for me, and I didn't need to update reanimated, just add this line in the Podfile

peterdee, 2023-09-11:
Hi. I had this issue when running my app on Android device:

```text
 LOG  Running "viewfinder" with {"rootTag":11}
 ERROR  system/frame-processors-unavailable: Frame Processors are not enabled!, js engine: hermes
```

I've installed [react-native-worklets-core](https://github.com/margelo/react-native-worklets-core) as it is suggested in the docs (https://www.react-native-vision-camera.com/docs/guides/frame-processors), and I was no longer getting the error mentioned above. 

However, my app started to crash / close when launching the camera. 

There was no error output, just this:

```text
 LOG  Running "viewfinder" with {"rootTag":11}
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
```

Module versions:

```text
    "react-native": "0.72.4",
    "react-native-vision-camera": "^3.0.0",
    "react-native-worklets-core": "^0.2.0"
```

I've deleted `node_modules` and `package-lock.json` and did a clean re-install of the modules.

After that the app could no longer be launched or built, I was getting the following output:

```text
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/Users/peter/Projects/viewfinder/node_modules/react-native-vision-camera/android/.cxx/Debug/4a1m2e2g/arm64-v8a'
  [1/1] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/4a1m2e2g/obj/arm64-v8a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cxx/Debug/4a1m2e2g/obj/arm64-v8a/libVisionCamera.so 
  : && /Users/peter/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/Users/peter/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -g  -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Qunused-arguments -Wl,--no-undefined -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/4a1m2e2g/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/Users/peter/Projects/viewfinder/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o  /Users/peter/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so  -landroid  /Users/peter/.gradle/caches/transforms-3/6fe00d6361b0f0c55549aec89a7f6742/transformed/jetified-react-android-0.72.4-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /Users/peter/.gradle/caches/transforms-3/6fe00d6361b0f0c55549aec89a7f6742/transformed/jetified-react-android-0.72.4-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so  /Users/peter/.gradle/caches/transforms-3/34c2120d0d3cd8dc97826fc02c9e8fc4/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  -lGLESv2  -lEGL   -latomic -lm && :
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
  >>> referenced by JFrameProcessor.cpp:37 (/Users/peter/Projects/viewfinder/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:37)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  >>> referenced by JVisionCameraProxy.cpp:57 (/Users/peter/Projects/viewfinder/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:57)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::~JVisionCameraProxy())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnJsThread(std::__ndk1::function<void (facebook::jsi::Runtime&)>&&)
  >>> referenced by JFrameProcessor.cpp:50 (/Users/peter/Projects/viewfinder/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:50)
[... cut ...]
I uninstalled `react-native-worklets-core`, removed `react-native-worklets-core` from `babel.config.js`, removed `worklet` notation from `useFrameProcessor()` handler and tried to launch the app, but I am back from where I started - 
I get `system/frame-processors-unavailable: Frame Processors are not enabled!, js engine: hermes` error.

I am not using Reanimated.

App and camera works fine if I am not using `frameProcessor` property on `<Camera />`.

I am testing this on a device with **Android 10**, I'll try to test on other Android versions.

w4t3r-45, 2023-09-19:
any solutions so far , am still facing this issue in both IOS and Android

nkloi, 2023-10-02:
The same with w4t3r-45
`
"react-native": "0.72.5",
"react-native-vision-camera": "3.0.0",
"react-native-reanimated": "3.0.0",
`

peterdee, 2023-10-03:
@w4t3r-45 Hi. Could you please provide your `package.json` file and Android version that you use for testing?

w4t3r-45, 2023-10-03:
> @w4t3r-45 Hi. Could you please provide your `package.json` file and Android version that you use for testing?
this are the versions of packages am using , besides that i tested on Android 10 physical device and on IOS 16.6 also physical device. 

"react-native": "0.71.13",
"react-native-reanimated": "^3.5.4",
"react-native-vision-camera": "^2.16.1",

peterdee, 2023-10-03:
@w4t3r-45 Thank you, I will try your solution.

w4t3r-45, 2023-10-03:
@peterdee just let me know if you have any issues.

nicks78, 2023-10-11:
@w4t3r-45 , I have tried your solution but I am still getting the same issue. 

`ReanimatedError: Property '__scanCodes' doesn't exist, js engine: reanimated`

"react-native": "0.71.4",
"react-native-reanimated": "^3.5.4",
"react-native-vision-camera": "^2.16.1",

ombogdanJoinToIt, 2023-10-19:
still breaking in IOS 17
The Camera is not ready yet! Wait for the onInitialized() callback!

samsonroyquickplay, 2023-10-23:
Still breaking on iOS 17. Please help!

karel-suchomel-ed, 2023-10-23:
The same issue after updating to iOS 17.

```
"react-native-vision-camera": "^2.15.2",
"react-native": "0.70.6",
"react-native-reanimated": "2.14.4"
```

tomyrodeghiero, 2023-10-24:
Hi! 

I've spent several hours working to resolve this issue with the react-native-vision-camera package [📷](https://emojipedia.org/es/c%C3%A1mara-de-fotos). I'd like to share the potential solution I've found. Here's a step-by-step guide: 

1. **Ensure Package Versions:** First and foremost, please ensure these versions are specified in your **package.json**:

**react-native-vision-camera:** 2.15.2
**react-native-reanimated:** 2.17.0
**vision-camera-code-scanner:** 0.2.0
**react-dom:** ^18.2.0
**react-native:** 0.70.6

2. **Android Cleanup:** Navigate to the Android directory and execute the cleaning command:

`cd android && ./gradlew clean`

3. **Yarn Cache and Android Start:** Clear the yarn cache and initiate the Android application with:

`yarn cache clean && yarn run android`

4. **iOS Build Cleanup:** For iOS concerns, please delete the build folder:

`rm -rf ios/build`

5. **CocoaPods Cleanup:** Refresh your CocoaPods configurations in the iOS directory:

```
cd ios/
pod cache clean --all
pod deintegrate
pod install
```

_**Additional Cleanup:**_ For a thorough cleanup, consider these steps:

```
watchman watch-del-all
rm -rf node_modules/ && npm install
rm -rf /tmp/metro-*.lock
```

**🧩 Frame Processors are now enabled with these changes!**

As a guiding note for future endeavors, for those starting new projects, I highly recommend using **react-native-vision-camera version 3**. This updated version offers enhanced features, better stability, and consistent support.

Furthermore, a special shoutout to [mrousavy](https://github.com/mrousavy) is essential. The development and enhancement of react-native-vision-camera have been monumental for the React Native community. Thank you, mrousavy, for your dedication and for arming developers with such invaluable tools.

Greetings!

hassanharoon321, 2023-11-14:
`/* eslint-disable eol-last */
/* eslint-disable react-hooks/exhaustive-deps */
import * as React from 'react';

import {
  Alert,
  BackHandler,
  Platform,
  StatusBar,
  StyleSheet,
  Text,
  View,
} from 'react-native';
import {useCameraDevices} from 'react-native-vision-camera';
import {Camera} from 'react-native-vision-camera';
import {useScanBarcodes, BarcodeFormat} from 'vision-camera-code-scanner';
import {useIsFocused} from '@react-navigation/native';
import CustomAppStatusBar from '../../Helpers/AppBar';
import BackNavigation from '../../Components/BackNavigation/BackNavigation';
import {
  widthPercentageToDP as wp,
  heightPercentageToDP as hp,
} from 'react-native-responsive-screen';

const styles = StyleSheet.create({
  barcodeTextURL: {
    fontSize: 20,
    color: 'white',
    fontWeight: 'bold',
  },
  parentCam: {
    flex: 1,
    backgroundColor: '#0D6E72',
  },
  camOne: {
    height: hp('7%'),
    width: wp('100%'),
  },
  camTwo: {
    marginTop: 0,
    paddingTop: 0,
    height: hp('93%'),
    width: wp('100%'),
  },
});

export default function Scanner({navigation}) {
  const [hasPermission, setHasPermission] = React.useState(false);
  const devices = useCameraDevices();
  const device = devices.back;

  const isFocused = useIsFocused();

  const [frameProcessor, barcodes] = useScanBarcodes([BarcodeFormat.QR_CODE], {
    checkInverted: true,
  });

  const filteredBarcodesRef = React.useRef([]);

  React.useEffect(() => {
    const backAction = () => {
      navigation.navigate('Dashboard');
      return true;
    };

    const backHandler = BackHandler.addEventListener(
      'hardwareBackPress',
      backAction,
    );

    return () => backHandler.remove();
  }, []);

  React.useEffect(() => {
    (async () => {
      const status = await Camera.requestCameraPermission();
      setHasPermission(status === 'authorized');
    })();
  }, []);

  React.useEffect(() => {
    // Update the filteredBarcodesRef with the detected barcodes
    filteredBarcodesRef.current = barcodes.filter(
      barcode => barcode.format === BarcodeFormat.QR_CODE,
    );

    // Check if the filteredBarcodes array has any barcodes
    if (filteredBarcodesRef.current.length > 0) {
      navigation.navigate('TransactionWithCamera', {
        filteredBarcodes: filteredBarcodesRef.current,
      });

      // Reset filteredBarcodesRef after navigation
      filteredBarcodesRef.current = [];
    }
  }, [barcodes, navigation]);

  return (
    device != null &&
    hasPermission &&
    isFocused && (
      <View style={styles.parentCam}>
        <StatusBar animated={true} backgroundColor="#0D6E72" />
        <CustomAppStatusBar
          backgroundColor={`${Platform.OS === 'ios' ? '#fff' : '#0D6E72'}`}
          barStyle="dark-content"
        />
        <View style={styles.camOne}>
          <BackNavigation name={'Qr Code'} route={'Dashboard'} />
        </View>

        <View style={styles.camTwo}>
          <StatusBar animated={true} backgroundColor="#0D6E72" />
          <CustomAppStatusBar
            backgroundColor={`${Platform.OS === 'ios' ? '#fff' : '#0D6E72'}`}
            barStyle="dark-content"
          />

          <Camera
            style={StyleSheet.absoluteFill}
            device={device}
            isActive={true}
            frameProcessor={frameProcessor}
            frameProcessorFps={5}
          />
          {/* {filteredBarcodes.map((barcode, idx) => (
            <Text key={idx} style={styles.barcodeTextURL}>
              {barcode.displayValue}
            </Text>
          ))} */}
        </View>
      </View>
    )
  );
}
`

still shows me an error frameprocessor is not defined