margelo/react-native-vision-camera #627: ✨  `CameraCaptureError` error handling: Handle no storage space

paulrostorp, 2021-12-05:
### What feature or enhancement are you suggesting?

Currently, if a recording fails to save for lack of storage space, we get a `capture/unknown` error code. I would be nice if there was a `capture/no-storage` error or something.

Here is the `CameraCaptureError` on ios for such an error:

```json
{
  "cause": {
    "NSLocalizedDescription": "Operation Stopped",
    "NSLocalizedRecoverySuggestion": "There is not enough available space to continue the file writing. Make room by deleting existing videos or photos.",
    "NSUnderlyingError": null,
    "message": "AVFoundationErrorDomain: -11807 Error Domain=AVFoundationErrorDomain Code=-11807 \"Operation Stopped\" UserInfo={NSLocalizedRecoverySuggestion=There is not enough available space to continue the file writing. Make room by deleting existing videos or photos., NSLocalizedDescription=Operation Stopped, NSUnderlyingError=0x2816a0e10 {Error Domain=NSPOSIXErrorDomain Code=28 \"No space left on device\"}}"
  },
  "code": "capture/unknown",
  "message": "capture/unknown: An unknown recording error occured! Error Domain=AVFoundationErrorDomain Code=-11807 \"Operation Stopped\" UserInfo={NSLocalizedRecoverySuggestion=There is not enough available space to continue the file writing. Make room by deleting existing videos or photos., NSLocalizedDescription=Operation Stopped, NSUnderlyingError=0x2816a0e10 {Error Domain=NSPOSIXErrorDomain Code=28 \"No space left on device\"}}"
}
```

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

parsing the error message string to identify this error is a workaround

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2021-12-06:
Yes, makes sense! How does this look on Android?

mrousavy, 2024-01-19:
Created #2422 for iOS 👍

closed 2024-01-24 by mrousavy